xtrx-sdr/images #3: dkms build error

hmne, 2018-09-25:
hi after i enter 

/usr/sbin/dkms build -m xtrx -v "0.0.1-1" 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.18.0-kali1-amd64 -C /lib/modules/4.18.0-kali1-amd64/build M=/var/lib/dkms/xtrx/0.0.1-1/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.18.0-kali1-amd64 (x86_64)
Consult /var/lib/dkms/xtrx/0.0.1-1/build/make.log for more information.

the log info 

DKMS make.log for xtrx-0.0.1-1 for kernel 4.18.0-kali1-amd64 (x86_64)
Mon Sep 24 20:41:27 EDT 2018
make: Entering directory '/usr/src/linux-headers-4.18.0-kali1-amd64'
  CC [M]  /var/lib/dkms/xtrx/0.0.1-1/build/xtrx.o
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c: In function ‘xtrx_uart_init’:
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c:514:17: error: ‘struct timer_list’ has no member named ‘data’
  dev->uart_timer.data = (unsigned long )dev;
                 ^
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c:515:27: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
  dev->uart_timer.function = xtrx_uart_timer;
                           ^
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c: In function ‘xtrx_probe’:
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c:1061:8: error: implicit declaration of function ‘pci_enable_msi_range’; did you mean ‘pci_enable_msix_range’? [-Werror=implicit-function-declaration]
  err = pci_enable_msi_range(pdev, XTRX_MSI_COUNT, XTRX_MSI_COUNT);
        ^~~~~~~~~~~~~~~~~~~~
        pci_enable_msix_range
At top level:
/var/lib/dkms/xtrx/0.0.1-1/build/xtrx.c:702:20: warning: ‘xtrx_msi_irq_single’ defined but not used [-Wunused-function]
 static irqreturn_t xtrx_msi_irq_single(int irq, void *data)
                    ^~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-4.18.0-kali1-common/scripts/Makefile.build:329: /var/lib/dkms/xtrx/0.0.1-1/build/xtrx.o] Error 1
make[2]: *** [/usr/src/linux-headers-4.18.0-kali1-common/Makefile:1518: _module_/var/lib/dkms/xtrx/0.0.1-1/build] Error 2
make[1]: *** [Makefile:146: sub-make] Error 2
make: *** [Makefile:8: all] Error 2
make: Leaving directory '/usr/src/linux-headers-4.18.0-kali1-amd64'

arglas864, 2018-09-26:
Hi, I'm having the same issue with DKMS driver. I'm runnig Ubuntu 16.04 x86 64 bit. Did you manage to solve the issue?

hmne, 2018-09-26:
> Hi, I'm having the same issue with DKMS driver. I'm runnig Ubuntu 16.04 x86 64 bit. Did you manage to solve the issue?

I didn't solve this issue 

but i think if i use the xtrx without usb3 adapter

zehm, 2018-10-03:
I'm having the same issue as well. I'm sure it's a simple fix, but I haven't had time to write kernel code yet so I feel a bit helpless ;-)

hmne, 2018-10-03:
the issue fixed download the new files 

https://github.com/xtrx-sdr/images

chemeris, 2018-10-03:
Closing this issue now. Thank you, everyone, for reporting and apologies for not pushing the fix earlier. Sergey has been cleaning up a lot of code which went into this push.

closed 2018-10-03 by chemeris