xtrx-sdr/images #4: Windows drivers?

OhSoGood, 2018-10-05:
Hi,
I'm pleased to have received my xtrx today. But where are windows drivers?

sergforce, 2018-10-08:
Hi,

As we stated initially during the campaign we had plans to do windows driver but it's in development now and completely unusable. I'd suggest either to use USB3 adapter on Windows or switch to Linux.

DG9BFC, 2019-02-01:
i also received my xtrx today ... i wanna use it with the above mentioned usb3 adaptor ... but my win7 system does not find any driver
Usb3 + 1 PCIe + RC mode - 88 pin single row QFN (no driver found)
so .. how can i use the xtrx if windows does not find a driver??
greetz sigi
Ps i am no linux guy .. hopefully i do not own a nice letter weight now

DG9BFC, 2019-02-02:
ok ... i found drivers ...and installed them (was tricky but ok so be it) ... now xtrx is in the devices list in windows ... so it "seems" to work with those drivers ... but not found in sdrangel even if the programmer said it IS supported in sdrangel it is not found in the devices list
or did i install the wrong drivers??? or do i have to copy the "x64 drivers manually?? if yes ... in what folder??

OhSoGood, 2019-05-04:
@sergforce : on October 8, 2018, you wrote the windows driver was in development. Any news 7 months after? I'm sure many people would enjoy it!

DG9BFC, 2019-05-07:
i could install the drivers in both my windows machines .. so xtrx is found by the system
but ... sadly no programmer of any sdrsoft included it yet ... and there is no extio driver for hdsdr available ... so ... a nice but expensive paperweight ... i will send mine to a programmer and hope he then will include it .... greetz sigi dg9bfc

OhSoGood, 2019-05-07:
Could you share the driver bin & sources and describe how you compiled it?
Regarding sdr soft, I thought there was at least a soapy driver ( 
https://github.com/xtrx-sdr/libxtrx/tree/master/soapy ). Have you looked 
into it?
Regards

DG9BFC, 2019-05-09:
i installed the precompiled windows drivers NO IDEA IF THEY WORK!!!) found here
https://github.com/xtrx-sdr/images/tree/master/binaries under windows
i hoped (and still hope) that simon would include it in his lovely sdr console (that i now use with the pluto for working over eshail) ... maybe one day i will send simon my xtrx and he will then (also maybe) include it ... drivers are there ... just the programmers need to include that in their soft