xtrx-sdr/images #5: warnings running test_xtrx -t l2

o8ko8k, 2018-10-05:
I get odd warnings when running anything with the xtrx.
ubuntu16.04:~/source_code/images/sources/build/libxtrx$ sudo ./test_xtrx -t -l2

Im running the xtrx in the mini pcie slot on my thin client acer Veriton-N2620G. The USB errors out which i assume is ok as im not using it that way.

its the warnings from the PCI_3.  I get the same errors when running kalibrate or sdrangel.

****************************
running the test_xtrx
****************************
ubuntu16.04:~/source_code/images/sources/build/libxtrx$ sudo ./test_xtrx -t -l2
10:58:25.559696 ERROR:   USB3380: Unable to open USB3380!
10:58:25.560122 ERROR:   Unable to allocate USB3380 context: error: -6
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
10:58:25.563257 WARNING: NLMS7 PCI_3: LMS VER:7 REV:1 MASK:1 (13841)
10:58:25.585201 WARNING: NLMS7 PCI_3: CGEN: VCO/2=1184000000 k/2=37 int=91 frac=80659
10:58:25.586564 WARNING: NLMS7 PCI_3: CGEN: binary result: 155
10:58:25.587323 WARNING: NLMS7 PCI_3: CGEN: Retuned [156:161] -> 158
Master: 32000000.000000; RX rate: 4000000.000000; TX rate: 0.000000
10:58:25.589707 WARNING: NLMS7 PCI_3: SXR: initial VCO=3600000000 DIVH=2 VCOs:0010
10:58:25.589750 WARNING: NLMS7 PCI_3: SXX: VCO2 N=138 frac=483958
10:58:25.590516 WARNING: NLMS7 PCI_3: SXX: binary result: 139
10:58:25.593635 WARNING: NLMS7 PCI_3: SXR: VCO2 [141;159] -> 150
10:58:25.593660 WARNING: NLMS7 PCI_3: SXX: VCO2 N=138 frac=483958
10:58:25.593957 WARNING: NLMS7 PCI_3: SXX: binary result: 139
10:58:25.594917 WARNING: NLMS7 PCI_3: SXR: VCO2 [141;159] -> 150
RX tunned: 900000000.000000
10:58:25.595300 WARNING: NLMS7 PCI_3: 0x0124[18, 00]
10:58:25.595435 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
10:58:25.595579 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
10:58:25.596459 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
RX bandwidth: 5000000.000000
10:58:25.598324 WARNING: NLMS7 PCI_3: RFE: set_lna(15 -> 15) => 6
RX LNA gain: 15.000000
10:58:25.598575 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
10:58:25.598899 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
10:58:25.599055 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
10:58:25.599203 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
10:58:25.600696 WARNING: NLMS7 PCI_3: 0x0124[1c, 0c]
10:58:25.601165 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
10:58:25.601328 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
RX SAMPLES=524288 SLICE=524288 PARTS=1
PROCESSED RX SLICE 0 /0: res 0 TS:       0      65688 us DELTA     29 us LATE    182 us 524288 samples R=    ffeff8ffcffd DELTA=716617868
XXXXX Overruns:0 Zeros:0
xAND=[0x0000,0x0000,0x0000,0x0000] xOR=[0xffff,0xffff,0xffff,0xffff]
10:58:25.671528 WARNING: NLMS7 PCI_3: 0x0124[1c, 08]
10:58:25.671774 WARNING: NLMS7 PCI_3: 0x0124[1c, 08]
Success!
Processed RX 2 x 3.989 = 7.977 MSPS (WIRE: 31.908622)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s) 

**********************
With Kalibrate-rtl
**********************

Veriton-N2620G:~/source_code/kalibrate-rtl/build/src$ sudo ./kalibrate-xtrx -s GSM850
11:04:37.013230 ERROR:   USB3380: Unable to open USB3380!
11:04:37.013463 ERROR:   Unable to allocate USB3380 context: error: -6
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
11:04:37.016528 WARNING: NLMS7 PCI_3: LMS VER:7 REV:1 MASK:1 (13841)
11:04:37.031985 WARNING: NLMS7 PCI_3: CGEN: VCO/2=1178666644 k/2=34 int=90 frac=699048
11:04:37.034031 WARNING: NLMS7 PCI_3: CGEN: binary result: 151
11:04:37.034378 WARNING: NLMS7 PCI_3: CGEN: Retuned [152:159] -> 155
11:04:37.034586 WARNING: NLMS7 PCI_3: 0x0124[08, 00]
11:04:37.034710 WARNING: NLMS7 PCI_3: TIA: CCOMP=8 CFB=830 RCOMP=0
11:04:37.034854 WARNING: NLMS7 PCI_3: 0x0124[08, 08]
11:04:37.034972 WARNING: NLMS7 PCI_3: TIA: CCOMP=8 CFB=830 RCOMP=0
kalibrate-xtrx: Scanning for GSM-850 base stations.
11:04:37.039388 WARNING: NLMS7 PCI_3: 0x0124[08, 08]
11:04:37.040891 WARNING: NLMS7 PCI_3: 0x0124[0c, 08]
11:04:37.041657 WARNING: NLMS7 PCI_3: TIA: CCOMP=8 CFB=830 RCOMP=

...

10:57:09.232658 WARNING: NLMS7 PCI_3: SXR: initial VCO=3575200000 DIVH=2 VCOs:0010
10:57:09.232686 WARNING: NLMS7 PCI_3: SXX: VCO2 N=137 frac=532353
10:57:09.233481 WARNING: NLMS7 PCI_3: SXX: binary result: 129
10:57:09.235970 WARNING: NLMS7 PCI_3: SXR: VCO2 [130;148] -> 139
10:57:09.235997 WARNING: NLMS7 PCI_3: SXX: VCO2 N=137 frac=532353
10:57:09.236449 WARNING: NLMS7 PCI_3: SXX: binary result: 129
10:57:09.238196 WARNING: NLMS7 PCI_3: SXR: VCO2 [130;148] -> 139
10:57:09.238409 WARNING: NLMS7 PCI_3:  5c1=0010
10:57:09.238437 WARNING: NLMS7 PCI_3:  5c1=3000
10:57:09.238458 WARNING: NLMS7 PCI_3:  5c1=3020
10:57:09.238492 WARNING: NLMS7 PCI_3:  5c1=3020
10:57:09.238515 WARNING: NLMS7 PCI_3:  5c1=0000
10:57:09.238537 WARNING: NLMS7 PCI_3:  5c1=0010
10:57:09.238589 WARNING: NLMS7 PCI_3:  5c1=0000
10:57:09.238615 WARNING: NLMS7 PCI_3:  5c1=0000
10:57:09.238636 WARNING: NLMS7 PCI_3:  5c1=0010
10:57:09.238657 WARNING: NLMS7 PCI_3:  5c1=0010
10:57:09.238757 WARNING: NLMS7 PCI_3:  TX[0]=0000
10:57:09.238838 WARNING: NLMS7 PCI_3:  TX[1]=0000
10:57:09.239040 WARNING: NLMS7 PCI_3:  TX[2]=0000
10:57:09.239090 WARNING: NLMS7 PCI_3:  TX[3]=0000
10:57:09.239140 WARNING: NLMS7 PCI_3:  RX[0]=0042
10:57:09.239218 WARNING: NLMS7 PCI_3:  RX[1]=0012
10:57:09.239269 WARNING: NLMS7 PCI_3:  RX[2]=0000
10:57:09.239319 WARNING: NLMS7 PCI_3:  RX[3]=0000

o8ko8k, 2018-10-05:
**************************************
when running sdrangel...
***************************************
5.10.2018 11:27:38.285 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/iqbalance/_iqbalance_swig.so'
05.10.2018 11:27:38.290 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/noaa/_noaa_swig.so'
05.10.2018 11:27:38.295 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/pager/_pager_swig.so'
05.10.2018 11:27:38.299 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/qtgui/_qtgui_swig.so'
05.10.2018 11:27:38.303 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/trellis/_trellis_swig.so'
05.10.2018 11:27:38.306 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/_uhd_swig.so'
05.10.2018 11:27:38.309 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/video_sdl/_video_sdl_swig.so'
05.10.2018 11:27:38.312 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/vocoder/_vocoder_swig.so'
05.10.2018 11:27:38.315 (W) PluginManager::loadPlugins:  Plugin verification data mismatch in '/usr/local/lib/python2.7/dist-packages/gnuradio/wavelet/_wavelet_swig.so'
7:38.370921 INFO:    Probing 'libxtrxll_pcie.so.0' low-level library
11:27:38.371341 INFO:    Probing 'libxtrxll_libusb3380.so.0' low-level library
11:27:38.376471 ERROR:   USB3380: Unable to open USB3380!
11:27:38.376523 ERROR:   Unable to allocate USB3380 context: error: -6
11:27:38.376625 INFO:    XTRX PCI_29: XTRX Rev4 (04000013)
11:27:38.376641 INFO:    XTRX PCI_29: Device `/dev/xtrx0` was opened
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
11:27:38.376720 INFO:    LMS PMIC: DCDC out set to V18=1880mV V33=3100mV V14=1480mV V12=1340mV
11:27:38.376727 INFO:    FPGA V_IO set to 1800mV
11:27:38.387093 WARNING: NLMS7 PCI_29: LMS VER:7 REV:1 MASK:1 (13841)
11:27:38.390853 INFO:    XTRX PCI_29: Device closing
05.10.2018 11:27:39.533 (I) WebAPIServer::start: starting web API server at http://127.0.0.1:8091

sergforce, 2018-10-08:
XTRX software tries to look USB first that's why you see these errors. I'll refactor this part to get rid of error messages.

These warning messages mostly are LMS7 debugs and also will be cleaned up.

o8ko8k, 2018-10-09:
great thanks! 
any idea why sdrangel is doing this:

11:27:38.390853 INFO: XTRX PCI_29: Device closing   

i suspected it was relative to the warnings but either way xtrx isnt working for me at the moment, this is as far as i have gotten so far.  

thanks,
o8k

kurtulmehtap, 2018-10-16:
We have a similar problem. We get also the "Device Closing". 
See:https://github.com/xtrx-sdr/libxtrx/issues/1

sergforce, 2018-10-20:
I cleaned up a lot of warnings in master, please update.

Can you retest SDRangel, please?

sergforce, 2018-10-23:
That part of the test is for RX only. Specify -T instead -t for TX.

I updated rebased XTRX support for SDRangel. The current master in `libxtrx_new` branch. Also, update all subrepos in `images`.

When you first run I won't see confusing "Device closing". After the first start, you need to select XTRX device.

If you have problems copy all console output here.

sergforce, 2018-10-24:
Great!

I noticed you were running sdrangel under `sudo` which is generally a bad idea. You can install `udev` rules to make `/dev/xtrx0` available for normal users. Just run
```
cp sources/xtrx_linux_pcie_drv/50-xtrx.rules /etc/udev/rules.d/
```
And reboot (or restart `udevd` and reload the driver)

sdrangel support RX only at the moment.

Since there were so many different problems in the issue and main was resolved I close it. If I missed something please open a new bug report with the corresponding topic.

closed 2018-10-24 by sergforce