xtrx-sdr/images #6: TX tuning and rate looks problematic

mpbraendli, 2018-10-17:
Hi all,

I was able to install the libraries incl. Soapy bindings and the PCIe kernel driver on debian stretch.

My TX application [ODR-DabMod](https://github.com/Opendigitalradio/ODR-DabMod) requires a sampling rate of 2048kSps, and can use SoapySDR to drive SDR devices. I've observed the following:

* My application runs and outputs several hundred lines `WARNING: XTRX PCI_13: TX DMA STAT 63|63/33/37/37 RESET:0 Full:8 TxS:4  34/37 FE:2 FLY:3 D:0 TS:2496563 CPL:0000071f`. No RF output measured on the spectrum analyser. So I thought it would be better to use the test tool first.

* `test_xtrx -f 234208000 -t -l1 -s 2048000` seems to work fine, it concludes with `Processed RX 2 x 2.046 = 4.092 MSPS`
* The TX equivalent `test_xtrx -F 234208000 -T -l1 -S 2048000` doesn't seem to respect the `-S` parameter. It underruns and terminates with

```
...
16:05:06.318165 ERROR:   xtrx_send_burst_sync: TX DMA Current skip due to TO buffers: 24, prev: 23 TS:245760 STPS:253952
PROCESSED TX SLICE 0/60: res 0 TS:140308534178676 D     1917 us DELTA     26 us LATE -75705 us  0 samples
PROCESSED TX SLICE 0/61: res 0 TS:140308534178676         22 us DELTA     31 us LATE -77652 us  8192 samples
PROCESSED TX SLICE 0/62: res 0 TS:140308534178676       1900 us DELTA     26 us LATE -77724 us  8192 samples
PROCESSED TX SLICE 0/63: res 0 TS:140308534178676       1963 us DELTA     35 us LATE -77725 us  8192 samples
TX Underruns:8
Success!
Processed RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 5.216 = 10.431 MSPS (WIRE: 41.724609 MB/s)
```

Is this expected, am I doing it wrong? Or are there some issues with TX?

sergforce, 2018-10-19:
Thanks for the feedback, I'll check it on my side.
Does master of ODR-DabMod good enough to test with?

mpbraendli, 2018-10-19:
Yes the `master` branch is good for the test. But you need to set up quite some things. If you need an ETI source (input to ODR-DabMod) contact me directly. Contact at bottom of https://mpb.li/ page.

Were you able to reproduce the `text_xtrx` transmit behaviour?

mathlizee, 2018-10-23:
Hi everyone,

I have the same issue with my TX output on the XTRX. When streaming with a sampling rate of 12.5 Msps, I get a lot of underruns, making the signal unstable. Here's the console output of the example provided with the _libxtrx_ library: `./test_xtrx --dmatx --txfreq 1500000000 --txrate 12500000`

    ...
    10:33:58.290117 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current skip due to TO buffers: 20, prev: 19 TS:425984 STPS:442368
    PROCESSED TX SLICE 0/52: res 0 TS:       0 D      649 us DELTA      3 us LATE -22131 us  0 samples
    PROCESSED TX SLICE 0/53: res 0 TS:       0          3 us DELTA      4 us LATE -22778 us  8192 samples
    PROCESSED TX SLICE 0/54: res 0 TS:       0        633 us DELTA      3 us LATE -22797 us  8192 samples
    PROCESSED TX SLICE 0/55: res 0 TS:       0        641 us DELTA     10 us LATE -22800 us  8192 samples
    PROCESSED TX SLICE 0/56: res 0 TS:       0        642 us DELTA     12 us LATE -22801 us  8192 samples
    PROCESSED TX SLICE 0/57: res 0 TS:       0        643 us DELTA     11 us LATE -22801 us  8192 samples
    PROCESSED TX SLICE 0/58: res 0 TS:       0        650 us DELTA     12 us LATE -22794 us  8192 samples
    PROCESSED TX SLICE 0/59: res 0 TS:       0        616 us DELTA     11 us LATE -22820 us  8192 samples
    PROCESSED TX SLICE 0/60: res 0 TS:       0        664 us DELTA     11 us LATE -22800 us  8192 samples
    PROCESSED TX SLICE 0/61: res 0 TS:       0        642 us DELTA     11 us LATE -22801 us  8192 samples
    10:33:58.296015 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current delayed buffers: 29, prev: 20 TS:507904
    PROCESSED TX SLICE 0/62: res 0 TS:       0 D      659 us DELTA     11 us LATE -22785 us  0 samples
    PROCESSED TX SLICE 0/63: res 0 TS:       0          4 us DELTA      5 us LATE -23431 us  8192 samples
    TX Underruns:5
    Success!
    Processed RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 14.161 = 28.322 MSPS (WIRE: 113.287113 MB/s) 

It seems that there are less underruns as the sampling rate becomes smaller. However, even with a very small sampling rate of 120 Ksps, I still have underruns with my device: 
`./test_xtrx --dmatx --txfreq 1500000000 --txrate 120000 -l1`

    10:45:35.448801 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current skip due to TO buffers: 20, prev: 19 TS:425984 STPS:442368
    PROCESSED TX SLICE 0/52: res 0 TS:       0 D    68243 us DELTA      7 us LATE -2320796 us  0 samples
    PROCESSED TX SLICE 0/53: res 0 TS:       0          9 us DELTA     10 us LATE -2389042 us  8192 samples
    PROCESSED TX SLICE 0/54: res 0 TS:       0      68230 us DELTA      8 us LATE -2389070 us  8192 samples
    PROCESSED TX SLICE 0/55: res 0 TS:       0      68242 us DELTA     24 us LATE -2389071 us  8192 samples
    PROCESSED TX SLICE 0/56: res 0 TS:       0      68241 us DELTA     24 us LATE -2389072 us  8192 samples
    PROCESSED TX SLICE 0/57: res 0 TS:       0      68221 us DELTA     23 us LATE -2389093 us  8192 samples
    PROCESSED TX SLICE 0/58: res 0 TS:       0      68254 us DELTA     24 us LATE -2389081 us  8192 samples
    PROCESSED TX SLICE 0/59: res 0 TS:       0      68244 us DELTA     23 us LATE -2389079 us  8192 samples
    PROCESSED TX SLICE 0/60: res 0 TS:       0      68256 us DELTA     23 us LATE -2389066 us  8192 samples
    PROCESSED TX SLICE 0/61: res 0 TS:       0      68232 us DELTA     24 us LATE -2389076 us  8192 samples
    10:45:36.063211 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current delayed buffers: 29, prev: 20 TS:507904
    PROCESSED TX SLICE 0/62: res 0 TS:       0 D    68272 us DELTA     24 us LATE -2389046 us  0 samples
    PROCESSED TX SLICE 0/63: res 0 TS:       0          9 us DELTA     10 us LATE -2457293 us  8192 samples
    TX Underruns:5
    Success!
    Processed RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 0.137 = 0.274 MSPS (WIRE: 1.096964 MB/s) 

Hoping this helps resolving the issue,
Mathieu

mpbraendli, 2018-10-25:
@sergforce informed my by email there was an issue in the Soapy device driver. With the latest update ODR-DabMod now seems to work, thanks for that fix.

Running `test_xtrx -F 234208000 -T -l1 -S 2048000` now outputs
```
CPU Features: SSE2+ SSE4.1+ AVX+ FMA-
Master: 32768000.000000; RX rate: 0.000000; TX rate: 2048000.000000
TX tunned: 234208000.000000
TX bandwidth: 2000000.000000
TX SAMPLES=16384 SLICE=8192 PARTS=2
PROCESSED TX SLICE 0/0: res 0 TS:       0         65 us DELTA     40 us LATE  -1893 us  8192 x 1 samples
PROCESSED TX SLICE 0/1: res 0 TS:       0         28 us DELTA     38 us LATE  -3826 us  8192 x 1 samples
TX Underruns:0
Success!
Processed 1 devs, each: RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 49.086 = 98.172 MSPS (WIRE: 392.689796 MB/s)
```

which doesn't seem to take into account the `-S` option.

I will also try DAB reception now.

sergforce, 2018-10-25:
The sample rate is set correctly, but the issue why it's estimated wrongly is different.  The utility measures all the time spent to send out the number of samples. But actually, 16k samples was just copied to DMA buffer. It doesn't wait for them to be transmitted out. And if I add these code the error on such small number is huge. I'd recommend increasing the number of cycles to run to see how it's going.

```
$ ./test_xtrx -T -S4e6 -l2
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
Master: 64000000.000000; RX rate: 0.000000; TX rate: 4000000.000000
TX tunned: 450000000.000000
TX bandwidth: 2000000.000000
TX SAMPLES=16384 SLICE=8192 PARTS=2
PROCESSED TX SLICE 0/0: res 0 TS:       0         16 us DELTA     10 us LATE   -996 us  8192 x 1 samples
PROCESSED TX SLICE 0/1: res 0 TS:       0          8 us DELTA      7 us LATE  -2004 us  8192 x 1 samples
TX Underruns:0
Success!
Processed 1 devs, each: RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 191.040 = 382.081 MSPS (WIRE: 1528.322567 MB/s) 
```

And with 10000 iterations:
```
$ ./test_xtrx -T -S4e6 -l2 -C10000 -L1000
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
Master: 64000000.000000; RX rate: 0.000000; TX rate: 4000000.000000
TX tunned: 450000000.000000
TX bandwidth: 2000000.000000
TX SAMPLES=16384 SLICE=8192 PARTS=2
PROCESSED TX SLICE 0/0: res 0 TS:       0         18 us DELTA     11 us LATE   -994 us  8192 x 1 samples
PROCESSED TX SLICE 0/1: res 0 TS:       0          9 us DELTA      8 us LATE  -2000 us  8192 x 1 samples
PROCESSED TX SLICE 1000/0: res 0 TS:       0       1023 us DELTA      1 us LATE -31115 us  8192 x 1 samples
PROCESSED TX SLICE 1000/1: res 0 TS:       0       1021 us DELTA     22 us LATE -31094 us  8192 x 1 samples
PROCESSED TX SLICE 2000/0: res 0 TS:       0       1023 us DELTA      1 us LATE -31111 us  8192 x 1 samples
PROCESSED TX SLICE 2000/1: res 0 TS:       0        986 us DELTA     23 us LATE -31125 us  8192 x 1 samples
PROCESSED TX SLICE 3000/0: res 0 TS:       0       1048 us DELTA      1 us LATE -31119 us  8192 x 1 samples
PROCESSED TX SLICE 3000/1: res 0 TS:       0        963 us DELTA     24 us LATE -31155 us  8192 x 1 samples
PROCESSED TX SLICE 4000/0: res 0 TS:       0       1045 us DELTA      1 us LATE -31151 us  8192 x 1 samples
PROCESSED TX SLICE 4000/1: res 0 TS:       0        978 us DELTA     24 us LATE -31173 us  8192 x 1 samples
PROCESSED TX SLICE 5000/0: res 0 TS:       0       1045 us DELTA      1 us LATE -31141 us  8192 x 1 samples
PROCESSED TX SLICE 5000/1: res 0 TS:       0        952 us DELTA     23 us LATE -31189 us  8192 x 1 samples
PROCESSED TX SLICE 6000/0: res 0 TS:       0       1001 us DELTA      1 us LATE -31182 us  8192 x 1 samples
PROCESSED TX SLICE 6000/1: res 0 TS:       0        990 us DELTA     24 us LATE -31191 us  8192 x 1 samples
PROCESSED TX SLICE 7000/0: res 0 TS:       0       1015 us DELTA      1 us LATE -31219 us  8192 x 1 samples
PROCESSED TX SLICE 7000/1: res 0 TS:       0       1007 us DELTA     24 us LATE -31211 us  8192 x 1 samples
PROCESSED TX SLICE 8000/0: res 0 TS:       0       1049 us DELTA      1 us LATE -31197 us  8192 x 1 samples
PROCESSED TX SLICE 8000/1: res 0 TS:       0        968 us DELTA     23 us LATE -31229 us  8192 x 1 samples
PROCESSED TX SLICE 9000/0: res 0 TS:       0       1021 us DELTA      1 us LATE -31273 us  8192 x 1 samples
PROCESSED TX SLICE 9000/1: res 0 TS:       0       1013 us DELTA     23 us LATE -31260 us  8192 x 1 samples
TX Underruns:0
Success!
Processed 1 devs, each: RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 4.006 = 8.012 MSPS (WIRE: 32.048950 MB/s) 
```

mathlizee, 2018-10-25:
Hi,

I have tested it on my side and it works much better than before.

However, with a sampling rate of 12.5 Msps and with a center frequency of 1575.42 MHz, I can get underruns with 100 or 1000 cycles. They always happen at the beginning of the stream and always total to 12 underruns.

I don't always have the same result with the same configuration (sampling rate, center frequency and cycles). Sometimes I get 0 underruns, sometimes I get 12.

sergforce, 2018-10-25:
Try to provide more delay before transmission (add `-K 131072`) and also high samplerates most ikely need RT priority for working stable

ducnguyenAstro, 2018-10-27:
Hi sergforce,

Thanks for shedding the light for me on this from the beginning.
Running test_xtrx with adding delay can reach sampling rate about 22M without any under-run, while without any delay just get ~ 4M sps.

My question is : 
- If i am using GRC and connect to osmocom sink to feed into XTRX, this error still shows up:
"17:09:49.900108 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current delayed buffers: 455, prev: 446 TS:9349992" even with sps = 300k. 
- And how can I add delay onto the stream from  GRC.

Thank you,

m4ver1cks, 2018-11-24:
We have the same problem in the GRC. 

How can i solve it?

b3sholt, 2019-01-18:
Just started using the xtrx and am running into the same problem.  Running the test_xtrx command runs fine and I get no under-runs.  However, I am trying to use existing flowgraphs that previously used UHD SInk/Source blocks, but I am getting "delayed buffers" on the simplest of flowgraphs using the osmocom Sink block.  Interested in a solution if one exists for GRC users.

sergforce, 2019-01-26:
> Just started using the xtrx and am running into the same problem. Running the test_xtrx command runs fine and I get no under-runs. However, I am trying to use existing flowgraphs that previously [...]
Are you using SoapySDR or our branch of gr-osmosdr with direct XTRX support?

b3sholt, 2019-01-28:
I think the answer to that question is both?  I am definitely using the xtrx branch of gr-osmosdr with the last commit being e238911..., but from my understanding  gr-osmosdr has a dependency on SoapySDR correct? So I installed SoapySDR first, then installed the xtrx branch of gr-osmosdr. Is there a way to get around using soapy entirely?  I did notice that you can disable soapy support while building libxtrx...

sergforce, 2019-01-28:
If you're using our gr-osmosdr you don't need to disable Soapy. The actual gr-osmosdr backend can be selected using device arguments

Can you specify dev args for goth sink/source block
`xtrx,loglevel=5`
and copy the output?

b3sholt, 2019-01-28:
Hi Sergforce,
Looks like that works! (at least the xtrx argument...it wouldn't run with loglevel=5).  For trhe osmocom SInk, what do each of the gain values mean?  and what is their acceptable range?

sergforce, 2019-01-28:
For sink, only RF gain is valid with range [-31;0]

b3sholt, 2019-01-28:
Awesome!  And how about on the source?