xtrx-sdr/images #8: XTRX Device Not Detected

mathlizee, 2018-10-19:
Hi,

We have multiple issues with our XTRX Device.

First, even after compiling the host libraries and after completing the steps described in the Git repository ([https://github.com/xtrx-sdr/images](url)), the device is not recognised by our computer with a USB 3.0 connection. It is not visible when executing `lsusb`. Also, when we execute `./test_xtrx -t -l2`, we get the following error messages:

`15:22:10.566497 ERROR:   USB3380: Unable to open USB3380!
15:22:10.566627 ERROR:   Unable to allocate USB3380 context: error: -6
15:22:10.566643 ERROR:   Can't open device `/dev/xtrx0`: No such file or directory
Failed xtrx_open: -19`

When the device is plugged in via a USB 3.0 cable, a green LED is flashing with a 4 sec ON/ 1sec OFF pattern.

Then, we tried to connect the XTRX Device to a computer in a PCIe slot by using a miniPCIe to PCIe adapter. After installing the device in the PCIe slot, the computer would start, but nothing appeared on the screen. Is it possible that the XTRX could interfere with the GPU of the computer? 

We changed the XTRX for a WiFi Network miniPCIe and every was working correctly. It is therefore unlikely that it was the adapter's fault. Still, do you have any suggestion as to what miniPCIe to PCIe adapter we should use?

In the meantime, I would really like to get your support so that we can use our XTRX Device with a USB 3.0 cable. In this regard, do you have any idea of what could be wrong, or a suggestion to try on our side?

Thank you!
Mathieu Lizee

sergforce, 2018-10-19:
Hi,

Are you using our USB3 adapter or just using USB2.0 cable connected directly to XTRX (USB micro connector)?

Using miniPCIe to PCIe should be fine, that might be an issue if it's a very specific adapter. Can you provide the exact specification of your hardware (+ photos would be great):
Motherboard, CPU, and which slot was used for XTRX and which was occupied.

For further support send all data to xtrx@fairwaves.co with this bug reference.

Best Regards,
Sergey

mathlizee, 2018-10-23:
Hi,

We were finally able to use the XTRX with a miniPCIe to PCIe adapter. I am therefore closing the incident.

Thanks, 
Mathieu

closed 2018-10-23 by mathlizee