xtrx-sdr/images #10: make error

hmne, 2018-10-20:
[ 85%] Generating lms7002m_defs.h
+ python /root/images/sources/liblms7002m/regs/enum_parser.py /root/images/sources/liblms7002m/regs/afe.json /root/images/sources/liblms7002m/regs/bias.json /root/images/sources/liblms7002m/regs/cgen.json /root/images/sources/liblms7002m/regs/dc.json /root/images/sources/liblms7002m/regs/defaults.json /root/images/sources/liblms7002m/regs/endir.json /root/images/sources/liblms7002m/regs/ldo.json /root/images/sources/liblms7002m/regs/lml.json /root/images/sources/liblms7002m/regs/rbb.json /root/images/sources/liblms7002m/regs/rfe.json /root/images/sources/liblms7002m/regs/rxnco.json /root/images/sources/liblms7002m/regs/rxtsp.json /root/images/sources/liblms7002m/regs/sxx.json /root/images/sources/liblms7002m/regs/tbb.json /root/images/sources/liblms7002m/regs/trf.json /root/images/sources/liblms7002m/regs/txnco.json /root/images/sources/liblms7002m/regs/txtsp.json /root/images/sources/liblms7002m/regs/xbuf.json
Traceback (most recent call last):
  File "/root/images/sources/liblms7002m/regs/enum_parser.py", line 10, in <module>
    from Cheetah.Template import Template
ImportError: No module named Cheetah.Template
make[2]: *** [liblms7002m/CMakeFiles/lms7compact.dir/build.make:79: liblms7002m/lms7002m_defs.h] Error 1
make[2]: *** Deleting file 'liblms7002m/lms7002m_defs.h'
make[1]: *** [CMakeFiles/Makefile2:658: liblms7002m/CMakeFiles/lms7compact.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

sergforce, 2018-10-20:
Install `python-cheetah` package

sergforce, 2018-10-23:
CMake check was added and installation instruction was corrected

closed 2018-10-23 by sergforce