xtrx-sdr/images #11: xtrx: module verification failed: signature and/or required key missing - tainting kernel

ducnguyenAstro, 2018-10-22:
Hi guys,

That was so excited to try out the XTRX.
Having followed successfully the instructions until loading driver with the command: "modprobe xtrx"

Nothing shows up on /proc/modules. Checking out the dmesg there was an error: "xtrx: module verification failed: signature and/or required key missing - tainting kernel"

And there is no xtrx0 on /dev/ location and not with lspci either.

Thank you a lot for great help.

"

sergforce, 2018-10-22:
It seems you're using secure boot (UEFI) enabled kernel. That's why you get this message; however, this shouldn't affect you.

The device should appear in lspci regardless UEFI settings or driver presence. 

What system are you using?  Motherboard, CPU, slot where your XTRX is installed. This information will help to trigger the problem. You may send it xtrx@fairwaves.co

ducnguyenAstro, 2018-10-22:
Thanks Sergey, (forgive me if i call it wrong)

Output of ~$ lspci does not include the XTRX as below:
~$ lspci
00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:14.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB xHCI Host Controller (rev 04)
00:16.0 Communication controller: Intel Corporation 7 Series/C216 Chipset Family MEI Controller #1 (rev 04)
00:16.3 Serial controller: Intel Corporation 7 Series/C210 Series Chipset Family KT Controller (rev 04)
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C216 Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C216 Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.1 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 2 (rev c4)
00:1c.2 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C216 Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation QM77 Express Chipset LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C216 Chipset Family SMBus Controller (rev 04)
02:00.0 System peripheral: Ricoh Co Ltd MMC/SD Host Controller (rev 07)
03:00.0 Network controller: Intel Corporation Centrino Advanced-N 6205 [Taylor Peak] (rev 34)

XTRX is install on WMAX/WWAN pcie slot next to Wifi adapter on Thinkpad X230 running Ubuntu18.04.01.

Cheers,

sergforce, 2018-10-22:
AFAIK X230 has two miniPCIe slots and only one has PCIe lanes. Can you install XTRX instead of Wifi adapter and check?

sergforce, 2018-10-23:
I see. It's a significant problem with some laptops called "white list" where BIOS doesn't allow unlisted VID/PID combinations.

None of our team have X230 so we can't help, but I suggest to reach specific forums like this to see if there's solution:
https://www.bios-mods.com/forum/Thread-REQUEST-Lenovo-Thinkpad-X230-i-G2ETxxWW-Whitelist-Removal
https://www.thinkwiki.org/wiki/Custom_BIOS
https://forum.thinkpads.com/viewtopic.php?t=124541

I think USB3 is the safest option.

I close this bug since it's not related to the software. Feel free to reopen if you get it unlocked.

closed 2018-10-23 by sergforce