xtrx-sdr/images #12: Not able to get RX signal

ynour, 2018-10-25:
Hi,

I just received my xtrx and have been able to install xtrx software from source successfully, "test_xtrx -t -l2" output also seems ok.

Then I compiled xtrx version of gr-osmosdr. using osmocom_fft to display spectrum, as start I know there is FM channel on 106Mhz, using the following command:

./osmocom_fft -a "xtrx=0" -s 4000000 -f 106000000

I can't see any strong signal, just noise tried different frequencies like 433M, 900M and 2.4Ghz, still no signal.

So I think I'm making something wrong, or device argument above is wrong, or something is missing?

[image]

[image]

I'm using Lenovo Thinkpad T410 (with patched BIOS to overcome lenovo whitelisting issue), tried with laptop broadband antenna and external antenna connected to RXA, the same result.

Thanks

sergforce, 2018-10-25:
I see LNA & TIA gains are zero. Have you tried to set them to the max?

ynour, 2018-10-25:
Yes I tried set them to maximum, still not see a good signal on any band. Is there a way to use RXB instead of RXA to see if there is a difference? 

Also tried it with a desktop PC using a generic mPCIe to PCIe adapter card and externel SMA antenna, see the same issue.

sergforce, 2018-10-25:
Ok, I see. Try to change frequency after the GUI is started. And do the same with LNAs.

mgw93, 2018-12-04:
I seem to encounter the same issue.
TX seems to work. Although I still need to figure out how to switch antennas, and I only get a proper signal if I change frequency again after starting the program.
Receiving however does not give me any useful signal at all. Not even the noise level changes when I change from a 50Ω resistor to an antenna on both RX ports.
A signal transmitted with a USRP next to it does not show up either.
I used gqrx, and played with all the gain settings, as well as changing the frequency after starting the reception.
Is there any documentation on what each of the different gain settings "LNA", "TIA" and "PGA" does?
My xtrx came with one of the cables in the USB case somewhat damaged. But I don’t think that explains these symptoms.

Given the lack of a user guide, it is entirely possible, I’m just not setting the right parameters. But I think I tried all of the obvious options.

RobertK66, 2019-01-14:
@mgw93:  Have you tried to put some string into the Antenna field of the  gr-osmosdr block?
I started with using "B1" and got Rx on one of the antennas. When I tried to switch to other, i found that leaving the field empty does not give any RX.

Is there some documentation anywhere on the 'hallo-sdr settings' possible/needed via this gr block?