xtrx-sdr/images #13: Strange compilation error

kurtulmehtap, 2018-10-25:
I tried the necessary steps for compilation and  got a compilation error:
**fatal error: xtrx_api.h: No such file or directory**

Here are the details:
hp@hp:~$ git clone https://github.com/xtrx-sdr/images.git
Cloning into 'images'...
remote: Enumerating objects: 44, done.
remote: Counting objects: 100% (44/44), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 138 (delta 14), reused 31 (delta 7), pack-reused 94
Receiving objects: 100% (138/138), 4.51 MiB | 441.00 KiB/s, done.
Resolving deltas: 100% (55/55), done.
Checking connectivity... done.
hp@hp:~$ git submodule init
fatal: Not a git repository (or any of the parent directories): .git
hp@hp:~$ cd images/
hp@hp:~/images$ git submodule init
Submodule 'sources/LMS7002M-driver' (https://github.com/xtrx-sdr/LMS7002M-driver) registered for path 'sources/LMS7002M-driver'
Submodule 'sources/liblms7002m' (https://github.com/xtrx-sdr/liblms7002m.git) registered for path 'sources/liblms7002m'
Submodule 'sources/libusb3380' (https://github.com/xtrx-sdr/libusb3380) registered for path 'sources/libusb3380'
Submodule 'sources/libxtrx' (https://github.com/xtrx-sdr/libxtrx) registered for path 'sources/libxtrx'
Submodule 'sources/libxtrxdsp' (https://github.com/xtrx-sdr/libxtrxdsp) registered for path 'sources/libxtrxdsp'
Submodule 'sources/libxtrxll' (https://github.com/xtrx-sdr/libxtrxll) registered for path 'sources/libxtrxll'
Submodule 'sources/xtrx_linux_pcie_drv' (https://github.com/xtrx-sdr/xtrx_linux_pcie_drv) registered for path 'sources/xtrx_linux_pcie_drv'
hp@hp:~/images$
hp@hp:~/images$ git submodule update
Cloning into 'sources/LMS7002M-driver'...
remote: Enumerating objects: 2412, done.
remote: Total 2412 (delta 0), reused 0 (delta 0), pack-reused 2412
Receiving objects: 100% (2412/2412), 18.15 MiB | 4.66 MiB/s, done.
Resolving deltas: 100% (1403/1403), done.
Checking connectivity... done.
Submodule path 'sources/LMS7002M-driver': checked out 'c244dfb92f84c027ede5faee7527a5947fc48480'
Cloning into 'sources/liblms7002m'...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 38 (delta 14), reused 31 (delta 11), pack-reused 0
Unpacking objects: 100% (38/38), done.
Checking connectivity... done.
Submodule path 'sources/liblms7002m': checked out '07baade5df34946cbdd0c26753844a4de9e82ac0'
Cloning into 'sources/libusb3380'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 42 (delta 2), reused 4 (delta 2), pack-reused 33
Unpacking objects: 100% (42/42), done.
Checking connectivity... done.
Submodule path 'sources/libusb3380': checked out '9e555415e438d88a9c98087b946642b00b5f9edf'
Cloning into 'sources/libxtrx'...
remote: Enumerating objects: 44, done.
remote: Counting objects: 100% (44/44), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 198 (delta 19), reused 29 (delta 12), pack-reused 154
Receiving objects: 100% (198/198), 171.34 KiB | 0 bytes/s, done.
Resolving deltas: 100% (109/109), done.
Checking connectivity... done.
Submodule path 'sources/libxtrx': checked out '54284cd3780d0ee3cbd5e5a2e103b2b08f032910'
Cloning into 'sources/libxtrxdsp'...
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 50 (delta 0), reused 1 (delta 0), pack-reused 47
Unpacking objects: 100% (50/50), done.
Checking connectivity... done.
Submodule path 'sources/libxtrxdsp': checked out '84c6f4c8bdb9e26e7c9a62bd96f02e35d61aaf45'
Cloning into 'sources/libxtrxll'...
remote: Enumerating objects: 31, done.
remote: Counting objects: 100% (31/31), done.
remote: Compressing objects: 100% (21/21), done.
[... 142 lines cut ...]
[ 87%] Built target xtrx
Scanning dependencies of target test_xtrx
[ 88%] Building C object libxtrx/CMakeFiles/test_xtrx.dir/test_xtrx.c.o
[ 90%] Linking C executable test_xtrx
[ 90%] Built target test_xtrx
Scanning dependencies of target mainwindow_automoc
[ 91%] Automatic moc, uic and rcc for target mainwindow
Generating ui_mainwindow.h
Generating ui_mainwindow.h
Generating moc_mainwindow.cpp
Generating moc_rxthread.cpp
Generating moc_waterfall.cpp
[ 91%] Built target mainwindow_automoc
Scanning dependencies of target mainwindow
[ 92%] Building CXX object libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/mainwindow.cpp.o
In file included from /home/hp/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:20:0:
**/home/hp/images/sources/libxtrx/examples/xtrx_fft/mainwindow.h:25:22: fatal error: xtrx_api.h: No such file or directory**
compilation terminated.
libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/build.make:62: recipe for target 'libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/mainwindow.cpp.o' failed
make[2]: *** [libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/mainwindow.cpp.o] Error 1
CMakeFiles/Makefile2:830: recipe for target 'libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/all' failed
make[1]: *** [libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
hp@hp:~/images/sources/build$

ducnguyenAstro, 2018-10-25:
Hi sergforce,

Come across the same problem. could you tell me how to apply the above patch?

Thanks

ducnguyenAstro, 2018-10-25:
Just follow-up,

After changing CMakeLists.txt does not solve problem.
go to : /images/sources
manually  copy file: cp libxtrx/xtrx_api.h libxtrx/examples/xtrx_fft/
working properly.

Thanks,

sergforce, 2018-10-25:
Thanks all!
There was one more miss, LIBXTRX_INCLUDE_DIRS wasn't set properly.

I committed and pushed already.

closed 2018-10-25 by sergforce