xtrx-sdr/images #16: TX: high DC signal and transmitting in bursts

ducnguyenAstro, 2018-10-26:
Hi guys,

There are two things that i found when recently working with XTRX:
 -DC signal is always there (~ 12dbm) upon powering XTRX up.
- Transmitting signals always in bursts with error on terminal: 
"11:23:44.092663 ERROR:   xtrx_send_burst_sync[0]: TX DMA Current skip due to TO buffers: 2797, prev: 2796 TS:36667227 STPS:36675414"

Signals are all observed on Signal Analyzer.

Any comments or though are really appropriated to make this board more beautiful.
Cheers,

sergforce, 2018-10-26:
I see you're using the old version. `test_xtrx` utility was updated, see the last comments about TO events here https://github.com/xtrx-sdr/images/issues/6

ducnguyenAstro, 2018-10-26:
Thanks sergforce,

I will close this issue and move the conversation into the other. Sorry for not realizing they are both the same.

closed 2018-10-26 by ducnguyenAstro