xtrx-sdr/images #17: Build fails with QCustomPlot installed

darkstar007, 2018-10-28:
Hi,

I have QCustomPlot 2.0.1 installed (on Debian unstable) - which appears to be the current/latest version. This is detected when cmake runs. Upon building I get the following errors:

```
[ 86%] Building CXX object libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/mainwindow.cpp.o
/home/matt/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp: In member function ‘void MainWindow::redraw(int)’:
/home/matt/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:111:34: error: ‘rpImmediate’ is not a member of ‘QCustomPlot’
  customPlot->replot(QCustomPlot::rpImmediate);
                                  ^~~~~~~~~~~
/home/matt/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp: In member function ‘void MainWindow::on_btStartStop_clicked()’:
/home/matt/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:144:25: error: ‘class QCPGraph’ has no member named ‘clearData’; did you mean ‘userData’?
   ui->widget->graph(0)->clearData();
                         ^~~~~~~~~
                         userData
/home/matt/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:145:25: error: ‘class QCPGraph’ has no member named ‘clearData’; did you mean ‘userData’?
   ui->widget->graph(1)->clearData();
                         ^~~~~~~~~
                         userData
make[2]: *** [libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/build.make:63: libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/mainwindow.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:839: libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

If I remove the QCustomplot package, then cmake now grumbles about not being able to find it, but everything else can at least now build.

What version of QCustomPlot do you require?

Thanks,

Matt

sergforce, 2018-10-28:
Hi Matt,

It seems when you removed QCustomPlot the CMake used cached values in CMakeCache.txt. Try to remove CMakeCache.txt files (or build directory completely) and run again. If QCustomPlot isn't installed the build should go without it.

The code was written for 1.x QCustomPlot. I see there are few changes in API for 2.0 so it should be easy to port

darkstar007, 2018-10-28:
Yeah, I delete the whole build dir to try to keep things sane.....

OK - 3 errors going from V1 to V2 doesn't seem that bad..........

zhovner, 2018-12-08:
I get the same error on Kali Linux 2019.1
Here is the `QCustomPlot` sections from `CMakeCache.txt`
```
//The directory containing a CMake configuration file for QCustomPlot.
QCustomPlot_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/QCustomPlot
QCustomPlot_INCLUDE_DIR:PATH=/usr/include
QCustomPlot_LIBRARIES:FILEPATH=/usr/lib/x86_64-linux-gnu/libqcustomplot.so
//Details about finding QCustomPlot
FIND_PACKAGE_MESSAGE_DETAILS_QCustomPlot:INTERNAL=[/usr/include][/usr/lib/x86_64-linux-gnu/libqcustomplot.so][v()]
```
The only solution is to use this patch? https://github.com/xtrx-sdr/images/issues/20#issuecomment-444054600

sergforce, 2018-12-27:
I updated the code to support QCustomPlot 2.0.1, so systems using QCP 2.0.1 should work just fine. Unfortunately, QCustomPlot 2.0.0 doesn't provide a way to detect major version change (2.0.1 does), so 2.0.0 will fail. However, most Linux distros are already using 2.0.1 and this should not be a problem. 

Feel free to reopen if you hit any other problems with QCP

closed 2018-12-27 by sergforce

jgalak, 2023-12-31:
I am having a similar issue on Debian 12 (Bookworm) on Raspberry Pi.  The problem here is that I can't install QCustomPlot 2.0.x, only 2.1.x.  Specifically, 2.1.0-dfsb1-3.1.  Trying to build with that generates the following error:

```
[ 88%] Linking CXX executable xtrx_fft
/usr/bin/ld: CMakeFiles/xtrx_fft.dir/mainwindow.cpp.o: in function `MainWindow::MainWindow(QWidget*)':
/home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:51: undefined reference to `QCustomPlot::addGraph(QCPAxis*, QCPAxis*)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:52: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:52: undefined reference to `QCPAbstractPlottable::setPen(QPen const&)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:56: undefined reference to `QCustomPlot::addGraph(QCPAxis*, QCPAxis*)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:57: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:57: undefined reference to `QCPAbstractPlottable::setPen(QPen const&)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:63: undefined reference to `QCPLayerable::setVisible(bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:64: undefined reference to `QCPAxis::setTickLabels(bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:65: undefined reference to `QCPLayerable::setVisible(bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:66: undefined reference to `QCPAxis::setTickLabels(bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:69: undefined reference to `QCPAxis::setRange(double, double)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:70: undefined reference to `QCPAxis::setRange(double, double)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:72: undefined reference to `QCustomPlot::setInteractions(QFlags<QCP::Interaction> const&)'
/usr/bin/ld: CMakeFiles/xtrx_fft.dir/mainwindow.cpp.o: in function `MainWindow::redraw(int)':
/home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:113: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:113: undefined reference to `QCPGraph::setData(QVector<double> const&, QVector<double> const&, bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:116: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:116: undefined reference to `QCPGraph::setData(QVector<double> const&, QVector<double> const&, bool)'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:121: undefined reference to `QCustomPlot::replot(QCustomPlot::RefreshPriority)'
/usr/bin/ld: CMakeFiles/xtrx_fft.dir/mainwindow.cpp.o: in function `MainWindow::on_rescale_clicked()':
/home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:131: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:131: undefined reference to `QCPAbstractPlottable::rescaleAxes(bool) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:133: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:133: undefined reference to `QCPAbstractPlottable::rescaleAxes(bool) const'
/usr/bin/ld: CMakeFiles/xtrx_fft.dir/mainwindow.cpp.o: in function `MainWindow::on_btStartStop_clicked()':
/home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:159: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: /home/jg/images/sources/libxtrx/examples/xtrx_fft/mainwindow.cpp:160: undefined reference to `QCustomPlot::graph(int) const'
/usr/bin/ld: CMakeFiles/xtrx_fft.dir/mainwindow.cpp.o: in function `Ui_MainWindow::setupUi(QMainWindow*)':
/home/jg/images/sources/build/libxtrx/examples/xtrx_fft/xtrx_fft_autogen/include/ui_mainwindow.h:241: undefined reference to `QCustomPlot::QCustomPlot(QWidget*)'
collect2: error: ld returned 1 exit status
make[2]: *** [libxtrx/examples/xtrx_fft/CMakeFiles/xtrx_fft.dir/build.make:173: libxtrx/examples/xtrx_fft/xtrx_fft] Error 1
make[1]: *** [CMakeFiles/Makefile2:666: libxtrx/examples/xtrx_fft/CMakeFiles/xtrx_fft.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

Any way to make this work with 2.1?