xtrx-sdr/images #18: cmake complains about missing python-cheetah

FlashPackets, 2018-10-28:
flash@Flash64Super:~/cyb/sdr/xtrx/images$  sudo apt-get install build-essential libusb-1.0-0-dev cmake dkms python-cheetah
[sudo] password for flash: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
python-cheetah is already the newest version (2.4.4-3.fakesyncbuild1).  <<<<<<<<<<<<<<<<<<<<<<
cmake is already the newest version (3.5.1-1ubuntu3).
dkms is already the newest version (2.2.0.3-2ubuntu11.5).
0 upgraded, 0 newly installed, 0 to remove and 39 not upgraded.

flash@Flash64Super:~/cyb/sdr/xtrx/images/sources/build$ cmake ..
-- Building for x86_64
-- Building for x86_64
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Building for x86_64
-- xtrxll: PCIe support enabled
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- xtrxll: USB3380 support enabled
-- Building for x86_64
-- Building for x86_64
CMake Error at liblms7002m/CMakeLists.txt:47 (message):
  Python module Cheetah is reuired! Install 'python-cheetah' package.  <<<<<<<<<<< THE ERROR

Running "cheetah test " from the command line yielded this result
flash@Flash64Super:~/cyb/sdr/xtrx/images/sources$ cheetah test
................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................FFF.......................FFF.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Compiling /tmp/tmp8yC4ym.tmpl -> /tmp/tmp8yC4ym.py
..Compiling /tmp/tmpUx3CLO.tmpl -> /tmp/tmpUx3CLO.py
.Compiling /tmp/tmpXegqjY.tmpl -> /tmp/tmpXegqjY.py
.Compiling /tmp/tmpzTZrlW.tmpl -> /tmp/tmpzTZrlW.py
.Compiling /tmp/tmpHIUc6r.tmpl -> /tmp/tmpHIUc6r.py
....................................................
======================================================================
FAIL: test10 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_MacEOL)
func placeholder - with ('''\nstring\n''')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 845, in test10
    "\naoeu\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
''')(end)'

    Expected Output =
(end)

    Actual Output =

aoeu
(end)

======================================================================
FAIL: test11 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_MacEOL)
func placeholder - with ('''\nstring'\n''')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 850, in test11
    "\naoeu'\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
''')(end)'

    Expected Output =
(end)

    Actual Output =

aoeu'
(end)

======================================================================
FAIL: test12 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_MacEOL)
func placeholder - with ("""\nstring\n""")
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 855, in test12
    "\naoeu\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
""")(end)"

    Expected Output =
(end)

    Actual Output =

aoeu
(end)

======================================================================
FAIL: test10 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_Win32EOL)
func placeholder - with ('''\nstring\n''')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 845, in test10
    "\naoeu\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
$aFunc('''
aoeu
''')(end)

    Expected Output =

aoeu
(end)

    Actual Output =

aoeu
(end)

======================================================================
FAIL: test11 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_Win32EOL)
func placeholder - with ('''\nstring'\n''')
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 850, in test11
    "\naoeu'\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
$aFunc('''
aoeu'
''')(end)

    Expected Output =

aoeu'
(end)

    Actual Output =

aoeu'
(end)

======================================================================
FAIL: test12 (Cheetah.Tests.SyntaxAndOutput.Placeholders_Calls_Win32EOL)
func placeholder - with ("""\nstring\n""")
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 855, in test12
    "\naoeu\n")
  File "/usr/local/lib/python2.7/dist-packages/Cheetah-2.4.0-py2.7-linux-x86_64.egg/Cheetah/Tests/SyntaxAndOutput.py", line 177, in verify
    assert output==expectedOutput, self._outputMismatchReport(output, expectedOutput)
AssertionError: 
Template output mismatch: 

    Input Template =
$aFunc("""
aoeu
""")(end)

    Expected Output =

aoeu
(end)

    Actual Output =

aoeu
(end)

----------------------------------------------------------------------
Ran 2103 tests in 11.570s

FAILED (failures=6)

FlashPackets, 2018-10-28:
Trying the patch produced 

s$ diff --git a/CMakeLists.txt b/CMakeLists.txt               << Ther is only one CMakeLists.txt in ../sources
diff: unrecognized option '--git'                                  
diff: Try 'diff --help' for more information.               

On the other hand using anaconda and switching to python 2.7 yielded 

$ source activate py27

xtrx/images/sources/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for x86_64
-- Building for x86_64
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb-1.0.so
-- Building for x86_64
-- xtrxll: PCIe support enabled
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- xtrxll: USB3380 support enabled
-- Building for x86_64
-- Building for x86_64
-- Found PythonInterp: /home/flash/anaconda3/envs/py27/bin/python (found version "2.7.14") 
CMake Error at liblms7002m/CMakeLists.txt:47 (message):
  Python module Cheetah is reuired! Install 'python-cheetah' package.

So, I guess it can assert  that python 2.7 is  called

sergforce, 2018-10-28:
Can you run `python -c "from Cheetah.Template import Template" ; echo $?` on your system?

FlashPackets, 2018-10-28:
~/cyb/sdr/xtrx/images/sources$ python -c "from Cheetah.Template import Template" ; echo $?
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named Cheetah.Template

FlashPackets, 2018-10-28:
returned 0                        
 
I'm surprised that setting anaconda to 2.7 is not sufficient.

sergforce, 2018-11-06:
Were you able to compile it?
You can specify python explicitly, e.g. 
`cmake .. -DPYTHONINTERP_FOUND=True -DPYTHON_EXECUTABLE=/usr/bin/python2.7`

jrri, 2018-11-20:
I'm also using an environment with python 3.6.4 and was able to go through the installation by adding Cheetah3 to the Anaconda env:

`
$ pip3 install cheetah3
`

Issueing "conda list" the following line shows Cheetah3 in the Anaconda env:

`
...
Cheetah3                  3.1.0                     <pip>
..
`

Note that there are differences between python 2 and python 3 and I needed to slightly modify the images/sources/liblms7002m/regs/enum_parser.py:

```diff
--- a/xtrx-sdr/images/sources/liblms7002m/regs/enum_parser.py
+++ b/xtrx-sdr/images/sources/liblms7002m/regs/enum_parser.py
@@ -8,6 +8,7 @@
 import re
 import json
 from Cheetah.Template import Template
+from functools import reduce
 
 def get_name(reg, field_name, category):
     #TODO optional reg[name]
@@ -30,7 +31,7 @@
 def get_options(reg, field_name):
     field = reg['fields'][field_name]
     if 'options' in field:
-        return [(('%s_%s'%(get_name(reg, field_name, "OPT"), k)).upper(), v) for k,v in sorted(field['options'].iteritems(), key=lambda x:x[1])]
+        return [(('%s_%s'%(get_name(reg, field_name, "OPT"), k)).upper(), v) for k,v in sorted(field['options'].items(), key=lambda x:x[1])]
     return None
 
 def get_shift_mask(reg, field_name):
@@ -45,7 +46,7 @@
     return int(bits), "0x1"
 
 def sorted_field_keys(reg):
-    return reversed(sorted(reg['fields'].keys(), key=lambda x: map(int, reg['fields'][x]['bits'].split(':'))[0]))
+    return reversed(sorted(reg['fields'].keys(), key=lambda x: list(map(int, reg['fields'][x]['bits'].split(':')))[0]))
 
 def sorted_field_keys_str(reg):
     return reduce(lambda x, y: str(x) + ", " + str(y), sorted_field_keys(reg))
@@ -121,7 +122,7 @@
 """
 
 if __name__ == '__main__':
-    print HEAD
+    print(HEAD)
 
     regs = list()
     for arg in sys.argv[1:]:
@@ -145,7 +146,7 @@
                 sorted_field_keys=sorted_field_keys,
                 sorted_field_keys_str=sorted_field_keys_str,
             )))
-            print code
+            print(code)
 
 
 
@@ -164,4 +165,4 @@
         sorted_field_keys_str=sorted_field_keys_str,
     )))
 
-    print code
+    print(code)

```

FlashPackets, 2018-12-16:
Setting up a virtualenv proved to be very effective:
from the xtrx directory I ran
virtualenv -p /usr/bin/python2.7 images
source images/bin/activate

sergforce, 2018-12-27:
I made an explicit dependency to Python 2.7 since some distros (like Ubuntu 18.04) don't provide packet Cheetah for python3.

In Ubuntu 18.04 you need to install python and python-cheetah
`sudo apt install python python-cheetah`

sergforce, 2019-01-13:
If someone still has the problem, please reopen.

closed 2019-01-13 by sergforce