xtrx-sdr/images #20: Build on Fedora 29 issues

dpreed, 2018-11-29:
I am trying to build and test on Fedora 29. There are some differences that cause problems.

1) `gcc -print-multiarch` on Fedora produces the null string. That's because "multiarch" is debian-only. I changed all the tests to use the `arch` command to fix the problems caused in the CMakeLists.txt files in the sources. `uname -m` would also work, as it prints the same result as `arch`.

2) `QCustomPlot` is packaged differently on Fedora. In particular, there appears to be no ``FindQCustomPlot.cmake` file. This causes `cmake ..` to complain and not build the fft example. I have patched around that by creating a FindQCustomPlot.cmake file that sets QCustomPlot_FOUND to true. I will be suggesting to the Fedora package maintainer to add the cmake file necessary to the -devel package.

3) I have found that I encounter Issue #17 due to the change in version of QCustomPlot.
Will probably try to guess what the fix is.

I hope this is helpful. I am looking forward to testing my new XTRX CS over USB 3 shortly.

drakedevel, 2018-12-04:
@dpreed I was able to work around point 3 with the following patch applied to sources/libxtrx:

```patch
diff --git a/examples/xtrx_fft/CMakeLists.txt b/examples/xtrx_fft/CMakeLists.txt
index 6f5ee74..0400a72 100644
--- a/examples/xtrx_fft/CMakeLists.txt
+++ b/examples/xtrx_fft/CMakeLists.txt
@@ -12,7 +12,7 @@ message(${CMAKE_MODULE_PATH})
 # Find the QtWidgets library
 find_package(Qt5Widgets CONFIG REQUIRED)
 find_package(Qt5PrintSupport CONFIG REQUIRED)
-find_package(QCustomPlot CONFIG REQUIRED)
+find_package(QCustomPlot REQUIRED)
 find_package(LibXTRX REQUIRED)
 
 # Populate a CMake variable with the sources
diff --git a/examples/xtrx_fft/FindQCustomPlot.cmake b/examples/xtrx_fft/FindQCustomPlot.cmake
new file mode 100644
index 0000000..ffb0bae
--- /dev/null
+++ b/examples/xtrx_fft/FindQCustomPlot.cmake
@@ -0,0 +1,19 @@
+# Try to find the QCustomPlot librairies
+#  QCustomPlot_FOUND - system has QCustomPlot lib
+#  QCustomPlot_INCLUDE_DIR - the GMP include directory
+#  QCustomPlot_LIBRARIES - Libraries needed to use QCustomPlot
+
+# Copyright (c) 2013, Anton Gladky <gladk@debian.org>
+#
+# Redistribution and use is allowed according to the terms of the GPL-3 license.
+
+
+IF (QCustomPlot_INCLUDE_DIR AND QCustomPlot_LIBRARIES)
+  SET(QCustomPlot_FIND_QUIETLY TRUE)
+ENDIF (QCustomPlot_INCLUDE_DIR AND QCustomPlot_LIBRARIES)
+
+FIND_PATH(QCustomPlot_INCLUDE_DIR NAMES qcustomplot.h )
+FIND_LIBRARY(QCustomPlot_LIBRARIES NAMES qcustomplot-qt5 )
+
+include(FindPackageHandleStandardArgs)
+FIND_PACKAGE_HANDLE_STANDARD_ARGS(QCustomPlot DEFAULT_MSG QCustomPlot_INCLUDE_DIR QCustomPlot_LIBRARIES)
diff --git a/examples/xtrx_fft/main.cpp b/examples/xtrx_fft/main.cpp
index 14582d2..6be7ed8 100644
--- a/examples/xtrx_fft/main.cpp
+++ b/examples/xtrx_fft/main.cpp
@@ -23,8 +23,9 @@
 
 #include "mainwindow.h"
 
-void qInitResources_application()
+int qInitResources_application()
 {
+    return 0;
 }
 
 int main(int argc, char *argv[])
diff --git a/examples/xtrx_fft/mainwindow.cpp b/examples/xtrx_fft/mainwindow.cpp
index d1bfe40..3cff873 100644
--- a/examples/xtrx_fft/mainwindow.cpp
+++ b/examples/xtrx_fft/mainwindow.cpp
@@ -108,7 +108,7 @@ void MainWindow::redraw(int idx)
 		customPlot->graph(1)->setData(x, *pz[idx]);
 	}
 
-	customPlot->replot(QCustomPlot::rpImmediate);
+	customPlot->replot(QCustomPlot::rpImmediateRefresh);
 }
 
 
@@ -141,8 +141,8 @@ void MainWindow::on_btStartStop_clicked()
 	QString devstr = ui->cbDev->currentText();
 
 	if (dev == NULL) {
-		ui->widget->graph(0)->clearData();
-		ui->widget->graph(1)->clearData();
+		ui->widget->graph(0)->data()->clear();
+		ui->widget->graph(1)->data()->clear();
 
 		ui->statusbar->showMessage(QString("Samplerate %1 MSPS Gain: %2").arg(samplerate).arg(gain));
 		res = xtrx_open(devstr.toLatin1(), 4, &dev);
```

Note that you'll need `qcustomplot-qt5-devel`, if you link against the Qt4 `qcustomplot-devel` you'll crash and burn on startup since the rest of the app is Qt5.

With this and `-DFORCE_ARCH=x86_64` everything builds and runs for me.

When I run the `xtrx_fft` command, it seems as though my X-axis is bogus (always 0-500 regardless of what I set the frequency to). I'm not sure if that's introduced by qcustomplot 2.x or if it's otherwise broken.

Separately, I can't seem to get any sort of valid data from my USB3 XTRX in any of GNU Radio, SDRangel, or this tool, and everything is very crashy. Not sure if F29 specific or if there's something I'm missing, but figured you might find the data point helpful in case you have a similar experience.

sergforce, 2018-12-27:
Try to update all libs, Fedora 29 should work fine.

closed 2018-12-27 by sergforce