xtrx-sdr/images #21: USB3 adapter drivers for Windows10

jkassot, 2018-12-15:
USB3 adaptor/XTRX are not recognized by Win10. 
Windows 10 device manager requires a driver for the UDB3/XTRX  Combo.
I understood that the USB3 driver for Win10 is ready. Where can it be found as pre-combiled binary?

sergforce, 2019-01-14:
USB3380 implementation relies on cross-platform `libusb`. Libusb on Windows can use different kernel driver as a backend (https://github.com/libusb/libusb/wiki/Windows). I used libusbK and WinUSB.

Unfortunately, Microsoft has changed signing policy (https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-) and now both the kernel driver itself and INF file must be signed. We don't have signed INF for USB3380 pointing to WinUSB (or libusbK) right now. Anyway, I pushed what I have now so you can do more tests. To install this driver you need to disable driver signature enforcement (google for that).

Windows build can be compiled using MinGW-w64, and the last build is uploaded here https://github.com/xtrx-sdr/images/tree/master/binaries/Windows

I did some tests and found it very slow on the latest Windows 10 system, used to have much more performance. Will look into this later

chemeris, 2019-02-02:
Btw, **if someone wants to take ownership of maintaining Windows builds and testing under Windows - please step up**.
Quite a few people would welcome a working Windows driver but we, unfortunately, don't have much time to work on it.

DG9BFC, 2019-02-02:
i also want a working windows solution ... i am not a programmer so can not help on that ... but could be a "betatester" or so ?!? ... i just installed the windiws drivers 
found the inf file and the usb3380driver ... downloading and installing was a bit tricky but at the end now the xtrx is found in windows device manager ... so far so good
(for what are the other drivers in the x64 folder??) ....
but the device is not found in sdrangel (i also asked the programmer of sdrangel for help)
this is tested on a win7 system but i also have a newer tablet with win10
in win 7 the system asked to install that "unsigned " driver (and i could install it when clicking on install anyway) .... is that different in win10??