xtrx-sdr/images #24: [3380] Unable to reset USB3380

ccsh23, 2018-12-18:
Hi all,

I have assembled XTRX+USB3 and now I am struggling with (hopefully) software problem on Xubuntu 14.04. I followed all the steps mentioned in README file from this repo, and I believe all components were installed successfully:

Here is looks like XTRX module has been installed correctly:
```
$ sudo dkms status
xtrx, 0.0.1-1, 4.4.0-21-generic, x86_64: installed
```
I could also seen that USB3 adatper was detected:
```
$ lsusb | grep 0525:3380
Bus 002 Device 003: ID 0525:3380 Netchip Technology, Inc. 
```

The problem begins when I try to test if XTRX is alive. When typing following command `test_xtrx -t -l2` it cannot find the text_xtrx binary, so I have to manually enter the directory which contains it. Then, when I execute test command I am getting following error and USB3 adapter disappears from lsusb list:
```
$ ./test_xtrx -t -l2
14:07:39.091318 WARN:   [3380] Need to reinitialize the USB device dueto configuration change
14:07:39.226266 ERROR:  [3380] Unable to reset USB3380: error `Entity not found` (-5)!
14:07:39.226464 ERROR:  [USB3] Unable to reinitialize context: error: -13
Failed xtrx_open: -19
```
I got two such XTRX+USB3 setups, and the same problems happen for the second one as well. I am connecting them to USB3 port on the front of my PC case. My motherboard is Z170-Gaming K3. 

Could you help me with that issue?

Peterthegreat, 2019-01-08:
Hello ccsh23
try the following:
- another USB3 port, if available
- install udev rules.
> cp sources/xtrx_linux_pcie_drv/50-xtrx.rules /etc/udev/rules.d/
- uninstall linux modemmanager

RobertK66, 2019-01-10:
Hi all,
'I have the same issue here. I am on Ubuntu 16.04 and downloaded the images today. Because of #25 I could not install the xtrx_linux_pcie_drv package yet. But as I want to use USB3 for now I continued and have now same result as described by  @ccsh23. Only difference is an additional warning line for the missing pcie driver:

robert@sdr-pad:~/Downloads$ /usr/lib/xtrx/test_xtrx -t -l2
22:26:54.490656 WARN:   [PCIE] XTRX PCIe driver isn't loaded
22:26:54.496889 ERROR:  [3380] Unable to open USB3380!
22:26:54.497232 ERROR:  [USB3] Unable to allocate USB3380 context: error: -6
Failed xtrx_open: -19

I tried all 3 USB3 Ports of my laptop -> no change
I uninstalled modemmanger -> device numbers in lsusb did change, but no change in output of test_xtrx

RobertK66, 2019-01-10:
yes, thx, I already tried this. Unfortunately this also didn't help. But for today I have to finish up now. Thx. a lot for your support. Maybe tomorrow is a better day for SDR Initiation 😄 ...

sergforce, 2019-01-11:
Let me clarify things

`sources/xtrx_linux_pcie_drv/50-xtrx.rules` is for PCIe driver only, it has nothing to do for USB. Also you don't need to install xtrx_linux_pcie_drv for USB operations.

Create `50--usb-xtrx.rules` in `/etc/udev/rules.d/`
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0525", ATTRS{idProduct}=="3380", GROUP="usb", MODE="0666"
```
after creating you need to unplug USB3 adapter and restart `udevd`
``` sh
sudo /etc/init.d/udev restart
```

After insertion, and the first run you may still see the error
```
14:07:39.091318 WARN:   [3380] Need to reinitialize the USB device dueto configuration change
```
but next runs should work just fine

RobertK66, 2019-01-11:
Ok, thx. for clarification ( I switched to a linux system because of this: https://github.com/xtrx-sdr/images/issues/4#issuecomment-427747256 😉 ).

I tried your recipe but unfortunately no success:

```
robert@sdr-pad:~$ cat /etc/udev/rules.d/50--usb-xtrx.rules 
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0525", ATTRS{idProduct}=="3380", GROUP="usb", MODE="0666"

robert@sdr-pad:~$ sudo /etc/init.d/udev restart
[ ok ] Restarting udev (via systemctl): udev.service.
robert@sdr-pad:~$ lsusb
Bus 002 Device 004: ID 8087:07da Intel Corp. 
Bus 002 Device 003: ID 04f2:b330 Chicony Electronics Co., Ltd Asus 720p CMOS webcam
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:0139 Realtek Semiconductor Corp. RTS5139 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 014: ID 0525:3380 Netchip Technology, Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0b95:7e2b ASIX Electronics Corp. AX88772B
Bus 003 Device 004: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx
Usage: DMATX and/or DMARX must be enabled (or TX REPEAT is set)!
09:38:19.867221 WARN:   [PCIE] XTRX PCIe driver isn't loaded
[00] usb3380://4/2/14                 4.2:14           USB              5Gbit           
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx -t -l7
09:38:30.901635 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
09:38:30.914733 ERROR:  usb3380_context_init_ex:1232 [3380] Unable to open USB3380!
09:38:30.914925 ERROR:  xtrxllusb3380v0_open:469 [USB3] Unable to allocate USB3380 context: error: -6
Failed xtrx_open: -19
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/14  -l7
09:39:38.313629 DEBUG:  xtrxllusb3380v0_open:452 [USB3] usb3380: comparing devices 4/2/14 <> 4/2/14
09:39:38.313699 ERROR:  usb3380_context_init_ex:1226 [3380] Unable to initialize DEVICE: -3
09:39:38.313707 ERROR:  xtrxllusb3380v0_open:469 [USB3] Unable to allocate USB3380 context: error: -3
Failed xtrx_open: -19
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx
Usage: DMATX and/or DMARX must be enabled (or TX REPEAT is set)!
09:39:51.671715 WARN:   [PCIE] XTRX PCIe driver isn't loaded
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx
Usage: DMATX and/or DMARX must be enabled (or TX REPEAT is set)!
09:40:01.959638 WARN:   [PCIE] XTRX PCIe driver isn't loaded
[00] usb3380://4/2/15                 4.2:15           USB              5Gbit           
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/15  -l7
09:40:07.975293 DEBUG:  xtrxllusb3380v0_open:452 [USB3] usb3380: comparing devices 4/2/15 <> 4/2/15
09:40:07.975370 ERROR:  usb3380_context_init_ex:1226 [3380] Unable to initialize DEVICE: -3
09:40:07.975383 ERROR:  xtrxllusb3380v0_open:469 [USB3] Unable to allocate USB3380 context: error: -3
Failed xtrx_open: -19
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx -t -d usb3380://4/2/15  -l7
Creating multidev for 1 devices
09:40:30.676702 DEBUG:  xtrxllusb3380v0_open:452 [USB3] usb3380: comparing devices 4/2/15 <> 4/2/15
09:40:30.676774 ERROR:  usb3380_context_init_ex:1226 [3380] Unable to initialize DEVICE: -3
09:40:30.676783 ERROR:  xtrxllusb3380v0_open:469 [USB3] Unable to allocate USB3380 context: error: -3
Failed xtrx_open: -19
robert@sdr-pad:~$ 

```

* When I plug out/in the device it gets a new device Id but the error never changes.

* The device has alll 3LEDs steady on.

* I also tried another USB3 device (hdd) and this got also connected to BUS 4 and it was mounted by the system and worked normal. So I asume the installation and drivers for the USB3 root hub are ok.

sergforce, 2019-01-11:
Error -3 means "Access denied (insufficient permissions) " which is strange

Does `sudo /usr/lib/xtrx/test_xtrx -t  -l7` works normally?

RobertK66, 2019-01-11:
wow, somehow that worked now. I am not sure in which sequence I did the connect / reconnect and sudo calls (got interrupted by other stuff) but finally I got some 'proof of life' from the SDR. 👍: 
```
robert@sdr-pad:~$ sudo /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/16
[sudo] password for robert: 
10:22:56.516583 WARN:   [3380] Need to reinitialize the USB device dueto configuration change
10:22:56.702372 ERROR:  [3380] Unable to reset USB3380: error `Entity not found` (-5)!
10:22:56.702633 ERROR:  [USB3] Unable to reinitialize context: error: -13
Failed xtrx_open: -19
robert@sdr-pad:~$ sudo /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/17
10:23:11.757681 ERROR:  [USB3] Unable to allocate USB3380 context: error: -19
Failed xtrx_open: -19
robert@sdr-pad:~$ 
robert@sdr-pad:~$ lsusb
Bus 002 Device 004: ID 8087:07da Intel Corp. 
Bus 002 Device 003: ID 04f2:b330 Chicony Electronics Co., Ltd Asus 720p CMOS webcam
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0bda:0139 Realtek Semiconductor Corp. RTS5139 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 018: ID 0525:3380 Netchip Technology, Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0b95:7e2b ASIX Electronics Corp. AX88772B
Bus 003 Device 004: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
robert@sdr-pad:~$ sudo /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/18
[sudo] password for robert: 
12:11:36.684421 WARN:   [3380] Need to reinitialize the USB device dueto configuration change
12:11:36.870375 ERROR:  [3380] Unable to reset USB3380: error `Entity not found` (-5)!
12:11:36.870673 ERROR:  [USB3] Unable to reinitialize context: error: -13
Failed xtrx_open: -19
robert@sdr-pad:~$ /usr/lib/xtrx/test_xtrx -t -D usb3380://4/2/18
12:12:02.229333 ERROR:  [USB3] Unable to allocate USB3380 context: error: -19
Failed xtrx_open: -19
robert@sdr-pad:~$ sudo /usr/lib/xtrx/test_xtrx -t 
12:12:33.069705 WARN:   [PCIE] XTRX PCIe driver isn't loaded
CPU Features: SSE2+ SSE4.1+ AVX+ FMA-
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4083 us DELTA     31 us LATE   2067 us 16384 samples
RX STAT Overruns:0
Success!
Processed 1 devs, each: RX 2 x 1.978 = 3.956 MSPS (WIRE: 15.825017)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s) 
robert@sdr-pad:~$ 
```

sergforce, 2019-01-11:
If you have just one XTRX device in the system you may omit device specification (-D key), it'll automatically autodetect it.

Which system are you using? I wonder why udev rules haven't taken place.

RobertK66, 2019-01-11:
Hi, this is a ASUS ZENBOOK UX32VD with intel i5.
I cleaned all partitions and made a fresh install of ubuntu-16.04.5-desktop-amd64.iso with only one user and password......
Then I struggled a little with #25.  But in the end I got a clean install with all deb packages prepared by @mazsi-on-xtrx  here:   https://github.com/xtrx-sdr/images/issues/27#issuecomment-453273479

RobertK66, 2019-01-11:
I solved the mystery: The udev rulefile must be named **50-usb-xtrx.rules** iso 50--usb-xtrx.rules like I typed it from your recipe 😉 .

if the file has the 2 -- in it it seems not to work. renaming and restarting with
`sudo /etc/init.d/udev restart`
does the trick. 
50-usb-xtrx.rules -> test_xtrx  works without sudo
50--usb-xtrx.rules -> test_xtrx only  works with sudo

Thx. a lot for all your help. Heavy weather for occasional Linux users 😄.

sergforce, 2019-01-11:
Thanks, good to know!

sergforce, 2019-01-13:
I updated images repo to include this rules file. If someone still has the problem, please reopen.

closed 2019-01-13 by sergforce