xtrx-sdr/images #25: linux driver .deb manifest not updated

mazsi-on-xtrx, 2019-01-04:
the .deb manifest files (postinst & prerm) for xtrx-pcie-linux-driver-0.0.1-2_all.deb were not upgraded, they still have version "0.0.1-1" hardcoded in them.

this makes installation fail with "Directory .. does not exist":

>Selecting previously unselected package xtrx-pcie-linux-driver.
(Reading database ... 63247 files and directories currently installed.)
Preparing to unpack xtrx-pcie-linux-driver-0.0.1-2_all.deb ...
Unpacking xtrx-pcie-linux-driver (0.0.1-2) ...
Setting up xtrx-pcie-linux-driver (0.0.1-2) ...
/var/lib/dpkg/info/xtrx-pcie-linux-driver.postinst: 6: /var/lib/dpkg/info/xtrx-pcie-linux-driver.postinst: status: not found
0
Error! Could not find module source directory.
Directory: /usr/src/xtrx-0.0.1-1 does not exist.
dpkg: error processing package xtrx-pcie-linux-driver (--install):
 subprocess installed post-installation script returned error exit status 2
Errors were encountered while processing:
 xtrx-pcie-linux-driver

sergforce, 2019-01-11:
Thanks for the catch!

Fixed and uploaded new DEB

sergforce, 2019-01-13:
Thanks for the catch!

closed 2019-01-13 by sergforce