xtrx-sdr/images #27: libusb3380 dependency libusb-1.0 vs libusb-1.0-0

mazsi-on-xtrx, 2019-01-04:
the libusb1 dependency for libusb3380-*.deb is listed as "libusb-1.0" in the .deb manifest (control).

probably that should be "libusb-1.0-0" (with an extra "-0"); without that i think there is no installation candidate in the repository to satisfy the dependency.

RobertK66, 2019-01-10:
I have the same problem here
I used a fresh installed machine with current Ubuntu 16.04 LTS. Got my xtrx yesterday.
Now trying to install with 
`sudo apt install ./libusb3380-0.0.1-2-amd64.deb
`
i get: 
...
The following packages have unmet dependencies:
 libusb3380 : Depends: libusb-1.0 but it is not installable

RobertK66, 2019-01-10:
With help of this thread: https://ubuntuforums.org/showthread.php?t=110458 I was able to modify the dependency of the libusb3380-0.0.1-2-amd64.deb  from libusb-1.0 to libusb-1.0-0

This helped to install it without errors.

mazsi-on-xtrx, 2019-01-10:
RobertK66, i think the "could not locate dkms.conf file" error is probably because of issue #25.
i've uploaded a set of fixed .deb files in [this fork](https://github.com/mazsi-on-xtrx/images) for your convenience.

RobertK66, 2019-01-10:
HI @mazsi-on-xtrx, 
Thx. a lot. I will give it a try....
Maybe this also solves my struggling with #24 .....

mazsi-on-xtrx, 2019-01-10:
sorry @RobertK66, this might not help you with issue #24: i think with your setup (accessing XTRX via usb) you don't even need the low level pcie driver (see the "XTRX Host Software Block Diagram" in the Dec 10, 2017 update on the crowdsource page, that shows why).

sergforce, 2019-01-11:
Fixed and uploaded new DEB

sergforce, 2019-01-13:
Thanks for the catch!

closed 2019-01-13 by sergforce