xtrx-sdr/images #28: xtrx board in (mini) PCIe slot prevents computer from booting - SMB/i2c problem?

mazsi-on-xtrx, 2019-01-13:
this is probably the FAQ 2 ("prevents system to boot") issue.

after installing an xtrx board (directly in a mini PCIe slot, or using a commercial PCIe to mini PCIe adatper), some computers wouldn't boot anymore, they won't even POST. in a different computer the same xtrx board works without any problem.

i think this might be caused by the non-standard usage of the SMB pins on the mini PCIe connector (pin 30 and pin 32). if i disconnect those traces, then the failing setups start working (the computers boot, the xtrx board is recognised and shown in 'lspci' output).

checking the SMB CLK line on a failing setup without the xtrx board installed shows activity at boot time, i think the BIOS is accessing the EEPROM on the DIMMs. checking the same line with the xtrx board installed shows no activity, with the line being pulled down to GND. i think the xtrx board pulls it down, which prevents the BIOS from accessing the DIMM EEPROM, which prevents booting.

on the xtrx board the SMB CLK is repurposed as MHZ_OUT, the block diagram on the crowdsource page shows that it comes directly from the onboard VCTCXO. is that accurate? then probably there is no way to put that output to HIGHZ until after system boot has completed, right? (maybe something to change in a future hardware revision.)

workarounds that might work:
- using a motherboard which has a separate i2c bus for the PCIe slot where the xtrx is installed,
- using the official adapter board: that board has the SMB signals unconnected on the PCIe fingers,
- cutting the SMB traces on a commercial carrier board, if connected,
- cutting the SMB traces on the back of the xtrx card (i did not thry this, but there are jumpers on the PCB designed for this).

just a final question: will the VCTCXO not be demaged by the 3V3 on SMB CLK (= on its output)? i never see any signal on the MHZout output, how do i turn that on to see that it's working?

sergforce, 2019-01-13:
Thank you for the notes. By default, all used reserved pins by XTRX are in hi-z and can be activated to in or out by software. SMB CLK is a different thing. VCTCXO is exposed directly through the non-populated resistor J9. It's open by default. To expose internal clock to that pin you need manually solder J9. Since it's very unlikely that most really need it and as you mentioned it can cause interference on some problem we left J9 open. 
I'll update diagrams to actually reflect this.

mazsi-on-xtrx, 2019-01-13:
you are absolutely right, of course: i mixed up SMB CLK and SMB DATA (i was probing CLK first, i forgot about switching to DATA; for the fix i did cut both traces). sorry for the confusion!

it's just like you said, jumper for CLK (pin 30, right side) is open; jumper for DATA (pin 32, left side) is closed:

[image]

(this also clearly explains why i see no MHZOut signal.)

probing SMB CLK (ch1, blue) and SMB DATA (ch2, yellow) with no xtrx installed shows the lines normally pulled up (as expected for i2c), and then the transactions when the system boots:

[image]

but after installing the xtrx board SMB DATA, is pulled down and stays like that (SMB CLK is not affected, just like you explained):

[image]

i'm still thinking that it's shomehow about the SMB pins, i.e. SMB DATA, since disconnecting those traces allows the computer to boot.

mazsi-on-xtrx, 2019-01-14:
please note: SMB SDA is not at GND, it is ~0.76V (i measured that separately).

could it be that the PMICs are not programmed yet; there is no VCCO, i.e. it is ~GND and can sink some current; and the SMB DATA pin's clamping diode inside the FPGA pulls down the line? (see e.g. xilinx AR# 45985.)

i've been doing these tests with the factory default flash content, i'll check now the new bitfiles available in this git repo. they might be different, i.e. they might set up the PMICs.

mazsi-on-xtrx, 2019-01-14:
the available bitfiles (factory default, 3b4c604, 747a8fa) show no difference: SMB DATA is stuck ~0.76V.

mazsi-on-xtrx, 2019-03-07:
i created a minimal test design which initializes the PMICs upon powerup: mazsi-on-xtrx/xtrx-system-test@8bdab14d27a4347e513b4433db67816fdd61b259, bitfile: https://github.com/mazsi-on-xtrx/xtrx-system-test/raw/v20190307/top.bin

i tested this design / bitfile in a machine (VIA ARTiGO A1200) which was not booting earlier, now the XTRX board works as expected (computer boots, card shows up in lspci).