xtrx-sdr/images #31: XTRX freezes streaming when GPS is not locked

f4exb, 2019-01-19:
When GPS is not locked the streaming is frozen briefly and regularly.

sergforce, 2019-01-19:
Can you describe more about streaming got frozen vs GPS lock?

Can you describe what happened with GPS?

f4exb, 2019-01-19:
For the GPS itself maybe it was due to a bad connection because after trying the antenna on another receiver and plugging it back to XTRX it is working again. 

However when the GPS is not locked (easy to do by unplugging the GPS antenna) then periodically the stream of samples collected with `xtrx_recv_sync_ex` hangs briefly (half a second or so) which is quite noticeable on a spectrum display and annoying. This means that it does need GPS lock for proper operation. I think operation should be smooth even when GPS cannot be locked.

sergforce, 2019-01-19:
Good, that GPS is working.

Can you provide a log with maximum level (5 or 6) when it freezes without GPS and works smoothly with GPS sync? 

GPS sync isn't used in streaming (i.e. realigning data block etc), only corresponding timestamps are generated. So there's no dependency on lock nither in FPGA nor in host library (except especially timed streaming). But during GPS lock XTRX generates 1PPS interrupt events (and also generate UART data if a port is open) and this may cause some interference.

f4exb, 2019-01-19:
Indeed the problem does not seem to be with the streaming so I looked at the issue differently and I think it comes from my software but I still need your help.

1. How do I set the message level to maximum? I usually only get INFO messages from the XTRX library which is OK in normal conditions but it could indeed be useful to get the debug messages.

2. The problem itself comes from this piece of code that checks regularly for GPS status. It runs fine when GPS is locked but blocks for some time when GPS is not locked. It became obvious with the debug messages. Here is the code:
```
bool DeviceXTRXShared::get_gps_status()
{
    uint64_t val = 0;

    qDebug("DeviceXTRXShared::get_gps_status: before");
    int res = xtrx_val_get(m_dev->getDevice(), XTRX_TRX, XTRX_CH_AB, XTRX_WAIT_1PPS, &val);
    qDebug("DeviceXTRXShared::get_gps_status: after");

    if (res) {
        return false;
    }

    return val != 0;
}
```
The program clearly stalls between the "before" and "after" message.

I have inferred that way to check for GPS status but this is probably not the way to do it. Can you please indicate how I could do this without impacting the normal work of the device?

sergforce, 2019-01-19:
Hi,

There's 2 logging function that are exposed in high level:
```
XTRX_API void xtrx_log_setfunc(xtrx_logfunc_t func);
XTRX_API void xtrx_log_setlevel(int sevirity, const char sybsystem[4]);
```

First one alter custom output, so you can redirect all messages out of the console, to e.g. a GUI widget.
Second does what you want. `sevirity` tells how much you're interested in
```
1 - Errors
2 - Warning
3 - Info
4 - Detail
5 - LMS7 interaction
6 - Driver calls
7 - MAX
```

`sybsystem` is reserved for future message filtration and can be set to NULL, indicating ALL categories.

You get it right `XTRX_WAIT_1PPS` actually waits for the next 1PPS event and immediately exit when it happens. It has around 1.5sec timeout, if 1PPS wasn't observed in the period error is reported.

`XTRX_OSC_LATCH_1PPS` returns the number of calculated ref clocks between 1PPS signal. When you see a non-zero value that means there were 1PPS events. However if GPS loss lock, this value isn't updated and you'll just see the old value. The problem here you may trim TCXO that you'll be getting the same value for a long while. Internally there's a special register that reports a number of 1PPS events since the start, and if you keep seeing the same number that means GPS sync was lost.

So you inspired me to add a simpler way just to get GPS status. I'll add it in Feb release. Meantime I can expose the counter that reports the number of 1PPS events.

f4exb, 2019-01-19:
Thanks for the details. For sure a simple function returning GPS status will be much appreciated.

f4exb, 2019-01-19:
In fact when lock is acquired the value returned by XTRX_OSC_LATCH_1PPS changes by 1 digit over a certain amount of calls and when it is not locked it is always the same probably because the counter never gets updated. This can be a workaround.

sergforce, 2019-01-22:
I think you can use that approach while there's no better way to detect GPS lock

chemeris, 2019-09-27:
Closing this ticket since the issue looks resolved.

closed 2019-09-27 by chemeris