xtrx-sdr/images #32: xtrx-0.0.1-1 changed to xtrx-0.0.1-2

IslamAlam, 2019-01-20:
Currently, there is an issue during the build because the xtrx-0.0.1-1 was updated to xtrx-0.0.1-2
Therefore, the README.md needs to be updated to include thus modification.

Now, the only issue which I am facing is related to loading the driver `modprobe xtrx`.

What is the proposal to solve this issue?

>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$ sudo ln -s /usr/local/src/xtrx-0.0.1-1/ /usr/src/
>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$ sudo /usr/sbin/dkms add -m xtrx -v "0.0.1-1"
Error! Could not find module source directory.
Directory: /usr/src/xtrx-0.0.1-1 does not exist.
>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$ sudo ln -s /usr/local/src/xtrx-0.0.1-1/ /usr/src/
ln: failed to create symbolic link '/usr/src/xtrx-0.0.1-1': File exists
>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$ sudo ln -s /usr/local/src/xtrx-0.0.1-2/ /usr/src/
..
..
..
>DKMS: build completed.
>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$ modprobe xtrx
>modprobe: FATAL: Module xtrx not found in directory /lib/modules/4.15.0-43-generic
>ge49gar@ge49gar-ThinkPad-X230-Tablet:~/xtrx_images/sources/build$

sergforce, 2019-01-24:
Agh, yes. White list VID/PID is a very annoying problem! I'll add this into the FAQ
Did you manage to patch it?

IslamAlam, 2019-01-28:
Not yet, it would be required to create a mod bios and physically flash the bios to the chip via EEPROM bios programmer.

After solving the issue, I will write about it here then.

closed 2019-01-28 by IslamAlam