xtrx-sdr/images #33: udev rules for XTRX Octopack not working

dmlonard, 2019-01-23:


dmlonard, 2019-01-23:
Update -
lspci is still showing <access denied>, but xtrx_tests appears to be working.

 lspci -v -d 10ee:
04:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 151
	Memory at df810000 (32-bit, non-prefetchable) [size=4K]
	Memory at df800000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

05:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 153
	Memory at df710000 (32-bit, non-prefetchable) [size=4K]
	Memory at df700000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

06:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 154
	Memory at df610000 (32-bit, non-prefetchable) [size=4K]
	Memory at df600000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

07:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 155
	Memory at df510000 (32-bit, non-prefetchable) [size=4K]
	Memory at df500000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

08:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 156
	Memory at df410000 (32-bit, non-prefetchable) [size=4K]
	Memory at df400000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

09:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 157
	Memory at df310000 (32-bit, non-prefetchable) [size=4K]
	Memory at df300000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

0a:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 158
	Memory at df210000 (32-bit, non-prefetchable) [size=4K]
	Memory at df200000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

0b:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 159
	Memory at df110000 (32-bit, non-prefetchable) [size=4K]
	Memory at df100000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: xtrx
	Kernel modules: xtrx

sergforce, 2019-01-23:
To get capabilities information you need to run `lspci` under superuser. If `test_xtrx` works under a normal user that udev rules work.

chemeris, 2019-09-27:
Closing this ticket since the issue seems resolved.

closed 2019-09-27 by chemeris