xtrx-sdr/images #34: SoapySDRUtil hangs when xtrx module in place

unixpunk, 2019-01-26:
Things seem to be hanging:

steve@tweak:~$ SoapySDRUtil --probe="driver=xtrx"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=xtrx
12:45:18.560297 DEBUG:  xtrxllpciev0_discovery:264 [PCIE] pcie: Found `pcie:///dev/xtrx0`
[INFO] Make connection: 'pcie:///dev/xtrx0'
12:45:18.568533 INFO:   [CTRL] PCI:/dev/xtrx0: XTRX Rev4 (04000013)
12:45:18.568545 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO @0.0
12:45:18.568553 INFO:   [PCIE] PCI:/dev/xtrx0: Device `pcie:///dev/xtrx0` was opened
CPU Features: SSE2+ SSE4.1+ AVX- FMA-
**---HUNG HERE---**

**Additional info:**
steve@tweak:~$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-g2da7c192
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libHackRFSupport.so   (0.3.3-3c514ce)
Module found: /usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so     (18.10.0-59d51d5)
Module found: /usr/local/lib/SoapySDR/modules0.7/libXTRXSupport.so
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspySupport.so   (0.1.2-99756be)
Module found: /usr/local/lib/SoapySDR/modules0.7/libairspyhfSupport.so (0.1.0-e8582d4)
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so   (0.5.1-2a7363e)
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so   (0.3.1-4534646)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so  (0.2.0-ff37683)
Available factories... airspy, airspyhf, hackrf, lime, remote, rtlsdr, sdrplay, xtrx
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

unixpunk, 2019-01-26:
To add, using sudo doesn't help either.

sergforce, 2019-01-26:
Can you run `SoapySDRUtil --probe="driver=xtrx,loglvl=6"` to see where it stuck?

unixpunk, 2019-01-27:
I've just come to find this isn't an issue with SoapySDR specifically but with either the hw or the rest of the drivers/software in play from this repo.  I have the exact same behavior when attempting to use the device in sdrangel as well.  I suppose the next step here would be to start from scratch, compiling each piece of this puzzle individually?  Unless you have another method to see where the issue might be?

Here is the dmesg output: https://pastebin.com/FzcRbBA5

From sdrangel when run from bash:
21:02:46.138787 INFO:   [CTRL] PCI:/dev/xtrx0: XTRX Rev4 (04000013)
21:02:46.138836 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO @0.0
21:02:46.138851 INFO:   [PCIE] PCI:/dev/xtrx0: Device `pcie:///dev/xtrx0` was opened
CPU Features: SSE2+ SSE4.1+ AVX- FMA-

Thanks in advance for any help you're able to offer!  This is my first attempt at getting this device to work, though I've had it since the first batch was shipped.  The laptop is an older Dell e6400, in case that also helps.  No usb3 adapter, just straight pcie.

unixpunk, 2019-01-27:
In case this also helps:
steve@tweak:~$ cat /proc/cpuinfo
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 23
model name	: Intel(R) Core(TM)2 Duo CPU     T9550  @ 2.66GHz
stepping	: 10
microcode	: 0xa0b
cpu MHz		: 800.000
cache size	: 6144 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm sse4_1 xsave lahf_lm kaiser tpr_shadow vnmi flexpriority dtherm ida
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf
bogomips	: 5319.90
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 23
model name	: Intel(R) Core(TM)2 Duo CPU     T9550  @ 2.66GHz
stepping	: 10
microcode	: 0xa0b
cpu MHz		: 800.000
cache size	: 6144 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm sse4_1 xsave lahf_lm kaiser tpr_shadow vnmi flexpriority dtherm ida
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf
bogomips	: 5319.90
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

unixpunk, 2019-01-27:
Figured it out, had old driver and dsp deb packages installed...wiped them, started from scratch here and seems good to go now.  Sorry for the fire drill!

closed 2019-01-27 by unixpunk