xtrx-sdr/images #35: Build on CentOS 7 Issues

fstop22, 2019-01-28:
I am trying to build the sources for the images on CentOS 7.6; however, I am running into multiple problems with CMake and GCC:
1. In CMakeLists.txt ```gcc -print-multiarch``` returns null string.  Using ```gcc -dumpmachine``` seems to work.
2. std=c99 must be explicitly set for GCC 4.8.5.
3. Building libxtrxdsp keeps erroring with
```
images/sources/libxtrxdsp/xtrxdsp.c: In function ‘cpu_features_init’:
images/sources/libxtrxdsp/xtrxdsp.c:64:2: error: Parameter to builtin not valid: fma
features->fma = __builtin_cpu_supports("fma");
```

b3sholt, 2019-02-13:
I ran into the same issues.  After trying a bunch of different stuff I gave up and have just been using the XTRX on Ubuntu 16.04.  Sorry I don't have a solution, but you are not alone.

sergforce, 2019-02-13:
Can you update only `libxtrxdsp` and retest?
You can use `-DFORCE_ARCH` to get rid of `    1. gcc -print-multiarch` problem

``` cmake -DFORCE_ARCH=x86_64 -DCMAKE_C_FLAGS="--std=gnu99" ..```

b3sholt, 2019-04-03:
Hi @sergforce, I just pulled the latest libxtrxdsp and was able to successfully build on CentOS 7.6.  Running into a different issue with the xtrx kernel module, but it looks like your gcc updates worked.

chemeris, 2019-09-27:
@b3sholt Are you posting your RPMs or build instructions for CentOs somewhere? We'd love to add it a list of supported Linux distros.

Otherwise, I'm closing this tickets since the issue seems resolved.

closed 2019-09-27 by chemeris