xtrx-sdr/images #36: Error from GNURadio block

Astromis, 2019-01-29:
Hello.

I use the XTRX in my projects and recently I've found an error that causes when I'm trying using it in GNURadio with Audio Sink block.

In the end of post I've attached a log from GNURadio terminal as well as GNURadio simple FM receiver graph where problem causes.

I use latest program modules including driver, libxtrxll, libxtrx and so on. XTRX is connected through miniPCIe. OS Ubuntu 16.06. osmocom Source is built from xrtx repo as well. 

---------------------------------------------
File: 

[untitled.grc.zip](https://github.com/xtrx-sdr/images/files/2807604/untitled.grc.zip)
---------------------------------------------
Generating: '/home/xtrx/top_block.py'

Generating: '/home/xtrx/top_block.py'

Executing: /usr/bin/python -u /home/xtrx/top_block.py

CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
Using avx for xtrxdsp_iq16_sc32
Using avx for xtrxdsp_iq8_ic16
Using avx for xtrxdsp_iq16_ic16i
Using avx for xtrxdsp_iq8_ic8i
Using avx for xtrxdsp_sc32i_iq16
Using avx for xtrxdsp_iq8_sc32
Using avx for xtrxdsp_iq8_sc32i
Using avx for xtrxdsp_iq16_sc32i
Using avx for xtrxdsp_sc32_iq16
Using avx for xtrxdsp_ic16i_iq16
gr-osmosdr v0.1.4-104-ge239811d (0.1.5git) gnuradio 3.7.10
built-in source types: file fcd rtl_tcp hackrf rfspace soapy redpitaya xtrx 
15:01:21.466894 DEBUG:  xtrxllpciev0_discovery:264 [PCIE] pcie: Found `pcie:///dev/xtrx0`
15:01:21.480104 DEBUG:  xtrxllpciev0_discovery:264 [PCIE] pcie: Found `pcie:///dev/xtrx0`
[INFO] Make connection: 'pcie:///dev/xtrx0'
15:01:21.485120 INFO:   [CTRL] PCI:/dev/xtrx0: XTRX Rev4 (04000013)
15:01:21.485133 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA STOP MIMO @0.0
15:01:21.485139 INFO:   [PCIE] PCI:/dev/xtrx0: Device `pcie:///dev/xtrx0` was opened
15:01:21.600756 INFO:   [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x000304
15:01:21.700973 INFO:   [CTRL] PCI:/dev/xtrx0: FPGA V_GPIO set to 3280mV
15:01:21.701020 INFO:   [CTRL] PCI:/dev/xtrx0: LMS PMIC DCDC out set to VA18=1880mV VA14=1480mV VA12=1340mV
15:01:21.704268 INFO:   [CTRL] PCI:/dev/xtrx0: FPGA V_IO set to 1800mV
15:01:21.714423 INFO:   [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x000306
15:01:21.724746 INFO:   [LSM7] PCI:/dev/xtrx0: LMS VER:7 REV:1 MASK:1 (3841)
15:01:21.724764 INFO:   [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x00031e
15:01:21.724851 INFO:   [DBGP] Starting XTRX debug thread
15:01:21.724899 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA SKIP MIMO @0.0
15:01:21.724954 INFO:   [LSM7] PCI:/dev/xtrx0: 0x0124[00, 00]
15:01:21.725051 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA STOP MIMO (BLK:0 TS:0); TX DMA SKIP MIMO @0.0
15:01:21.726832 INFO:   [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x00031e
15:01:21.752471 INFO:   [XTRX] PCI:/dev/xtrx0: Set INT RefClk to 26000000 based on 26059985 measurement
15:01:21.752508 INFO:   [LSM7] PCI:/dev/xtrx0: CGEN: VCO/2=1152000000 k/2=24 int=88 frac=645277
15:01:21.753054 INFO:   [LSM7] PCI:/dev/xtrx0: CGEN: binary result: 145
15:01:21.753370 INFO:   [LSM7] PCI:/dev/xtrx0: CGEN: Retuned [147:152] -> 149
15:01:21.753530 INFO:   [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x00031e
15:01:21.753542 INFO:   [CTRL] PCI:/dev/xtrx0: FPGA V_IO set to 1800mV
15:01:21.753553 INFO:   [LMSF] PCI:/dev/xtrx0: rxrate=6.000MHz txrate=0.000MHz actual_master=48.000MHz rxdecim=2(h_1) txinterp=1(h_1) RX_ADC=12.000MHz TX_DAC=6.000MHz hintr=0 hdecim=0 delay=0 NRXFWD=0 LML1HID=3 LML2HID=1 RX_div=0 TX_div=0 RX_TSP_div=1 TX_TSP_div=1 FclkRX=0.000 (PHS=0) RXx2=0
15:01:21.753627 INFO:   [LMSF] PCI:/dev/xtrx0: FE_FREQ rx=1 type=0 freq=452000000.000000 ch=3
15:01:21.753746 INFO:   [LSM7] PCI:/dev/xtrx0: SXR: initial VCO=3616000000 DIVH=3 VCOs:0010
15:01:21.753757 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: VCO2 N=139 frac=80659
15:01:21.754041 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: binary result: 255
15:01:21.754084 INFO:   [LSM7] PCI:/dev/xtrx0: SXR: VCO2 [0;0] -> 0
15:01:21.755163 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: VCO2 N=139 frac=80659
15:01:21.755479 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: binary result: 151
15:01:21.756351 INFO:   [LSM7] PCI:/dev/xtrx0: SXR: VCO2 [153;175] -> 164
15:01:21.756381 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: VCO2 N=139 frac=80659
15:01:21.756686 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: binary result: 155
15:01:21.757423 INFO:   [LSM7] PCI:/dev/xtrx0: SXR: VCO2 [158;175] -> 166
15:01:21.757482 INFO:   [LMSF] PCI:/dev/xtrx0: NCO ch=1 type=3 freq=0
15:01:21.757565 INFO:   [LMSF] PCI:/dev/xtrx0: Set gain 30.0 to 0 on 1 channel
15:01:21.757589 INFO:   [LSM7] PCI:/dev/xtrx0: RFE: set_lna(0 -> 0) => 15
15:01:21.757637 INFO:   [LMSF] PCI:/dev/xtrx0: Set gain 20.0 to 0 on 1 channel
15:01:21.757649 INFO:   [LSM7] PCI:/dev/xtrx0: RFE: set_lna(10 -> 12) => 7
15:01:21.757716 INFO:   [LMSF] PCI:/dev/xtrx0: Set RX band to 2 (L)
15:01:21.757856 INFO:   [CTRL] PCI:/dev/xtrx0: RX_ANT: 1 TX_ANT: 0
15:01:21.757940 INFO:   [LSM7] PCI:/dev/xtrx0: 0x0124[18, 00]
15:01:21.758054 INFO:   [LSM7] PCI:/dev/xtrx0: TIA: CCOMP=2 CFB=270 RCOMP=10
gr::log :INFO: audio source - Audio sink arch: alsa
15:01:21.789162 INFO:   [LMSF] PCI:/dev/xtrx0: AFE TX=[0;0] RX=[1;0]
15:01:21.789320 INFO:   [LSM7] PCI:/dev/xtrx0: 0x0124[1c, 00]
15:01:21.789457 INFO:   [LMSF] PCI:/dev/xtrx0: RBB Restore BW[0]=6000000
15:01:21.789465 INFO:   [LSM7] PCI:/dev/xtrx0: TIA: CCOMP=2 CFB=270 RCOMP=10
15:01:21.789548 INFO:   [LMSF] PCI:/dev/xtrx0: RBB Restore DSP[0]=0
15:01:21.789865 INFO:   [XTRX] PCI:/dev/xtrx0: RX ititialized to 65536 bytes paket size
15:01:21.789879 INFO:   [BPCI] PCI:/dev/xtrx0: RX DMA 16 bit SISO (BLK:16384 TS:4096); TX DMA SKIP MIMO @0.0
15:01:21.789886 INFO:   [CTRL] PCI:/dev/xtrx0: Placing TC @4096 on 0 data: 0 stat:20
[INFO] SoapyXTRX::activateStream(RX) 0 Samples per packet; res = 0
**15:01:22.109171 ERROR:  [BPCI] PCI:/dev/xtrx0: RX DMA STAT O- 00000000 Bytes YR03 07/38 I:32**

b3sholt, 2019-01-29:
Try placing 'xtrx' in the device arguments.  This will bypass Soapy and use the actual xtrx drivers.  See if that helps.

b3sholt, 2019-02-14:
Hi Astromis, just wanted to see if you have been able to test this?  If it worked that would be good to know so they can close this issue.  Thanks!

Brayan1710, 2021-04-15:
Hello @Astromis I am having the same error but I'm using USB 3. I would like to know if you can fix it. Thank you.

Astromis, 2021-04-15:
@Brayan1710  hi.

Unfortunately I can't help you now. You see, It was 2 years ago when I worked with xtrx but now I'm working in a different field. Sorry.