xtrx-sdr/images #37: DMA errors

schnet25, 2019-01-29:
I just received my xtrx and I am trying to get it running on a Solidrun Hummingboard edge running Armbian. Everything builds fine; however, when I try to load the kernel module I get DMA issues. Below are the relevant dmesg traces.

[    7.160363] xtrx: loading out-of-tree module taints kernel.
[    7.161727] xtrx: Initializing 0000:01:00.0
[    7.161761] xtrx 0000:01:00.0: enabling device (0140 -> 0142)
[    7.170147] pps pps1: new PPS source xtrx_pps
[    7.170166] xtrx 0000:01:00.0: Failed to enable MSI, falling back to legacy mode.
[    7.170235] xtrx: buf[0]=80008000 [virt (ptrval)] => 800087ff
[    7.170265] xtrx: buf[1]=80010000 [virt (ptrval)] => 800107ff
[    7.170292] xtrx: buf[2]=80018000 [virt (ptrval)] => 800187ff
[    7.170321] xtrx: buf[3]=80020000 [virt (ptrval)] => 800207ff
[    7.170355] xtrx: buf[4]=80028000 [virt (ptrval)] => 800287ff
[    7.170385] xtrx: buf[5]=80030000 [virt (ptrval)] => 800307ff
[    7.170421] xtrx: buf[6]=80038000 [virt (ptrval)] => 800387ff
[    7.170453] xtrx: buf[7]=80040000 [virt (ptrval)] => 800407ff
[    7.170491] xtrx: buf[8]=80048000 [virt (ptrval)] => 800487ff
[    7.170526] xtrx: buf[9]=80050000 [virt (ptrval)] => 800507ff
[    7.170560] xtrx: buf[10]=80058000 [virt (ptrval)] => 800587ff
[    7.170596] xtrx: buf[11]=80060000 [virt (ptrval)] => 800607ff
[    7.170630] xtrx: buf[12]=80068000 [virt (ptrval)] => 800687ff
[    7.170672] xtrx: buf[13]=80070000 [virt (ptrval)] => 800707ff
[    7.170878] xtrx: buf[14]=80078000 [virt (ptrval)] => 800787ff
[    7.170922] xtrx: buf[15]=80080000 [virt (ptrval)] => 800807ff
[    7.170954] xtrx: buf[16]=80088000 [virt (ptrval)] => 800887ff
[    7.170982] xtrx: buf[17]=80090000 [virt (ptrval)] => 800907ff
[    7.171014] xtrx: buf[18]=80098000 [virt (ptrval)] => 800987ff
[    7.171042] xtrx: buf[19]=800a0000 [virt (ptrval)] => 800a07ff
[    7.171077] xtrx: buf[20]=800a8000 [virt (ptrval)] => 800a87ff
[    7.171104] xtrx: buf[21]=800b0000 [virt (ptrval)] => 800b07ff
[    7.171137] xtrx: buf[22]=800b8000 [virt (ptrval)] => 800b87ff
[    7.171174] xtrx: buf[23]=800c0000 [virt (ptrval)] => 800c07ff
[    7.171205] xtrx: buf[24]=800c8000 [virt (ptrval)] => 800c87ff
[    7.171233] xtrx: buf[25]=800d0000 [virt (ptrval)] => 800d07ff
[    7.171275] xtrx: buf[26]=800d8000 [virt (ptrval)] => 800d87ff
[    7.171339] xtrx: buf[27]=800e0000 [virt (ptrval)] => 800e07ff
[    7.171421] xtrx: buf[28]=800e8000 [virt (ptrval)] => 800e87ff
[    7.171481] xtrx: buf[29]=800f0000 [virt (ptrval)] => 800f07ff
[    7.171512] xtrx: buf[30]=800f8000 [virt (ptrval)] => 800f87ff
[    7.171539] xtrx: buf[31]=80100000 [virt (ptrval)] => 801007ff
[    7.171608] xtrx: Failed to allocate 31 DMA buffer
[    7.171632] xtrx 0000:01:00.0: Failed to register TX DMA buffers.

Do you have any suggestions/advice on how I could resolve this? Thanks in advance!

Tom

sergforce, 2019-01-29:
It seems there's a limitation of DMA coherent pool on your ARM.

On NVIDIA Terga this problem is resolved by adding 'coherent_pool= ' into kernel boot parameter list https://devtalk.nvidia.com/default/topic/770634/jetson-tk1/large-coherent-dma-blocks/

Can you check this with your board?

schnet25, 2019-01-29:
Thanks for the quick response. Setting coherent_pool=32M on my cmdline worked! I appreciate the help

closed 2019-01-29 by schnet25

kcman100, 2021-12-11:
Hello, I am installing on a Raspberry Pi 4 Compute Module on a the Compute IO motherboard.  Using a mini-PCIe to PCIe 1x adapter card.  The Xlinix driver seems okay according to lspci.  But I am getting the dmesg error as above.  "xtrx: Failed to allocate 31 DMA buffer" and "xtrx 0000:01:00.0: Failed to register RX DMA buffers."

I see the comment about "coherent_pool=32M", but not sure that applies to a RPi version of Linux (RaspianOS, using 32-bit version for now).

Any ideas?  Thanks, Ken

kcman100, 2021-12-11:
Okay, got it.  Edit /boot/cmdline.txt and add "coherent_pool=32M" on that line.

For example (all on one line):
console=serial0,115200 console=tty1 root=PARTUUID=855fca69-02 rootfstype=ext4 fsck.repair=yes coherent_pool=32M rootwait

Now I get /dev/xtrx0 and the text_xtrx passes ("Success!").