xtrx-sdr/images #38: OpenBTS and OpenLTE

Peterthegreat, 2019-01-29:
Hello

Is there any support for OpenBTS or OpenLTE for XTRX? maybe using the USB3 adapter?

> **LTE Cellular**
> The combination of XTRX’s accurate, stable clock, on-board GPSDO, and low-latency PCIe bus makes LTE possible out of the box.

sergforce, 2019-01-29:
We haven't tested OpenBTS & OpenLTE.

Right now Amarisfot LTE works best. We're working to get srsLTE & OAI support as well

Peterthegreat, 2019-01-30:
Thanks!
Any software is fine. Glad to hear that it's a work in progress

closed 2019-01-30 by Peterthegreat

joeldelosangeles, 2019-05-08:
Hi Sergeforce, 

We use UMTrx for our GSM base stations (we just ordered 40 units recently).

Any update on development in using xtrx for srsLTE or OAI?  We have 2 units of XTRX CS.  We won't mind working with XTRX Pro if needed as long as srsLTE or OAI is used.

Thanks,
Joel Delos Angeles

Dastgsm, 2020-08-08:
How about OAI and OAI5G-NR support over xtrx?