xtrx-sdr/images #40: Corrupt data issues

schnet25, 2019-01-31:
I am having trouble receiving data from my xtrx. I think it is an issue with my setup; however, I am posting an issue just in case it isn't. When I try to capture using your test_xtrx binary I get bad data. Here is my call:

**$ ./test_xtrx --dmarx --o captured.bin --samples=262144**
RX_TO_FILE: thread start
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=262144 SLICE=16384 (PARTS=16)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4386 us DELTA     34 us LATE   2372 us 16384 samples
PROCESSED RX SLICE 0 /1: res 0 TS:   16384       1923 us DELTA     62 us LATE   2310 us 16384 samples
PROCESSED RX SLICE 0 /2: res 0 TS:   24576       2473 us DELTA     45 us LATE   2781 us 16384 samples
PROCESSED RX SLICE 0 /3: res 0 TS:   32768       1376 us DELTA    159 us LATE   2269 us 16384 samples
PROCESSED RX SLICE 0 /4: res 0 TS:   40960      11452 us DELTA     36 us LATE  11709 us 16384 samples
PROCESSED RX SLICE 0 /5: res 0 TS:   49152        364 us DELTA     97 us LATE  10122 us 16384 samples
PROCESSED RX SLICE 0 /6: res 0 TS:   57344        380 us DELTA     44 us LATE   8498 us 16384 samples
PROCESSED RX SLICE 0 /7: res 0 TS:   65536        389 us DELTA     48 us LATE   6889 us 16384 samples
PROCESSED RX SLICE 0 /8: res 0 TS:   73728        358 us DELTA     44 us LATE   5244 us 16384 samples
PROCESSED RX SLICE 0 /9: res 0 TS:   81920        344 us DELTA     49 us LATE   3590 us 16384 samples
PROCESSED RX SLICE 0 /10: res 0 TS:   90112        882 us DELTA     48 us LATE   2472 us 16384 samples
PROCESSED RX SLICE 0 /11: res 0 TS:   98304       2022 us DELTA     72 us LATE   2519 us 16384 samples
PROCESSED RX SLICE 0 /12: res 0 TS:  106496       2071 us DELTA     91 us LATE   2633 us 16384 samples
PROCESSED RX SLICE 0 /13: res 0 TS:  114688       1935 us DELTA    114 us LATE   2635 us 16384 samples
PROCESSED RX SLICE 0 /14: res 0 TS:  122880       1967 us DELTA    131 us LATE   2685 us 16384 samples
PROCESSED RX SLICE 0 /15: res 0 TS:  131072       1630 us DELTA    188 us LATE   2455 us 16384 samples
RX STAT Overruns:0
RX_TO_FILE: buffer 0 written
RX_TO_FILE: thread exit: 1:1
Success!
Processed 1 devs, each: RX 2 x 3.711 = 7.422 MSPS (WIRE: 29.689563)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s) 

All looks good; however, when I inspected the data I see that it is not from the device. Some mem mapping is wrong somewhere and I am getting random kernel buffers back....

**$ strings captured.bin**
1-8-cross.2.0-1ubuntu2~18.04
5.26.1-6ubuntu0.2
0.9.6~r156-6ubuntu0.18.04.1
6.1.1-2ubuntu0.1
reUnitE19UMeasureFormatWidthiR10UErrorCode
_ZNK6icu_6013MeasureFormat18getPluralFormatterERKNS_11MeasureUnitE19UMeasureFormatWidthiR10UErrorCode
_ZNK6icu_6013MeasureFormat15getPerFormatterE19UMeasureFormatWidthR10UErrorCode
_ZNK6icu_6013MeasureFormat20withPerUnitAndAppendERKNS_13UnicodeStringERKNS_11MeasureUnitERS1_R10UErrorCode
_ZNK6icu_6015SimpleFormatter15formatAndAppendEPKPKNS_13UnicodeStringEiRS1_PiiR10UErrorCode
_ZN6icu_6015SimpleFormatter22getTextWithNoArgumentsEPKDsi
_ZN6icu_6013MeasureFormat20createCurrencyFormatERKNS_6LocaleER10UErrorCode
_ZN6icu_6014CurrencyFormatC1ERKNS_6LocaleER10UErrorCode
_ZN6icu_6014CurrencyFormatD0Ev
_ZN6icu_6013MeasureFormat20createCurrencyFormatER10UErrorCode
_ZN6icu_6022MeasureFormatCacheDataD2Ev
_ZN6icu_6022MeasureFormatCacheDataD1Ev
_ZN6icu_6022MeasureFormatCacheDataD0Ev
_ZN6icu_6014LocaleCacheKeyINS_22MeasureFormatCacheDataEED2Ev
_ZN6icu_6014LocaleCacheKeyINS_22MeasureFormatCacheDataEED1Ev
_ZN6icu_6013MeasureFormat17initMeasureFormatERKNS_6LocaleE19UMeasureFormatWidthPNS_12NumberFormatER10UErrorCode
_ZN6icu_6011PluralRules20createSharedInstanceERKNS_6LocaleE11UPluralTypeR10UErrorCode
_ZN6icu_6013ListFormatter14createInstanceERKNS_6LocaleEPKcR10UErrorCode
_ZN6icu_6013MeasureFormatC2ERKNS_6LocaleE19UMeasureFormatWidthR10UErrorCode
_ZN6icu_6013MeasureFormatC1ERKNS_6LocaleE19UMeasureFormatWidthR10UErrorCode
_ZN6icu_6013MeasureFormatC2ERKNS_6LocaleE19UMeasureFormatWidthPNS_12NumberFormatER10UErrorCode
_ZN6icu_6013MeasureFormatC1ERKNS_6LocaleE19UMeasureFormatWidthPNS_12NumberFormatER10UErrorCode
_ZN6icu_6014LocaleCacheKeyINS_22MeasureFormatCacheDataEED0Ev
_ZN6icu_6013MeasureFormat22setMeasureFormatLocaleERKNS_6LocaleER10UErrorCode
_ZNK6icu_6014LocaleCacheKeyINS_22MeasureFormatCacheDataEE12createObjectEPKvR10UErrorCode
_ZNK6icu_6013MeasureFormateqERKNS_6FormatE
_ZNK6icu_6013MeasureFormat13formatMeasureERKNS_7MeasureERKNS_12NumberFormatERNS_13UnicodeStringERNS_13FieldPositionER10UErrorCode
_ZN6icu_6017QuantityFormatter12selectPluralERKNS_11FormattableERKNS_12NumberFormatERKNS_11PluralRulesERNS_13UnicodeStringERNS_13FieldPositionER10UErrorCode
_ZN6icu_6017QuantityFormatter6formatERKNS_15SimpleFormatterERKNS_13UnicodeStringERS4_RNS_13FieldPositionER10UErrorCode
_ZNK6icu_6013MeasureFormat6formatERKNS_11FormattableERNS_13UnicodeStringERNS_13FieldPositionER10UErrorCode
[... 125 lines cut ...]
[  639.604951] xtrx: buf[15]=8c210000 [virt f57229e4] => 8c210fff
[  639.605015] xtrx: buf[16]=8c220000 [virt 057d36ca] => 8c220fff
[  639.605087] xtrx: buf[17]=8c230000 [virt 2356bc9e] => 8c230fff
[  639.605152] xtrx: buf[18]=8c240000 [virt dc9aeff7] => 8c240fff
[  639.605206] xtrx: buf[19]=8c250000 [virt 3b635f2c] => 8c250fff
[  639.605267] xtrx: buf[20]=8c260000 [virt 86354546] => 8c260fff
[  639.605331] xtrx: buf[21]=8c270000 [virt 36c26db1] => 8c270fff
[  639.605385] xtrx: buf[22]=8c280000 [virt 5a98b568] => 8c280fff
[  639.605439] xtrx: buf[23]=8c290000 [virt aa8d0502] => 8c290fff
[  639.605487] xtrx: buf[24]=8c2a0000 [virt 885c6964] => 8c2a0fff
[  639.605535] xtrx: buf[25]=8c2b0000 [virt 19abf3a0] => 8c2b0fff
[  639.605580] xtrx: buf[26]=8c2c0000 [virt 58bab1c5] => 8c2c0fff
[  639.605624] xtrx: buf[27]=8c2d0000 [virt 4ef2f16e] => 8c2d0fff
[  639.605693] xtrx: buf[28]=8c2e0000 [virt adee954c] => 8c2e0fff
[  639.605741] xtrx: buf[29]=8c2f0000 [virt 9aab3d14] => 8c2f0fff
[  639.605784] xtrx: buf[30]=8c300000 [virt 04d9b7b8] => 8c300fff
[  639.605834] xtrx: buf[31]=8c310000 [virt 44da79cb] => 8c310fff
[  639.609009] xtrx: call: REGION=3 VMA=40fcb7a2 vma->vm_pgoff=16384
[  639.609172] xtrx: ioctl(123459, 10000) [f4ab1000]
[  639.660955] xtrx: ioctl(12345b, 0) [f4ab1000]
[  639.660963] xtrx:  3V3 CTRL:0
[  639.662140] xtrx: VMA close.
[  639.662159] xtrx: VMA close.

Do you guys have any tips/clues/thoughts? In the meantime I am trying to bring up a different board with a mini PCIe to test on that. Thanks in advance!

schnet25, 2019-01-31:
Sure, I am running armbian on a quad core imx6.

$ uname -a
Linux cubox 4.19.19-cubox #5.74 SMP Thu Jan 31 11:36:26 EST 2019 armv7l armv7l armv7l GNU/Linux

Same data corruption with your command line arguments:

$ ./test_xtrx --dmarx -o captured.bin --samples=262144
RX_TO_FILE: thread start
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=262144 SLICE=16384 (PARTS=16)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4375 us DELTA     42 us LATE   2369 us 16384 samples
PROCESSED RX SLICE 0 /1: res 0 TS:   16384       1940 us DELTA     57 us LATE   2320 us 16384 samples
PROCESSED RX SLICE 0 /2: res 0 TS:   24576       2009 us DELTA     55 us LATE   2337 us 16384 samples
PROCESSED RX SLICE 0 /3: res 0 TS:   32768       1971 us DELTA     59 us LATE   2319 us 16384 samples
PROCESSED RX SLICE 0 /4: res 0 TS:   40960       2158 us DELTA     56 us LATE   2486 us 16384 samples
PROCESSED RX SLICE 0 /5: res 0 TS:   49152       1948 us DELTA     89 us LATE   2475 us 16384 samples
PROCESSED RX SLICE 0 /6: res 0 TS:   57344       1986 us DELTA     91 us LATE   2505 us 16384 samples
PROCESSED RX SLICE 0 /7: res 0 TS:   65536       1861 us DELTA     98 us LATE   2416 us 16384 samples
PROCESSED RX SLICE 0 /8: res 0 TS:   73728       1893 us DELTA     76 us LATE   2338 us 16384 samples
PROCESSED RX SLICE 0 /9: res 0 TS:   81920       1930 us DELTA     72 us LATE   2292 us 16384 samples
PROCESSED RX SLICE 0 /10: res 0 TS:   90112       1983 us DELTA     42 us LATE   2270 us 16384 samples
PROCESSED RX SLICE 0 /11: res 0 TS:   98304       2043 us DELTA     36 us LATE   2302 us 16384 samples
PROCESSED RX SLICE 0 /12: res 0 TS:  106496       1998 us DELTA     44 us LATE   2297 us 16384 samples
PROCESSED RX SLICE 0 /13: res 0 TS:  114688       2075 us DELTA     41 us LATE   2366 us 16384 samples
PROCESSED RX SLICE 0 /14: res 0 TS:  122880       2226 us DELTA     66 us LATE   2610 us 16384 samples
PROCESSED RX SLICE 0 /15: res 0 TS:  131072       1943 us DELTA    117 us LATE   2623 us 16384 samples
RX STAT Overruns:0
RX_TO_FILE: buffer 0 written
RX_TO_FILE: thread exit: 1:1
Success!
Processed 1 devs, each: RX 2 x 3.320 = 6.641 MSPS (WIRE: 26.562143)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s) 

$ strings captured.bin
4c9c20d84937e0293fe598f44d785bce
5bb170970da4a03f1c53b6f0202148ef
browser-plugin-freshplayer-libpdf
0.3.5-1ubuntu7
89e57b37bb65a92b203e4b812489d0e7
browser-plugin-freshplayer-nacl
2ff74a7db1bb4fef979f63e63ec15efa
browser-plugin-freshplayer-pepperflash
efb8cdb83d2a1db04174dfb14143ebd9
bsdgames-nonfree
2.17-7
a0df62b17337e5cbd12de06fb52cd546
bugsx
1.08-12
multiverse/math
633d9d97dfd46cbccb5247af21f63dec
cbedic
kbedic
56a857f9100ac90cb66b2b50caddba94
b3fc6a06f6988440348dbafee80b3cee
chaplin
1.10-0.2ubuntu3
4b291946f818007dc5a07f07b10f6fbb
cicero
mbrola-fr4
fc49eb778e57e68629157bb32fd19f99
2.48.0-0ubuntu1
7d2f73ec99fe6ac800286af990c46c9d
302-5.3
9281d7f0d597f7492962b9135314d699
cl-sql-oracle
e2fa54c009744a1533f5486a15f863e6
cl-umlisp
1:2007ac.2-6
862801106fe3fce0392e1fbafafe4f06
cl-umlisp-orf
56177717a37fb571fe11ae0e6d490b77
981097f297a723afe41afec3b96d5a27
2.1+lgpl-6
a79728f91dcfed21b62f89dea1743319
1.53-1
mapleview
smile
fdd3e34e11f67c13bdf614925d3877ff
conserver-client
8.2.1-1
conserver
b9cc2e709d8b33dfd2ec4626d89d6cb5
conserver-server
e3162cc0887da73106fabc4b43ea815f
2012.06.27-4
432ddc1a98a461eff7937ebd9dac8fe9
2007.03.22-1
multiverse/tex

sergforce, 2019-01-31:
It looks like the problem is here
```
[ 639.604204] xtrx: buf[0]=8c010000 [virt 2befb0ad] => 8c010fff
[ 639.604258] xtrx: buf[1]=8c020000 [virt 021461be] => 8c020fff
[ 639.604298] xtrx: buf[2]=8c030000 [virt ca0ea717] => 8c030fff
```
The virtual addresses on DMA buffers are junk. However there's only one place they're assigned
```
pbufs[i].virt = pci_alloc_consistent(d->pdev, buflen, &pbufs[i].phys);
```
If you have another kernel it'd be worth trying. Unfortunatly I don't have  imx6 to test this kernel

Can you also do `lspci -vvv` for xtrx and `cat /proc/interrupts` right after the test.

sergforce, 2019-03-20:
I found the same problem on RK3399 chip. Fix for RK3399 was pushed already. 

Can you update the driver repo `xtrx_linux_pcie_drv` apply this patch and retest?
```diff
diff --git a/xtrx.c b/xtrx.c
index c9ac97b..6e2c626 100644
--- a/xtrx.c
+++ b/xtrx.c
@@ -177,9 +177,9 @@ MODULE_VERSION("0.1");
  * mmaped to userspce. Convertion DMA->PA->VA does the trick on that
  * platforms
  */
-#ifdef CONFIG_CPU_RK3399
+//#ifdef CONFIG_CPU_RK3399
 #define VA_DMA_ADDR_FIXUP
-#endif
+//#endif
 
 
 static void xtrx_writel(struct xtrx_dev *dev, unsigned int off, unsigned int value)
```