xtrx-sdr/images #42: TX Power seems very low

b3sholt, 2019-02-14:
Hello, I am still making my way through trying to use the XTRX with some of our existing flowgraphs in GnuRadio.  One of my latest finds is that the TX power seems very low on the XTRX.  On paper, the XTRX is spec'd to 0-10 dBm output (depending of frequency).  I am comparing it against a b205-mini which is spec'd at 10 dBm minimum output power.  But, when I receive on the same device (with the exact same settings) and transmit using the exact same flowgraph with the exception of the hardware sink (xtrx with gain set to 0 and b205 with gain set to 62), I am getting drastically different signals in terms of power.  I have attached two screenshots to show the b205-mini and the XTRX output.  I am receiving on another b205-mini.  Any thoughts on why the power output of the XTRX seems to be much lower that what would be spec'd on paper?  Thanks for your help!

---
## b205 Transmit
[image]
---
## XTRX Transmit
[image]

sergforce, 2019-02-14:
Can you share your GRC file?

b3sholt, 2019-02-14:
I can share some files.  I just re-ran the most basic example and have attached the plots and the .grc files used for TX and RX.

A few notes:
* There was 60 dB of physical attenuation added when transmitting from the b205
* There was only 30 dB of physical attenuation with the XTRX (couldn't see the tone with 60 dB)

---
## b205 w/60 dB attenuation
[image]
* Gain was set to 62 on the TX for b205; set to 0 for the XTRX
---
## XTRX w/30 dB attenuation
[image]
---
## GRC files for TX/RX

[xtrx_test_grc.zip](https://github.com/xtrx-sdr/images/files/2866141/xtrx_test_grc.zip)

b3sholt, 2019-02-28:
Hi @sergforce, just checking to see if you have had a chance to test this at all.  Thanks for your help!

b3sholt, 2019-03-07:
I have an update:  If I change frequencies after the XTRX is initialized, then the power jumps (assuming the amplifier gets engaged at this point??).  No idea why this is necessary.  Now, I am actually outputting more power from the XTRX than the b205m.  

I simply threw in a call to in the flowgraph .py file in main() to set_txFreq(<desired_freq>).  So for now, I have a workaround but definitely still looking forward to a proper fix if anyone finds one.