xtrx-sdr/images #44: xtrx FPGA source code

mazsi-on-xtrx, 2019-03-28:
hi,

so this might be the right place to ask: where can the xtrx FPGA source code be downloaded from?

thanks,

GNUtoo, 2019-06-24:
I hope it gets published soon.

https://www.crowdsupply.com/fairwaves/xtrx states:
<pre>
XTRX’s main FPGA code is open source and without a viral license, so not only can you modify the code, but you can also develop your own proprietary FPGA blocks. The FPGA is approximately 30% utilized. We will share a detailed utilization report in a future update. You can upload your own firmware with our USB 3 adapter board or with a JTAG cable and our PCIe adapter board. If you are good at soldering, you can even solder JTAG directly to the XTRX board – that’s how we programmed our first samples.
</pre>

As far as I understood, the source also have very few dependencies on Xilinx proprietary FPGA blocks, so it will probably be way easier to use that with a free software FPGA toolchain like project trellis (https://github.com/SymbiFlow/prjtrellis) once the source code gets published.

PalantirWielder, 2019-08-21:
I purposely upgraded to the XTRX Pro for the expanded FPGA resources that I could utilize. However without the FPGA source it'll be a bit difficult to do anything custom with the FPGA while retaining the SDR functionality. I would appreciate having access to the source too.

yahoo2016, 2019-08-21:
At minimum, FPGA pin assignment or constraint files should be provided.  We have planned using XTRXs in our projects and contacted Fairwaves multiple times and did not get response for my requests.  Fairwaves will loss many potential users/orders without releasing FPGA information as promised.

mazsi-on-xtrx, 2019-08-22:
i reverse engineered many pins, and got support from the the folks at fairwaves with the rest.

if you know what you are doing, then check out my xtrx-system-test repo: it has a minimal system to get you started, including code to bring up the PMICs -- no LMS interface. the pinout is in system.xdc.

it comes with no warranty of any kind - don't kill your device.

chemeris, 2019-08-22:
Just to update everyone - please contact us with the Contact Us form (https://xtrx.io/contact/) if you need source code access. We share the source code with selected customers under NDA once in a while on an "as is" basis, i.e. with no clean-up and no free support.

Once the code is cleaned up, we will publish it in open-source. It's just a matter of our resources to tidy everything up.

Again, apologies that it's taking a while.

FlashPackets, 2019-08-22:
Alex et.al:
Call me old school but having access to the FPGA source would certainly make everything a bit more "readable". 
I am more than happy to protect the proprietary nature of any code you are willing to provide. 

For myself,  and I would assume others,  a full understanding of PCIe has always been somewhat of a mystery. My original purchase of my XTRX board was based on the expectation that the radio and FPGA code could provide some light on the magic.

Regards,

Chuck Glasser

RobertK66, 2019-08-23:
I am old school and if I read 
> .. a matter of our resources to tidy everything up ..

my business experience translates this to: "Its not our current prio and its very likely that it will never happen."
I do back projects like this, because of the Open Source Spirit and the possibilities of learning and having fun and personal challenges in understanding the professional solution of others (e.g. '@FlashPackets full understanding of PCIe'). But considering that your documentation (cited by @GNUtoo ) says
> ... code  **is** open source  ...

you are not living up to your promises. Sorry for being that blunt. Just my 2 ¢.
--- end of rant ---

Far from 'really' needing it for myself, I am still looking forward to see your FPGA sources some when 😃

Robert

chemeris, 2019-09-27:
As mentioned [above](#issuecomment-523964183), the only issue preventing us from publishing the code publicly is that we need to do some cleanup. If there are volunteers who could help us with this - it would definitely speed the process up.

PalantirWielder, 2019-09-27:
Could you elaborate on what sort of cleanup tasks are needed?

yahoo2016, 2019-09-28:
I believe the first step could be releasing constraint files.  Advanced users such as @mazsi-on-xtrx could use XTRX creatively with constraint files.  There are open source PCIe IPs and drivers from Xilinx.  Open source LMS IPs are also available. 
As pointed out by @mazsi-on-xtrx, XTRX could be damaged if PMIC was not programmed properly.
It's very easy to brick XTRXs with modified firmware.  JTAG would be needed to reprogram flash.

PalantirWielder, 2019-11-19:
@chemeris Do you have a rough description of the "cleanup" assistance you'd be looking for from volunteers?

GNUtoo, 2020-04-26:
Publishing source code isn't hard.

The XTRX campaign never promised to have good quality source code for the FPGA code, or to offer free support on the source code, or even to accept patches for it.

Given that:
- Resources were highly optimized to improve bandwidth and minimize FPGA resources consumption: This is probably very hard to do in a way that keeps the code readable. It took several attempts and many years for projects to get clean code for doing USB bitbang in microcontrollers. Firmwares for DSP or specific CPU are probably most of the time very optimized but hard to read.
- "board bringup" source code is typically of bad quality.
- Vendors don't have time to cleanup the board bringup code.
- The politico economical system is incentivizing companies making hardware products to write source code quick and forget about it.

So it's to be expected that the source code quality is bad, and there is nothing wrong with that.
Many vendors do it this way.
Several drivers in Linux were even reverse engineered from bad quality source code like the wl1251 driver or made from cleaned up bad quality source code that lands in the staging area of Linux.

Making hardware is also hard and about half of the crowdfunding project fail because of that.

Even if Fairwaves has a good track record of making hardware that works, the complexity of doing a project like XTRX also meant that the risk of having hardware that didn't work was still present.

However crowdfunding project don't mean that you can take the money and buy some sport cars with it for instance: the projects are supposed to at least try to meet the goals set by the crowdfunding campaign.

In contract law it would correspond to having no guarantee of success but having some guarantee of putting necessary means of achieving the goal.

And here publishing source code isn't hard.

All you have to do is write a README telling that you don't give any free support on it.
I can even take care of that for you if you want.

As I understand, promising to publish the source code and not doing it would be FLOSS washing (like greenwashing but for free and open source software) and probably constitute misrepresentation according to the contract laws of many countries around the world.

Denis.

RobertK66, 2020-10-30:
Isn't this: https://github.com/xtrx-sdr/fpga-source the thing we waited for !?
So somebody could close this issue here ;-)  ....

chemeris, 2020-10-30:
Yes, we've pushed the FPGA source code to a public repo under an open-source license, finally. Thank you everyone for your patience.

closed 2020-10-30 by chemeris