xtrx-sdr/images #45: xtrx: module verification failed, tainted kernel

b3sholt, 2019-04-03:
Hello, 
I know there is already closed issue on this topic, but it was closed and I have different parameters.

---

### System:
Lex 2I610HW with i7
CentOS 7.6.1810

---

I can successfully run the xtrx using this same board and setup, with the only difference being OS.  When the xtrx works it is running Ubuntu 16.04.  I am now running CentOS 7 and am getting the below error after building/installing the dkms xtrx module.

dmesg output:
xtrx: loading out-of-tree module taints kernel.
xtrx: module verification failed: signature and/or required key missing - tainting kernel

lscpi does not show the xtrx.

Thanks for your help/suggestions.

jeffcandido, 2020-01-22:
Can you help me to solve this issue?
After installation of gr-osmosdr ( https://github.com/xtrx-sdr/gr-osmosdr ) the dmesg command gives me this output >>

xtrx: loading out-of-tree module taints kernel.
xtrx: module verification failed: signature and/or required key missing - tainting kernel

How can I recover my XTRX?
Before this installation, my device was being recognized (dmesg and lspci commands).

b3sholt, 2020-12-18:
Just tried again on Ubuntu 20.04 and following the instructions in xtrx-sdr/images and seeing the same "loading out-of-tree module taints kernel" error.  Does anyone have a solution to this?

earwin, 2020-12-18:
"loading out-of-tree module taints kernel" is not an error, it is a notification that will be present in the logs even when everything is working normally.
The taint flag (there are multiple of them in fact) was introduced so kernel developers and other community members don't have to waste their time debugging kernel issues that could be affected by code they have no source access to.
Google this message for more information.

Something else is going on, and there's not enough information to venture any guesses.

b3sholt, 2020-12-18:
Thanks @earwin!  Noted on the kernel tainting notification (rather than error).  Currently I am able to build the kernel module error free, but the xtrx is not showing in lspci output after kernel module has been added using modprobe.  I am still using the Lex 2I610HW board running Ubuntu 20.04.  Any help/suggestions are more than welcomed!

earwin, 2020-12-18:
xtrx should be visible on `lspci` even without the driver (as in https://github.com/xtrx-sdr/images#working-with-xtrx-over-pcie-bus).
If it's not, there's some physical problem with either host motherboard, or xtrx, or their connection (e.g. https://github.com/xtrx-sdr/images#i-dont-see-xtrx-in-lspci-output)
Your Lex board has an SMB issue as described in https://github.com/xtrx-sdr/images#xtrx-installed-into-a-minipcie-slot-prevents-the-system-from-booting-dont-even-get-to-bios, so it should probably not boot at all? If you applied the fix (by cutting the fuse or masking the pci connector SMB dataline with tape), are you sure you didn't cut too hard, or didn't mask wrong lines?

b3sholt, 2020-12-18:
earwin, I applied the fix to the xtrx and have used this xtrx and Lex board 1.5 years ago (they have essentially been in a box since then).  Back then, I was running Ubuntu 16.04 and I was able to use the xtrx without issue. Currently, the OS boots fine, but the xtrx just does the ON-OFF-ON-OFF pattern and appears to have no comms with the board.  Both the board and the xtrx should be in the same state as they were 1.5 years ago...main difference now being the OS (Ubuntu 20.04 vs Ubuntu 16.04).

b3sholt, 2020-12-21:
Closing this, for me it turned out to be a hardware issue.  I tried the same OS and xtrx on a second lex board, and everything is working now.

closed 2020-12-21 by b3sholt

earwin, 2020-12-21:
Good, thanks.
It's interesting though, Lex boards were pretty reliable for us.