xtrx-sdr/images #46: Can't compile on ArchLinux

teon, 2019-04-15:
Doesn't matter if i compile from images/sources or just the lib itself I have a compilation error:

```
[ 59%] Building C object libxtrxll/CMakeFiles/xtrxll.dir/xtrxll_log.c.o
/home/teon/x/images/sources/libxtrxll/xtrxll_base.c:832:12: warning: ‘PPS_CFG_TIME_PPS_EDGE’ defined but not used [-Wunused-const-variable=]
 localparam PPS_CFG_TIME_PPS_EDGE = 3;
            ^~~~~~~~~~~~~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_base.c:830:12: warning: ‘PPS_CFG_OSC_PPS_EDGE’ defined but not used [-Wunused-const-variable=]
 localparam PPS_CFG_OSC_PPS_EDGE  = 1;
            ^~~~~~~~~~~~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:76:12: warning: ‘ESPI_RDID_0’ defined but not used [-Wunused-const-variable=]
 localparam ESPI_RDID_0  = 0x9E;
            ^~~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:73:12: warning: ‘QIO_RST’ defined but not used [-Wunused-const-variable=]
 localparam QIO_RST      = 0x99;  //    0         0          0               0
            ^~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:72:12: warning: ‘QIO_RSTEN’ defined but not used [-Wunused-const-variable=]
 localparam QIO_RSTEN    = 0x66;  //    0         0          0               0
            ^~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:70:12: warning: ‘QIO_RDREAR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_RDREAR   = 0xC8;  //    0         0          0               0
            ^~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:69:12: warning: ‘QIO_WREAR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WREAR    = 0xC5;  //    0         0          0               0
            ^~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:68:12: warning: ‘QIO_EX4BYTEA’ defined but not used [-Wunused-const-variable=]
 localparam QIO_EX4BYTEA = 0xE9;  //    0         0          0               0
            ^~~~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:67:12: warning: ‘QIO_EN4BYTEA’ defined but not used [-Wunused-const-variable=]
 localparam QIO_EN4BYTEA = 0xB7;  //    0         0          0               0
            ^~~~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:65:12: warning: ‘QIO_WRECR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WRECR    = 0x61;  //    0         0          0               1
            ^~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:63:12: warning: ‘QIO_WRVCR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WRVCR    = 0x81;  //    0         0          0               1
            ^~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:61:12: warning: ‘QIO_CLFSR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_CLFSR    = 0x50;  //    0         0          0               2
            ^~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:60:12: warning: ‘QIO_RFSR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_RFSR     = 0x70;  //    0         0          2               0
            ^~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:59:12: warning: ‘QIO_WRLR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WRLR     = 0xE5;  //    3/4       0          0               1
            ^~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:58:12: warning: ‘QIO_RDLR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_RDLR     = 0xE8;  //    3/4       0          1+              0
            ^~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:57:12: warning: ‘QIO_WRSR’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WRSR     = 0x01;  //    0         0          0               1
            ^~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:54:12: warning: ‘QIO_PES’ defined but not used [-Wunused-const-variable=]
 localparam QIO_PES      = 0x75;  //    0         0          0               0
            ^~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:53:12: warning: ‘QIO_PER’ defined but not used [-Wunused-const-variable=]
 localparam QIO_PER      = 0x7A;  //    0         0          0               0
            ^~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:48:12: warning: ‘QIO_POTP’ defined but not used [-Wunused-const-variable=]
 localparam QIO_POTP     = 0x22;  //    3/4       0          0               1-65
            ^~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:47:12: warning: ‘QIO_QCPP_2’ defined but not used [-Wunused-const-variable=]
 localparam QIO_QCPP_2   = 0x12;  //    3/4       0          0               1-256
            ^~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:46:12: warning: ‘QIO_QCPP_1’ defined but not used [-Wunused-const-variable=]
 localparam QIO_QCPP_1   = 0x32;  //    3/4       0          0               1-256
            ^~~~~~~~~~
/home/teon/x/images/sources/libxtrxll/xtrxll_flash.c:42:12: warning: ‘QIO_WRDI’ defined but not used [-Wunused-const-variable=]
 localparam QIO_WRDI     = 0x04;  //    0         0          0               0
            ^~~~~~~~
[... 71 lines cut ...]
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq16_sc32_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_ic16i_iq16_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq16_sc32i_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_sc32_conv64_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_sc32i_iq16_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq12_sc32_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_sc32_iq16_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_sc32_conv64_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_ic8i_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_sc32_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_sc32i_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq16_conv64_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_ic16_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_sc32i_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_ic16_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_ic8i_avx'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq12_sc32_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_sc32_conv64_avx_fma'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_iq8_ic16i_sse2'
/usr/bin/ld: ../libxtrxdsp.so.0.0.1: undefined reference to `xtrxdsp_ic16i_iq16_avx'
collect2: error: ld returned 1 exit status
make[2]: *** [libxtrxdsp/tests/CMakeFiles/test_filter.dir/build.make:85: libxtrxdsp/tests/test_filter] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: libxtrxdsp/tests/CMakeFiles/test_filter.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

sergforce, 2019-04-15:
Can you provide more details about your distribution and version of compiler?

teon, 2019-04-16:
@sergforce CMake output for your convenience: https://git.io/fjYmd

- ArchLinux current (it's rolling release)
- cmake version 3.14.2
- gcc (GCC) 8.2.1 20181127
- GNU Make 4.2.1

teon, 2019-04-18:
@sergforce - any ideas/tips/fixes? Without the drivers my XTRX is useless..

lainy, 2019-04-18:
@teon Try this PKGBUILD, I have been using it on Arch successfully:
```shell
# Maintainer: lain <lain@hacktheinter.net>
pkgname=xtrx-images-git
pkgver=r49.a39d40c
pkgrel=1
pkgdesc="XTRX FPGA images, drivers, and binaries"
provides=("${pkgname%-git}")
conflicts=("${pkgname%-git}")
arch=('i686' 'x86_64')
url='https://github.com/xtrx-sdr/images'
md5sums=('SKIP')
license=('GPL' 'LGPL')
depends=(
  'git'
  'libusb'
  'cmake'
  'dkms'
  'python'
  'python-cheetah'
  'qcustomplot-qt5'
)
makedepends=()
source=('git+https://github.com/xtrx-sdr/images.git')

pkgver() {
  cd images
  #TODO: CMakeLists.txt has MAJOR_VERSION.API_COMPAT.MINOR_VERSION.MAINT_VERSION, extract that?
  printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

build() {
  cd "images/sources"
  git submodule init
  git submodule update

  # pkgbuild dkms stuff isn't happy about the version
  # string 0.0.1-2, specifically the hyphenation.
  # let's patch it to just match $pkgver
  sed -e "s/PACKAGE_VERSION=\"[^\"]\\+\"/PACKAGE_VERSION=\"${pkgver}\"/" \
      -i 'xtrx_linux_pcie_drv/dkms.conf'
  sed -e "s/\${LIBVER}-\${MAINT_VERSION}/${pkgver}/" \
      -i 'xtrx_linux_pcie_drv/CMakeLists.txt'

  mkdir -p build

  cd build
  cmake .. \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DINSTALL_UDEV_RULES=ON \
    -DFORCE_ARCH=$(uname -m)
  make
}

package() {
  cd "images/sources/build"
  make DESTDIR="$pkgdir/" install
}
```
It might be missing dependencies, I just whipped it up so I have a package I can use on different machines. In particular I seem to recall some issues with QCustomPlot cmake files, but that may be resolved by now.

teon, 2019-04-19:
@lainy did the job! Thank you! You shoud post it to AUR.

lainy, 2019-04-24:
Looks like somebody beat me to the punch by publishing cleaner, individual packages to AUR.
The [PCIe drivers](https://aur.archlinux.org/packages/xtrx-linux-pcie-drv-git/) and [xtrx-flavoured xc3sprog](https://aur.archlinux.org/packages/xtrx-xc3sprog-git/) were missing though so I've posted both.

You will of course have to uninstall xtrx-images-git to avoid conflicts.

If there is a compelling reason to have a package for the xtrx-images repository, I'd be happy to maintain that as well.

GNUtoo, 2019-06-24:
I've also some PKGBUILDS that I've been working on which are in Parabola::
* https://git.parabola.nu/abslibre.git/tree/pcr/libxtrx/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/libxtrxdsp/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/libxtrxll/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/liblms7002m/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/soapyosmo-xtrx/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/gnuradio-osmosdr-xtrx/PKGBUILD
* https://git.parabola.nu/abslibre.git/tree/pcr/gqrx-xtrx/PKGBUILD

However there are some limitations:
* I need to package the libraries required to make the USB3 version work
* I need to package the kernel module.

Without the kernel module being packaged, I'm not supposed to push that upstream, in AUR.

chemeris, 2019-09-27:
We've added a link to the ArchLinux AUR to the README file, so I'm closing this issue.

@GNUtoo Parabola sounds like another Linux distro but I'm not sure how does it work. If you want us to add it to a list of supported distros, please respond to this ticket with a link to mention, or open a new ticket.

closed 2019-09-27 by chemeris