xtrx-sdr/images #47: XTRX missing from Gnuradio

bartewert, 2019-04-23:
Hi,

I'm having issues compiling the Gr-OsmoSDR from git.
When trying to follow the instructions to the letter:
git clone git://git.osmocom.org/gr-osmosdr
cd gr-osmosdr/
mkdir build
cd build/
cmake ../

I get this error:
"
CMake Error at CMakeLists.txt:156 (MESSAGE):
  GnuRadio version required: >="3.7.10" found: "3.7.9.1"
"

I've also tried the libxtrx and master branches.

This is the output for both:
"
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.9
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.9
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_BLOCKS: /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
[... 240 lines cut ...]
#Failed xtrx_open: -19

sudo ./test_xtrx -t -l2
#CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
#Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
#RX tunned: 900000000.000000
#RX bandwidth: 2000000.000000
#RX LNA gain: 15.000000
#RX PGA gain: 0.000000
#RX TIA gain: 9.000000
#RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
#PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4229 us DELTA      3 us LATE   2185 us 16384 #samples
#RX STAT Overruns:0
#Success!
#Processed 1 devs, each: RX 2 x 1.933 = 3.865 MSPS (WIRE: 15.461600)    TX 2 x 0.000 = 0.000 MSPS ##(WIRE: 0.000000 MB/s) 

Has anyone come across these issues ?
Any help will be much appreciated.

Operating System: Ubuntu 16.04.6 LTS
Kernel: Linux 4.15.0-47-generic
Architecture: x86-64
cmake version 3.5.1
Gcc 4:5.3.1-1ubuntu1
Gnuradio 3.7.9-2ubuntu1

o8ko8k, 2019-08-23:
cmake .. | grep FALSE  should yield 

 Dependency SWIG_FOUND = FALSE
-- Dependency SWIG_VERSION_CHECK = FALSE
 Dependency LIBOSMOSDR_FOUND = FALSE

among other things (airspy and such which i expect they should be).  I would conclude you go install SWIG and libosmosdr   

if your using ubuntu or its variants consider installing synaptic and managing your dependencies with that tool.  i found it helpful.  i believe you will need the -dev variant of SWIG and libosmosdr in order to get those items to clear.

I presume you either figured this out already or gave up. but someone may find this post helpful.

chemeris, 2019-09-27:
@o8ko8k Thanks a lot for helping answer this!
I'm closing this issue so far.

closed 2019-09-27 by chemeris