xtrx-sdr/images #52: DMA errors for Nvidia Jetson TX2 dev kit

yahoo2016, 2019-06-04:
I used Startec mPCie to PCI adapter to install XTRX on PCIe slot of Jetson TX2 Dev kit, XTRX driver was loaded after changing BUFS from 32 to 14 in "xtrx.c" according to the link:
https://devtalk.nvidia.com/default/topic/1050762/pci-express-on-jetson-nano/

test_xtrx produced error:
[   26.177972] xtrx: SIM mctrl=1
[   26.180946] xtrx: Port closed: 1
[   26.184492] xtrx:  3V3 CTRL:0
[   26.187508] xtrx: Port closed: 0
[   26.191151] xtrx:  3V3 CTRL:1
[   26.194139] xtrx: Port opened: 1
[   26.197395] xtrx: SIM mctrl=1
[   26.200699] xtrx: Port opened: 0
[   32.178880] xtrx: SIM mctrl=1
[   32.182095] xtrx: Port closed: 1
[   32.188060] xtrx:  3V3 CTRL:0
[   32.191079] xtrx: Port closed: 0
[   99.361563] xtrx: call: REGION=1 VMA=ffffffc1d04f0000 vma->vm_pgoff=0
[   99.368071] xtrx: VMA open, virt 7f85ac2000, phys 0
[   99.373083] xtrx: call: REGION=0 VMA=ffffffc1c2536b00 vma->vm_pgoff=1024
[   99.380071] xtrx: VMA open, virt 7f85ac1000, phys 400000
[   99.385600] CPU0: SError detected, daif=140, spsr=0x20000000, mpidr=80000100, esr=bf000000
[   99.385619] CPU4: SError detected, daif=140, spsr=0x20000000, mpidr=80000102, esr=bf40c000
[   99.385622] CPU3: SError detected, daif=140, spsr=0x60000045, mpidr=80000101, esr=bf40c000
[   99.385624] CPU5: SError detected, daif=140, spsr=0x80000000, mpidr=80000103, esr=bf40c000
[   99.386947] ROC:CCE Machine Check Error:
[   99.386951] 	Address Type = Non-Secure DRAM
[   99.386981] 	Address = 0x4040110068 (Unknown Device)
[   99.386983] 		Request not going to SYSRAM or DRAM
[   99.393894] CPU0: SError detected, daif=140, spsr=0x40000145, mpidr=80000100, esr=bf40c000
[   99.431478] CPU0: SError detected, daif=140, spsr=0x80000000, mpidr=80000100, esr=bf000000
[   99.431519] CPU4: SError detected, daif=1c0, spsr=0x200000c5, mpidr=80000102, esr=bf40c000
[   99.432881] ROC:CCE Machine Check Error:
[   99.432886] 	Address Type = Non-Secure DRAM
[   99.432914] 	Address = 0x4040110000 (Unknown Device)
[   99.432916] 		Request not going to SYSRAM or DRAM
[   99.452090] CPU0: SError detected, daif=140, spsr=0x60000000, mpidr=80000100, esr=bf000000
[   99.452137] CPU4: SError detected, daif=1c0, spsr=0x800000c5, mpidr=80000102, esr=bf40c000
[   99.453369] ROC:CCE Machine Check Error:
[   99.453372] 	Address Type = Non-Secure DRAM
[   99.453392] 	Address = 0x4040110000 (Unknown Device)
[   99.453393] 		Request not going to SYSRAM or DRAM
[   99.482975] xtrx: ioctl(12345b, 1) [ffffff8014b01000]
[   99.487907] xtrx:  3V3 CTRL:1
[   99.490944] CPU0: SError detected, daif=140, spsr=0x20000000, mpidr=80000100, esr=bf000000
[   99.490999] CPU4: SError detected, daif=140, spsr=0x80000000, mpidr=80000102, esr=bf40c000
[   99.492244] ROC:CCE Machine Check Error:
[   99.492248] 	Address Type = Non-Secure DRAM
[   99.492276] 	Address = 0x4040110000 (Unknown Device)
[   99.492278] 		Request not going to SYSRAM or DRAM
[   99.511601] CPU5: SError detected, daif=140, spsr=0x20000000, mpidr=80000103, esr=bf000000
[   99.511603] CPU4: SError detected, daif=140, spsr=0x80000000, mpidr=80000102, esr=bf40c000
[   99.513006] ROC:CCE Machine Check Error:
[   99.513010] 	Address Type = Non-Secure DRAM
[   99.513034] 	Address = 0x4040110000 (Unknown Device)
[   99.513036] 		Request not going to SYSRAM or DRAM

chemeris, 2019-06-04:
Just checking - did you try this?
https://github.com/xtrx-sdr/images#ive-got-a-dma-buffer-issue-on-my-arm-device

sergforce, 2019-06-04:
You don't need to modify BUFS from 32 to 14 in "xtrx.c". Revert it back and check coherent pool size

yahoo2016, 2019-06-05:
Thanks for quick responses.
I cloned XTRX source using another TX2 rebuilt driver without changing driver source and only added "pci=noaer vmalloc=512M cma=64M coherent_pool=32M" to "extlinux.conf".  Driver loaded without error.
When I ran "test_xtrx -t -l2" I got 
$ ./test_xtrx -t -l2
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)

The program hang and "dmesg" showed 100's lines of 
"CPU4: SError detected, daif=140, spsr=0x80000000, mpidr=80000102, esr=bf000002"

I'm wandering if I need to change driver source at all, e.g., adding "V_DMA_ADDR_FIXUP"

yahoo2016, 2019-06-05:
Adding "#define VA_DMA_ADDR_FIXUP" to "xtrx.c" did not make difference.

sergforce, 2019-06-05:
Is there any diferrence in output with VA fix and without. Could you post
the full dmesg error?

yahoo2016, 2019-06-05:
sergforce,
I'm glad we can get prompt support for an open source project.  
The following is message from "lspci". I'll post dmesg before and after "test_xtrx".

nvidia@tegra-ubuntu:~$ sudo lspci -vv -d 10ee:
[sudo] password for nvidia: 
01:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 453
	Region 0: Memory at 50110000 (32-bit, non-prefetchable) [size=4K]
	Region 1: Memory at 50100000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [48] MSI: Enable+ Count=1/16 Maskable- 64bit+
		Address: 000000026082e000  Data: 0000
	Capabilities: [60] Express (v2) Endpoint, MSI 01
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 4096 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x2, ASPM L0s, Exit Latency L0s unlimited, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range B, TimeoutDis-, LTR-, OBFF Not Supported
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Device Serial Number 00-00-00-00-12-34-56-78
	Kernel driver in use: xtrx
	Kernel modules: xtrx

yahoo2016, 2019-06-05:
dmesg before and after "test_xtrx" are attached.
[xtrx1.txt](https://github.com/xtrx-sdr/images/files/3256658/xtrx1.txt)
[xtrx2.txt](https://github.com/xtrx-sdr/images/files/3256659/xtrx2.txt)

yahoo2016, 2019-06-05:
Some progresses were made.  I found there is a newer driver under xtrx/xtrx_linux_pcie_drv.
I was modifying the older driver under xtrx/images/sources/xtrx_linux_pcie_drv.
I cloned the newer driver and commanded out "#ifdef CONFIG_CPU_RK3399" to use "#define VA_DMA_ADDR_FIXUP" and other updated for ARM CPUs.
I got the folloing output without error:
" ........
Success!
Processed 1 devs, each: RX 2 x 1.861 = 3.723 MSPS (WIRE: 14.891405)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s) ".

I'm glad I could have XTRX working in one day with prompt support/help from Fairview and other users.
I'm closing this issue and open another one for throughputs.

yahoo2016, 2019-06-05:
I got a new issue:
When I ran "./test_xtrx -T", my TX2 completely locked.

sergforce, 2019-06-05:
I think there's something wrong anyway..
Try to reboot the system and start with -T on clean system. Try to do
./test_xtrx -T -l9 to see where it gets locked.

yahoo2016, 2019-06-05:
I'll try "./test_xtrx -T -l9" but I'm getting another issue for long run of RX DMAs, I was running:
./test_xtrx -t -N2000000000
but after 2 minutes, the test program quits with "Segmentation fault (core dumped)" error:

PROCESSED RX SLICE 0 /56533: res 0 TS:463126528       1911 us DELTA     53 us LATE   2888 us 16384 samples
Segmentation fault (core dumped)

yahoo2016, 2019-06-05:
Since TX2 was completely locked, I had to take a photo (attached) for outputs from "test_xtrx -T -l9".
TX is less important for our applications, I'll continue RX throughput tests. mtp://[usb:001,007]/Phone/DCIM/Camera/20190605_092450.jpg

yahoo2016, 2019-06-05:
/home/nvidia/Desktop/20190605_092450.jpg

yahoo2016, 2019-06-05:
Attaching jpg file doesn't seem to work.  I need to find anther way to attached log from "test_xtrx -T -l9"

yahoo2016, 2019-06-05:
The memory leak was the following code in "test_xtrx.c"  
			for (unsigned bc = 0; bc < buf_cnt; bc++) {
				stream_buffers[bc] += ri.samples * sdata->sample_host_size;
			}

After commenting out the above, I was able to make a long run of RC DMA.

yahoo2016, 2019-06-06:
RX DMA seems working.

closed 2019-06-06 by yahoo2016