xtrx-sdr/images #61: Ubuntu 18.x prebuilt binarys not working

zehm, 2019-07-02:
I've been trying to use my xtrx without having to custom-compile everything for some time now. I was excited to see the Ubuntu 18.x binaries show up, and I was able to get them all installed (after a `--force-overwrite` on `libgnuradio-osmosdr0.1.5git_0.1.5git-2_amd64.deb`). However, when I try to attach to xtrx via gqrx I get this:

```
gr-osmosdr v0.1.4-107-g7c59782a (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya freesrp xtrx 
[INFO] Make connection: '/dev/xtrx0'
gqrx: symbol lookup error: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so: undefined symbol: xtrx_log_setlevel
```
Any ideas on this? Maybe the binaries don't have the right version of one of the libs?

satunnainen, 2019-07-03:
Could you please provide the output of following commands?

`dpkg -l | grep xtrx`
`objdump -T /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so | grep xtrx`
`dpkg -S /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so`

Are you able to specify the details of the error, as a result of which you had to use the `--force-overwrite` key?

zehm, 2019-07-03:
For sure, here you go. Thanks for taking a look :)

Here's the output:

```
$ dpkg -l | grep xtrx
ii  gr-osmosdr-xtrx                                             0.1.5git-2                                                                  amd64        Gnuradio blocks from the OsmoSDR project
ii  gr-osmosdr-xtrx-dbgsym                                      0.1.5git-2                                                                  amd64        debug symbols for gr-osmosdr-xtrx
ii  libxtrx:amd64                                               0.0.1-5                                                                     amd64        High level XTRX API library
ii  libxtrx-dbgsym:amd64                                        0.0.1-5                                                                     amd64        debug symbols for libxtrx
ii  libxtrx-dev:amd64                                           0.0.1-5                                                                     amd64        High level XTRX API library
ii  libxtrxdsp:amd64                                            0.0.1-2                                                                     amd64        DSP specific function for SDR in general and XTRX in specific
ii  libxtrxdsp-dbgsym:amd64                                     0.0.1-2                                                                     amd64        debug symbols for libxtrxdsp
ii  libxtrxll:amd64                                             0.0.1-3                                                                     amd64        XTRX Low-level API library
ii  libxtrxll-dbgsym:amd64                                      0.0.1-3                                                                     amd64        debug symbols for libxtrxll
ii  soapysdr0.6-module-xtrx:amd64                               0.0.1-5                                                                     amd64        High level XTRX API library
ii  soapysdr0.6-module-xtrx-dbgsym:amd64                        0.0.1-5                                                                     amd64        debug symbols for soapysdr0.6-module-xtrx
ii  xtrx-dkms                                                   0.0.1-3                                                                     all          XTRX PCI driver for linux
ii  xtrx-examples:amd64                                         0.0.1-5                                                                     amd64        High level XTRX API library
ii  xtrx-examples-dbgsym:amd64                                  0.0.1-5                                                                     amd64        debug symbols for xtrx-examples
```

```
$ objdump -T /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so | grep xtrx
0000000000000000      DF *UND*  0000000000000000              xtrx_set_gain
0000000000000000      DF *UND*  0000000000000000              xtrx_log_setlevel
0000000000000000      DF *UND*  0000000000000000              xtrx_tune
0000000000000000      DF *UND*  0000000000000000              xtrx_open_string
0000000000000000      DF *UND*  0000000000000000              xtrx_stop
0000000000000000      DF *UND*  0000000000000000              xtrx_tune_rx_bandwidth
0000000000000000      DF *UND*  0000000000000000              xtrx_set_samplerate
0000000000000000      DF *UND*  0000000000000000              xtrx_run_ex
0000000000000000      DF *UND*  0000000000000000              xtrx_close
0000000000000000      DF *UND*  0000000000000000              xtrx_tune_tx_bandwidth
0000000000000000      DF *UND*  0000000000000000              xtrx_recv_sync_ex
0000000000000000      DF *UND*  0000000000000000              xtrx_send_sync_ex
0000000000000000      DF *UND*  0000000000000000              xtrx_discovery
0000000000000000      DF *UND*  0000000000000000              xtrx_tune_ex
0000000000000000      DF *UND*  0000000000000000              xtrx_val_get
0000000000000000      DF *UND*  0000000000000000              xtrx_set_ref_clk
0000000000000000      DF *UND*  0000000000000000              xtrx_set_antenna
000000000000a830 g    DF .text  0000000000000069  Base        _ZNK9SoapyXTRX16to_xtrx_channelsEm
000000000000a460 g    DF .text  00000000000000a8  Base        _ZN9SoapyXTRX12xtrx_logfuncEiPKc
```

```
$  dpkg -S /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so
soapysdr0.6-module-xtrx:amd64: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so
```

When I originally tried to install all those packages with `dpkg` I got this error:
```
dpkg: error processing archive libgnuradio-osmosdr0.1.5git_0.1.5git-2_amd64.deb (--install):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.1.5git', which is also in package libgnuradio-osmosdr0.1.4:amd64 0.1.4.98-myriadrf1~bionic
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

So apt didn't realize it was supposed to "upgrade" `libgnuradio-osmosdr0.1.4:amd64 0.1.4.98-myriadrf1~bionic` with the package in your repo. It wouldn't let me uninstall the old package because of dependancies, so to "fix" it I just ran:
```
sudo dpkg --force-overwrite -i libgnuradio-osmosdr0.1.5git_0.1.5git-2_amd64.ddeb
```
which got the package installed and supposedly should have overwritten the files from the old package with the version of them in your package. After that I ran an `ldconfig` to make sure the system knew about the new libs.

Brandon

satunnainen, 2019-07-04:
Please try to remove `libgnuradio-osmosdr0.1.4` package and repeat your attempt. If that wouldn't solve the issue, please provide output of `sudo /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx -t -l6` and `LD_DEBUG=all gqrx` (assuming you are using `bash`) commands.

zehm, 2019-07-04:
Solved it! (mostly, more questions below). I'll document what I went through in case it helps someone else.

I went through the whole process. Removed `libgnuradio-osmosdr0.1.4` (which forced removal of quite a few other things, including `gqrx`), removed all the xtrx packages. Then I reinstalled all the xtrx 18.04 packages from your repo. Then I installed gqrx. gqrx depends on `libgnuradio-osmosdr0.1.4` and `apt` wanted to install it but I worked around the dependency issue and installed it without the older `libgnuradio-osmosdr0.1.4` package by running the install manually like this: `dpkg --force-depends -i gqrx-sdr_2.11.5-gqrx1~bionic_amd64.deb`.

After all that I tried starting `gqrx` and connecting to xtrx, but I got the exact same error I did previously:
```
gr-osmosdr v0.1.4-107-g7c59782a (0.1.5git) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya freesrp xtrx 
[INFO] Make connection: '/dev/xtrx0'
gqrx: symbol lookup error: /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libXTRXSupport.so: undefined symbol: xtrx_log_setlevel
```

Then the `test_xtrx` command you gave me errors out too, odd:
```
$ sudo /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx -t -l6
/usr/lib/x86_64-linux-gnu/xtrx/test_xtrx: symbol lookup error: /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx: undefined symbol: xtrx_gtime_op
```

However the one from the wiki seems to work okay:
```
$ test_xtrx -t -l2
12:07:26.297469 ERROR:   USB3380: Unable to open USB3380!
12:07:26.297598 ERROR:   Unable to allocate USB3380 context: error: -6                                                                                                            
CPU Features: SSE2+ SSE4.1+ AVX+ FMA+
12:07:26.300641 WARNING: NLMS7 PCI_3: LMS VER:7 REV:1 MASK:1 (3841)
12:07:26.347018 WARNING: NLMS7 PCI_3: CGEN: VCO/2=1184000000 k/2=37 int=91 frac=80659                                                                                             
12:07:26.347479 WARNING: NLMS7 PCI_3: CGEN: binary result: 159                                                                                                                    
12:07:26.347749 WARNING: NLMS7 PCI_3: CGEN: Retuned [160:167] -> 163                                                                                                              
Master: 32000000.000000; RX rate: 4000000.000000; TX rate: 0.000000
12:07:26.347983 WARNING: NLMS7 PCI_3: SXR: initial VCO=3600000000 DIVH=2 VCOs:0010
12:07:26.347989 WARNING: NLMS7 PCI_3: SXX: VCO2 N=138 frac=483958
12:07:26.348207 WARNING: NLMS7 PCI_3: SXX: binary result: 255
12:07:26.348238 WARNING: NLMS7 PCI_3: SXR: VCO2 [0;0] -> 0
12:07:26.349319 WARNING: NLMS7 PCI_3: SXX: VCO2 N=138 frac=483958
12:07:26.349571 WARNING: NLMS7 PCI_3: SXX: binary result: 145
12:07:26.350245 WARNING: NLMS7 PCI_3: SXR: VCO2 [145;168] -> 156
12:07:26.350250 WARNING: NLMS7 PCI_3: SXX: VCO2 N=138 frac=483958
12:07:26.350469 WARNING: NLMS7 PCI_3: SXX: binary result: 153
12:07:26.350956 WARNING: NLMS7 PCI_3: SXR: VCO2 [153;170] -> 161
RX tunned: 900000000.000000
12:07:26.351049 WARNING: NLMS7 PCI_3: 0x0124[18, 00]
12:07:26.351144 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
12:07:26.351250 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
12:07:26.351342 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
RX bandwidth: 5000000.000000
12:07:26.351454 WARNING: NLMS7 PCI_3: RFE: set_lna(15 -> 15) => 6
RX LNA gain: 15.000000
12:07:26.351494 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
12:07:26.351575 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
12:07:26.351668 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
12:07:26.351748 WARNING: NLMS7 PCI_3: 0x0124[18, 08]
12:07:26.351972 WARNING: NLMS7 PCI_3: 0x0124[1c, 0c]
12:07:26.352177 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
12:07:26.352284 WARNING: NLMS7 PCI_3: TIA: CCOMP=3 CFB=326 RCOMP=9
RX SAMPLES=524288 SLICE=524288 PARTS=1
PROCESSED RX SLICE 0 /0: res 0 TS:       0      65549 us DELTA      5 us LATE     19 us 524288 samples R=    ffdffcff7ff9 DELTA=740734073
XXXXX Overruns:0 Zeros:0
xAND=[0x0000,0x0000,0x0000,0x0000] xOR=[0xffff,0xffff,0xffff,0xffff]
12:07:26.419560 WARNING: NLMS7 PCI_3: 0x0124[1c, 08]
12:07:26.419669 WARNING: NLMS7 PCI_3: 0x0124[1c, 08]
Success!
Processed RX 2 x 3.998 = 7.996 MSPS (WIRE: 31.983238)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s)
```

Then it hit me - the text_xtrx command you gave me, and the one I ran must be different! And they were. I had previously tried compiling from source (a much older version from last year), and those libs were sitting in `/usr/local/lib/`. I manually cleaned up all the old files in /usr/local/, and amazingly, now that test_xtrx command you gave me works!
```
$ sudo /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx -t -l6
12:26:24.891343 DEBUG:  xtrxllpciev0_discovery:264 [PCIE] pcie: Found `pcie:///dev/xtrx0`
12:26:24.891424 REGS:   internal_xtrxll_reg_in:156 [PCIE] PCI:/dev/xtrx0: Read  [001a] = 04000013
12:26:24.891432 INFO:   xtrxll_base_dev_init:1133 [CTRL] PCI:/dev/xtrx0: XTRX Rev4 (04000013)
[... 1824 lines cut ...]
Actual bandwidth   : 5e+06 Hz
setFreqCorr : 0 ppm
New LNB LO: 0 Hz
12:38:46.325735 INFO:   [LMSF] PCI:/dev/xtrx0: FE_FREQ rx=1 type=0 freq=30000000.000000 ch=3
12:38:46.325808 INFO:   [LSM7] PCI:/dev/xtrx0: SXR: initial VCO=1920000000 DIVH=6 VCOs:1000
12:38:46.325814 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: VCO0 N=73 frac=887256
12:38:46.326029 INFO:   [LSM7] PCI:/dev/xtrx0: SXX: binary result: 255
[... cut ...]
  what():  SoapyXTRX::setFrequency(BB) unable to tune!
Aborted
```

I'm out of time for now, but at least I figured out the reason for all the confusion around unknown symbols in the xtrx libs haha. I feel silly for not having realized that earlier, but thanks for asking the questions that led me to solving it.

If you have any insight on why `gqrx` still doesn't seem to be able to use the xtrx that would be awesome. It's the main SDR util I use for general spectrum browsing and it would be great to get it working.

Brandon

zehm, 2019-07-07:
Marking as "resolved", but would still love some feedback about the issue with gqrx. Or can anyone else confirm that `xtrx` can work with the pre-build 18.04 binaries and `gqrx` from the myriadrf repo? If I don't here back here in a few days I'll open a new issue and re-post.

Thanks for your help @satunnainen!

closed 2019-07-07 by zehm