xtrx-sdr/images #62: 120 MSPS SISO / 90 MSPS MIMO

markavr, 2019-07-03:
Hi!

Are "120 MSPS SISO / 90 MSPS MIMO" really supported samplerates? If not, what values are really supported?

I tested various samplerates using xtrx_fft application and gnuradio-companion. Applications works well without glitches with samplerates up to 70 MSPS.

My PC configuration:
motherboard: GA-H77M-D3H
CPU: Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz
RAM: 8GB
PCIe: x2 Gen 2.0

markavr, 2019-07-05:
Yesterday I tested MIMO mode. For my PC maximum samplerate in MIMO mode was 2 x 50 MSPS, i.e. total samplerate exceed declared 90 MSPS. Why SISO mode is not working with samplerate higher than 70 MSPS?

rmcleitao, 2019-07-05:
I have exactly the same problem. 
After 70 MSPS there is only noise, but I didn't tested enough to open an issue. Since you are reporting the same results I'm also interested to know the solution.

markavr, 2019-07-05:
Today I tested xtrx again and get the same issues on:
1) Supermicro + Intel(R) Xeon(R) CPU E5-2630 v2 @ 2.60GHz 2.60 GHz (2);
2) Kino + Intel Core i7.

yahoo2016, 2019-08-05:
I have the same issue. The glitches start after 70MSPS.  I suspect it's FPGA firmware issue, e.g., timing issues for >70MSPS within FPGA.
Fairwaves has not released FPGA source code and it's difficult to troubleshoot without FPGA source code.  Xilinx Vivodo Analyzer could be added to FPGA for diagnostics if source code is available.

yahoo2016, 2019-08-06:
It appears increasing vio to 2700~3300 solved the issue.  Example of setting Vio is in test_xtrx.c.

PalantirWielder, 2019-08-06:
Not sure if it's of any help, but this [OsmoDevCon 2018 document (pg 6)](https://pretalx.sysmocom.de/media/OsmoDevCon2018_XTRX_status_update.pdf) seems to confirm that VIO needs to be adjusted for higher sampling rates.

yahoo2016, 2019-08-06:
Very nice presentation for XTRX design.  VIOs are voltage supplies for Input/Output pins of Integrated Circuits (ICs). I'm surprised that they are adjustable for a small boards such as XTRX.

PalantirWielder, 2019-08-06:
Hmmm... if it's only for the Inputs/Output pins then perhaps the higher voltage are necessary to compensate for the slew rates of the Input+Output combinations. The faster the data rate the more the slew rate dominates the digital waveform 🤔

yahoo2016, 2019-08-07:
Artix FPGA supports >800MHz DDR RAM, but as pointed out in the OsmoDevCon2018_XTRX_status_update.pdf,  the LML is CMOS bus, hard for high rate (DDR RAMs are SSTL).
Phases between DDR data and clock need to be calibrated to get sampling rate >100MSPS within certain temperature range.

rmcleitao, 2019-08-20:
By changing the VIO I was able to get 95 MSPS in SISO. Did anyone get 100 MSPS? 

I tried different values for the VIO (based on the OsmoDevCon 2018 document) but after 95 MSPS the VCO can't be tune.

yahoo2016, 2019-09-05:
I was told by Fairwaves support that:
"LMS7 is CMOS only and 96MSPS requires 192Mhz DDR bus with an effective clock around 400Mhz. LVDS, HSTL, HUSL, etc would be more reliable in the case. The temperature also helps a lot.

VIO is not enough. You need to activate internal MMCM in order to recreate the clock signal inside the FPGA to reduce jitter."

i.e., new FPGA firmware would be needed to calibrate CMOS interface for very high data rates.

rmcleitao, 2019-12-12:
Does anyone have news about the new FPGA firmware or another way to achieve more than the 96 MSPS?

OhSoGood, 2020-05-03:
So, that means Fairwaves advertised 120 MSPS SISO but that has never been possible because they have never released the software for it? That's hard to hear.