xtrx-sdr/images #63: Reflashing FPGA image

markavr, 2019-07-08:
Does the test_xtrxflash utility allow to reflash FPGA firmware? If "yes", what command line options should be used?

P.S.: I assume that the problems that I encountered when working with XTRX are probably related to the fact that I do not have the latest FPGA firmware. I read flash, using "xtrx_xtrxflash -r filename" command, and compared received data with FPGA image from sources. They are very different. Probably I must start reading from nonzero offset, etc.

yahoo2016, 2019-09-05:
Flashing has risks of bricking/damaging XTRX.  I have Fairwaves PCIe adapter with JTAG as backup in case "test"xtrxflash" fails.
"sudo ./test_xtrxflash -w xtrxr4PRO_top.bin" worked for my XTRX PRO.  
Using "test_xtrxflash" is significantly faster than using Xilinx USB cable.
Fairwaves PCIe adapter is strongly recommended for high throughput (PCIeX2) and flash recovery.

chemeris, 2019-09-27:
@yahoo2016 Thanks a lot for helping your fellow XTRX user!
We've added a section about XTRX flashing to the README: https://github.com/xtrx-sdr/images/blob/master/README.md#flashing-xtrx

closed 2019-09-27 by chemeris