xtrx-sdr/images #66: gr-osmosdr xtrx unique options in the documentation.

o8ko8k, 2019-08-23:
Can we please put the valid antenna selection options as well as other xtrx unique arguments in the documentation section of the source and sink blocks (the ones specific to xtrx)  for example (the list of correct values for which antenna to use):

static const std::map<std::string, xtrx_antenna_t> s_ant_map = boost::assign::map_list_of
  ("AUTO", XTRX_TX_AUTO)
  ("B1", XTRX_TX_H)
  ("B2", XTRX_TX_W)
  ("TXH", XTRX_TX_H)
  ("TXW", XTRX_TX_W)

This folks wont have to dig through the code to find the valid options AND then ill know that i have the gr-osmo block from this branch and not the generic one.  

If i could figure out how to check-out/check-in i would do it myself, but im a bit of a hack when it comes to coding so i would recommend someone else do it so they dont have to clean up my mess.

dchard, 2019-12-25:
I have the same issue. My antenna is connected to RXA, so what command line should I use?

osmocom_fft -a xtrx -A RXW -s 4000000 -f 804000000 -F

Thanks in advance!

RFNOD, 2020-12-02:
@dchard any movement on this? I am curious as well before I dive in. Thanks!!