xtrx-sdr/images #69: Compiling the PCIE driver on Jetson TK1

thehamsterjam, 2019-09-05:
I am using the Jetson TK1 and I followed the instructions for building the XTRX PCIE driver. First I tried the dkms method, which failed. Then tried just building directly which produced the output:

```
make -C /lib/modules/3.10.40-ge78bb38/build M=/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv modules
make[1]: Entering directory `/usr/src/linux-headers-3.10.40-ge78bb38'
  CC [M]  /home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.o
In file included from /usr/src/linux-headers-3.10.40-ge78bb38/arch/arm/include/asm/irq.h:7:0,
                 from /usr/src/linux-headers-3.10.40-ge78bb38/arch/arm/include/asm/hardirq.h:6,
                 from include/linux/hardirq.h:8,
                 from include/linux/interrupt.h:12,
                 from /home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.c:26:
arch/arm/mach-tegra/include/mach/irqs.h:25:24: error: expected identifier or ‘(’ before numeric constant
 #define INT_GIC_BASE   0
                        ^
arch/arm/mach-tegra/include/mach/irqs.h:575:25: note: in expansion of macro ‘INT_GIC_BASE’
 #define INT_PRI_BASE   (INT_GIC_BASE + 32)
                         ^
arch/arm/mach-tegra/include/mach/irqs.h:610:25: note: in expansion of macro ‘INT_PRI_BASE’
 #define INT_SEC_BASE   (INT_PRI_BASE + 32)
                         ^
arch/arm/mach-tegra/include/mach/irqs.h:617:21: note: in expansion of macro ‘INT_SEC_BASE’
 #define INT_I2C    (INT_SEC_BASE + 6)
                     ^
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrxll_regs.vh:235:12: note: in expansion of macro ‘INT_I2C’
 localparam INT_I2C          = 8;
            ^
arch/arm/mach-tegra/include/mach/irqs.h:610:38: error: expected ‘)’ before ‘+’ token
 #define INT_SEC_BASE   (INT_PRI_BASE + 32)
                                      ^
arch/arm/mach-tegra/include/mach/irqs.h:617:21: note: in expansion of macro ‘INT_SEC_BASE’
 #define INT_I2C    (INT_SEC_BASE + 6)
                     ^
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrxll_regs.vh:235:12: note: in expansion of macro ‘INT_I2C’
 localparam INT_I2C          = 8;
            ^
arch/arm/mach-tegra/include/mach/irqs.h:617:34: error: expected ‘)’ before ‘+’ token
 #define INT_I2C    (INT_SEC_BASE + 6)
                                  ^
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrxll_regs.vh:235:12: note: in expansion of macro ‘INT_I2C’
 localparam INT_I2C          = 8;
            ^
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.c: In function ‘xtrx_process_l_interrupts’:
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.c:695:2: warning: left shift count >= width of type [enabled by default]
  if (imask & (1 << INT_I2C)) {
  ^
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.c: In function ‘xtrx_probe’:
/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.c:1221:2: error: implicit declaration of function ‘pci_enable_msi_range’ [-Werror=implicit-function-declaration]
  err = pci_enable_msi_range(pdev, XTRX_MSI_COUNT, XTRX_MSI_COUNT);
  ^
cc1: some warnings being treated as errors
make[2]: *** [/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv/xtrx.o] Error 1
make[1]: *** [_module_/home/ubuntu/Build/images/sources/xtrx_linux_pcie_drv] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.10.40-ge78bb38'
make: *** [modules] Error 2
```

I have tried using different kernel sources and cross compiling all with the same result.

yahoo2016, 2019-09-05:
I have no issue to build XTRX driver for TX1 or TX2 natively or cross compile.  I believe the kernel source for TK1 is too old.  Kernel 4.4.x and above should work.

chemeris, 2019-09-27:
Closing this issue since we haven't heard from the original reporter for a while.

closed 2019-09-27 by chemeris