xtrx-sdr/images #76: Data format of Recieved data?

farhan296, 2020-01-21:
Can anyone help me out in understanding the placement of IQ data in the receive buffer?
Following is my configuration:
 xtrx_wire_format_t tx_wire_fmt = XTRX_WF_16;
xtrx_host_format_t tx_host_fmt = XTRX_IQ_FLOAT32;
xtrx_direction_t dir = XTRX_TRX;

xtrx_wire_format_t rx_wire_fmt = XTRX_WF_16;
xtrx_host_format_t rx_host_fmt = XTRX_IQ_FLOAT32;//XTRX_IQ_FLOAT32;
    
params.dir = dir;
params.nflags = 0;
params.rx.wfmt = rx_wire_fmt;
params.rx.hfmt = rx_host_fmt;
params.rx.chs = XTRX_CH_A;
params.rx.flags = XTRX_RSP_SISO_MODE;
params.rx.paketsize = 8192;
params.rx_stream_start = 8192;

params.tx.wfmt = tx_wire_fmt;
params.tx.hfmt = tx_host_fmt;
params.tx.chs = XTRX_CH_A;
params.tx.flags = XTRX_RSP_SISO_MODE|XTRX_RSP_TEST_SIGNAL_A;

xtrx_recv_ex_info_t ri;
        ri.samples = 2*8192;
        ri.buffer_count = 1;
        ri.buffers = (void* const*) buffs;
        ri.timeout = 1000;
        ri.flags = 0;

float buffMemIQRd[65536];

I have a float type buffer. How will the IQ data be placed based on the above configuration in SISO Mode? Is there any documentation available? I see there is a serious lack of helping materials and API documentation.
buffMemIQRd[0] =? (AI,AQ)
buffMemIQRd[1] =? (AI,AQ)
buffMemIQRd[2] =? (AI,AQ)
buffMemIQRd[3] =? (AI,AQ)

farhan296, 2020-02-05:
@sergforce Can you please clarify, how the data is placed in the receive buffer in both SISO and MIMO modes? Also I would like to know in case of Tx how to place I/Q data in the buffer i.e the position of data AI,AQ or AQ,AI etc for both MIMO and SISO. I would appreciate your help in this regard.

arun1969, 2020-10-08:
In case of Tx IQ Data AI,AQ format.
I have tried sending iq data generated by python scrip.
[https://github.com/jgibbard/iqtool](url)
using test_xtrx got success.