xtrx-sdr/images #80: OpenAirInterface support

automeas, 2020-03-19:
Hi! I've integrated XTRX into OAI with Soapy. It works but phone does not see eNB. Comparing with USRP b200 spectrum, XTRX spectrum has +30dB peak in the center (DC position). Could it be removed somehow?

Dastgsm, 2020-06-10:
Please make a tutorial if it's run correctly... so on the usrp it's could connect to the network and on the xtrx it doesn't the same effect? did you use soapyUHD ?

Dastgsm, 2020-06-10:
or the best, tell me how did you install it on xtrx? may be i have some idea to resolve it

automeas, 2020-07-21:
Yes, I did fork IRIS lib, that uses soapy.

Dastgsm, 2020-08-07:
But on usrp it's see network? may be you should decrease a Tx for a while. I think when the LMS work stressly, it's result this pic.  i will send you their some suggestion if possible for Tx and Rx gain because your first experimentation of LimsdrMini with OAI also gives such things

Dastgsm, 2020-08-07:
if on usrp it's see the network , that's means that is a gain issue if not it'could be that some operators transmits in this frequency

Dastgsm, 2020-08-08:
Have a look on our discussion on it : https://discourse.myriadrf.org/t/oai-and-limesdr-mini/3058

Did you think, XTRX could support OAI5G-NR or not?

DevD3339, 2020-10-28:
> Yes, I did fork IRIS lib, that uses soapy.

Could you please share the working code of xtrx integrated with oai. Thanks

Crym4n, 2021-01-29:
Hi, i am also very interested in your solution? Could you give more information about how you integrated an XTRX into OAI?