xtrx-sdr/images #84: Bus error when running test XTRX Pro PCIe with a Jetson Xaver NX

enjector, 2020-07-02:
I have installed XTRX pro with a Jetson Xaver using PCIe. All looks good but I get a bus error when I try the test. Any ideas?

$ sudo lspci -v -d 10ee:
[sudo] password for user: 
0004:01:00.0 Memory controller: Xilinx Corporation Device 7012
	Subsystem: Xilinx Corporation Device 0007
	Flags: bus master, fast devsel, latency 0, IRQ 554
	Memory at 1740010000 (32-bit, non-prefetchable) [size=4K]
	Memory at 1740000000 (32-bit, non-prefetchable) [size=64K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable+ Count=4/16 Maskable- 64bit+
	Capabilities: [60] Express Endpoint, MSI 04
	Capabilities: [100] Device Serial Number 00-00-00-00-12-34-56-78
	Kernel driver in use: xtrx
	Kernel modules: xtrx

$ sudo ./test_xtrx -t -l2
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
Bus error

enjector, 2020-07-02:
I just tried increasing the coherent_pool, but I still got the same problem:

user@user-desktop:/boot/extlinux$ more extlinux.conf
TIMEOUT 30
DEFAULT primary

MENU TITLE L4T boot options

LABEL primary
      MENU LABEL primary kernel
      LINUX /boot/Image
      INITRD /boot/initrd
      APPEND ${cbootargs} quiet **vmalloc=512M cma=64M coherent_pool=32M**

enjector, 2020-07-02:
Just debugging (printing 8-) the code and found that it fails when it writes during ioctl_num 0x123459:
``
static void xtrx_update_rxdma_len(struct xtrx_dev *d, struct xtrx_dmabuf_nfo *pbufs, unsigned len)
{
	if ((GET_HWID_COMPAT(d->hwid) >= 1)) {
		uint32_t pktsz = (unsigned)((len / 16) - 1) | (1U << 31);
		**xtrx_writel(d, UL_RXDMA_ADDR + 32, pktsz);**
	} else {
		xtrx_set_dma_bufs(d, pbufs, 0x800, len);
	}
}
``

Any ideas?

enjector, 2020-07-03:
Looks like I found the problem using gdb ./test_xrtx

In xtrxll_pcie_linux.c:503: this is not supported on the ARM proc:

	uint32_t icnt = __atomic_exchange_n((uint32_t*)dev->mmap_stat_buf + XTRX_KERN_MMAP_RX_IRQS,
										0, __ATOMIC_SEQ_CST);

Workaround (a kludge!):

	uint32_t* p = (uint32_t*) dev->mmap_stat_buf + XTRX_KERN_MMAP_RX_IRQS;
	uint32_t icnt = *p;
	*p = 0;

Do this for TX on line 642

And now I get a good result:

RX:

$ ./test_xtrx -t -l2
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4182 us DELTA     61 us LATE   2195 us 16384 samples
RX STAT Overruns:0
**Success!**
Processed 1 devs, each: RX 2 x 1.836 = 3.672 MSPS (WIRE: 14.688163)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s)

TX:

$ ./test_xtrx -T -l2
Master: 64.000 MHz; RX rate: 0.000 MHz; TX rate: 4.000 MHz
TX tunned: 450000000.000000
TX bandwidth: 2000000.000000
TX PAD gain: 0.000000
TX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED TX SLICE 0/0: res 0 TS:       0        250 us DELTA     58 us LATE  -1739 us  16384 x 1 samples (16384)
TX STAT Underruns:0
**Success!**
Processed 1 devs, each: RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 21.623 = 43.247 MSPS (WIRE: 172.987583 MB/s) 

TODO: Need too look at the supported atomic exchange primitives.

RFNOD, 2020-09-30:
Hi, Do we replace the line with your workaround or add below it? and is it line 632?

[image]

RFNOD, 2020-09-30:
Also, does line 503-505 need to be uncommented? I cant tell from the instructions above what it should be. Thanks.

RFNOD, 2020-11-23:
Bump....Can anyone help with this?

Thank you!!

RFNOD, 2020-12-02:
@enjector can you help me with this, please? Thanks!!

enjector, 2020-12-09:
Hi, just saw your messages. Yes, I had to replace those lines at 503.