xtrx-sdr/images #85: Update installation instructions/plan for support?

nedonatelli, 2020-07-12:
I had quite the rollercoaster getting everything up and running for this library/talking to the board. After having to clean install Ubuntu at least once in the process, I was able to get things running. 

The installation instructions call for python 2 installation because "cheetah is not supported by python 3", which may have been correct at the time, but Cheetah3 exists now, and the build supports it. 

Just because I struggled with documentation to support any issues I had with building, SoapySDR was the best way to get something to communicate with the board in an efficient manner.

Like other people have found, things with the sample rates get fuzzy after 70 MHz, I saw the issue on modifying the VIO and the other issue regarding changing the buffer size on the board. While it's nice to know there are potential ways to make this board work, the support to translate these changes into troubleshoot approaches seems to be lacking. 

Is there any plan to continue supporting this project or any type of template plans to include more support for the low/high level APIs?

RFNOD, 2020-10-29:
Hi,

I am not part of the Fairwaves team, just more of an interested party. I am working on updating the documentation as I am testing and making sure it all works on my end. At some point, I will submit a pull request to update the instructions. As for future support, I have reached out to them via multiple methods to no avail. Hopefully, someone responds in the near future.

RobertK66, 2020-10-30:
Hi both,
I used this SDR 2 years ago and last week I had to move to a new PC. I also struggled to get something running there. Finally I used a Linux Mint 19.3 and the provided binary images for the libraries. But I think the gr-osmosdr module from XTRX code branch is broken (at least with the pre-build binaries ) and does not initialize my XTRX with current versions. So I had to hard code the USB device name in the sources as a quick fix now (Currently I am using the USB Adapter, but I plan to switch to an PCIe Adapter once the hardware arrives).  

@RFNOD : I am willing to help improve on documentation of current status by trying out and reviewing with my setup here. 

My plans are to use this SDR in a https://network.satnogs.org/ ground station. Currently I am trying to get my head around, which layers of libraries are best to use and how to integrate into other software without too much coding/changes on my behalves ...

So every peace of documentation of the current state of all APIs would definitely help a lot!

OhSoGood, 2020-11-05:
@RobertK66  : if you manage to have your XTRX work with the PCIe adapter, I'd glad to have details. 

I've installed mine on Linux Mint 19.3 for a while, managed to compile everything (I may provide info if needed), ran test_xtrx without error (but I don't know how to check its results), and I also run SoapySDR or osmocom based app without error but I always get a flat signal.

To my great disappointment, Fairwvare team has never answered me and had provided with no support. I even wonder sometimes if they are alive (their CTO has left a few months ago).

RobertK66, 2020-11-05:
@OhSoGood : OK, i will let you know when the adapter arrives. (this could take some time - there was a 3 week delivery schedule indicated and it has to pass customs here. so pls. be patient with me ...)
With my hard coded USB settings my gnuradio recording worked good. Only - when handling other apps or moving/resizing the gnuradio windows sometimes it gets stuck completely and i have to restart everything.

OhSoGood, 2020-11-05:
@RFNOD : are you working with the USB or the PCIe version of the XTRX ? And is it a XTRX (CS, 1st gen) or a XTRX Pro ?

RFNOD, 2020-11-23:
@OhSoGood  I am using the PCIe version of the XTRX Pro.

@RobertK66 - here is some info on where a fork of gr-osmosdr that works with the XTRX.  Apparently, it has been merged with the Master branch. I haven't tested it yet. [gr-osmosdr PR#20](https://github.com/osmocom/gr-osmosdr/pull/20).

RFNOD, 2020-12-02:
Updated instructions have been merged.

OhSoGood, 2020-12-27:
Even with latest update (from early December), my XTRX still receive a flat spectrum, with seldom jumps. I wonder if the board is worth trying again.

@RobertK66 : Have you managed to use the XTRX with the PCIe adapter?

@RFNOD: What's your status with your own board? Do you manage to receive and/or transmit?

Merry Christmas to All!

RobertK66, 2020-12-29:
Hi, I have got the board now, but no time to try till now, sorry. Hope i get hands on the hardware some when in January.