xtrx-sdr/images #87: XTRX signal issue

arun1969, 2020-08-29:
I got new xtrx.
installed driver on Ubuntu 19.04 from repo https://launchpad.net/~fairwaves/+archive/ubuntu/xtrx

and compiled osmo-trx from https://github.com/xtrx-sdr/osmo-trx

I can see signal on spectrum analyzer but phone can't register.
I checked Spectrum analyzer not demodulating GMSK.

Let me what mistake i am doing.

Arun

arun1969, 2020-08-31:
test_xtrx output
lcom@elcom-desktop:~$ /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx -t -l2
CPU Features: SSE2+ SSE4.1+ AVX+ FMA-
Master: 32.000 MHz; RX rate: 4.000 MHz; TX rate: 0.000 MHz
RX tunned: 900000000.000000
RX bandwidth: 2000000.000000
RX LNA gain: 15.000000
RX PGA gain: 0.000000
RX TIA gain: 9.000000
RX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED RX SLICE 0 /0: res 0 TS:    8192       4168 us DELTA      7 us LATE   2127 us 16384 samples
RX STAT Overruns:0
Success!
Processed 1 devs, each: RX 2 x 1.958 = 3.916 MSPS (WIRE: 15.662735)    TX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000 MB/s)

arun1969, 2020-08-31:
elcom@elcom-desktop:~$ /usr/lib/x86_64-linux-gnu/xtrx/test_xtrx -T -l2
CPU Features: SSE2+ SSE4.1+ AVX+ FMA-
Master: 64.000 MHz; RX rate: 0.000 MHz; TX rate: 4.000 MHz
TX tunned: 450000000.000000
TX bandwidth: 2000000.000000
TX PAD gain: 0.000000
TX CYCLES=1 SAMPLES=16384 SLICE=16384 (PARTS=1)
PROCESSED TX SLICE 0/0: res 0 TS:       0        120 us DELTA     11 us LATE  -1916 us  16384 x 1 samples (16384)
TX STAT Underruns:0
Success!
Processed 1 devs, each: RX 2 x 0.000 = 0.000 MSPS (WIRE: 0.000000)    TX 2 x 53.720 = 107.440 MSPS (WIRE: 429.761171 MB/s)

arun1969, 2020-10-07:
I have captured tx IQ data writeSamples function in XTRXDevice.cpp
got success in demodulating using matlab.

capture XTRX IQ data over the air and demodulate in matlab, no success.
even i tried to capture data of a active GSM 2G of running BTS on bladerf and use same matlab code.
it is decoding perfectly.

What is wrong with osmo-trx-xtrx radio interface code.
there is no problem if i send sample data using test_xtrx.

Looking for valuable input

arun1969, 2020-10-08:
can any one succeeded using **XTRX** with osmo-trx.
@chemeris 
requesting for response as it has been very frustrated.

gorkemmm, 2020-12-10:
I got a handset see the bts by swapping the I/Q on transmit. 

Try
--- a/Transceiver52M/device/xtrx/XTRXDevice.cpp
+++ b/Transceiver52M/device/xtrx/XTRXDevice.cpp
@@ -215,6 +215,10 @@ bool XTRXDevice::start()
                params.tx.flags |= XTRX_RSP_SWAP_AB | XTRX_RSP_SWAP_IQ;
        }
 
+        // TODO(gorkem): Investigating flipped spectrum issue.
+        params.tx.flags |= XTRX_RSP_SWAP_IQ;
+        LOG(INFO) << "(gorkem) Flipped XTRX_RSP_SWAP_IQ on";
+
        params.tx_repeat_buf = NULL;
        params.rx_stream_start = initialReadTimestamp();

This was in May '20, branched from I believe fairwaves/libxtrx-wip branch at the time. Things may have changed since.
This was just a quick try. There must be a better way to do this; or at the least the iq swap should be done in the config file.
I never had a chance to continue to solve the next problem where the handset couldn't connect to the bts.
My setup was an xtrx-pro over usb-c.

arun1969, 2020-12-15:
> 
> 
> I got a handset see the bts by swapping the I/Q on transmit after noticing that the frequency correction burst was flipped.
> 
> Try
> 
> ```
> -- a/Transceiver52M/device/xtrx/XTRXDevice.cpp
> +++ b/Transceiver52M/device/xtrx/XTRXDevice.cpp
> @@ -215,6 +215,10 @@ bool XTRXDevice::start()
>                 params.tx.flags |= XTRX_RSP_SWAP_AB | XTRX_RSP_SWAP_IQ;
>         }
>  
> +        // TODO(gorkem): Investigating flipped spectrum issue.
> +        params.tx.flags |= XTRX_RSP_SWAP_IQ;
> +        LOG(INFO) << "(gorkem) Flipped XTRX_RSP_SWAP_IQ on";
> +
>         params.tx_repeat_buf = NULL;
>         params.rx_stream_start = initialReadTimestamp();
> ```
> 
> This was in May, I believe on the fairwaves/libxtrx-wip branch at the time. Things may have changed since.
> This was just a quick try. There must be a better way to do this; or at the very least the iq swap should be done in the config file.
> 
> I haven't had a chance to continue to solve the next problem where the handset couldn't connect to the bts.
> 
> My setup was an xtrx-pro over usb-c.

Thanx For your Help 
It works in Tx Direction and i can see signal and able to decode in spectrum analyzer.
but hand phone is not getting registered, no log on osmo-bts, osmo-bsc and wireshark.