xtrx-sdr/images #91: GPSD Permissions Issues

arzulo, 2020-10-13:
I'm having an issue with permissions between my XTRX and gpsd to try and read GPS data.  No matter how I format permissions, I end up with the error...
``` sudo gpsd -D 5 -N -n /dev/ttyXTRX0
gpsd:INFO: launching (Version 3.20)
gpsd:IO: opening IPv4 socket
gpsd:IO: opening IPv6 socket
gpsd:INFO: listening on port gpsd
gpsd:PROG: NTP: shmat(14,0,0) succeeded, segment 0
gpsd:PROG: NTP: shmat(15,0,0) succeeded, segment 1
gpsd:PROG: NTP: shmat(16,0,0) succeeded, segment 2
gpsd:PROG: NTP: shmat(17,0,0) succeeded, segment 3
gpsd:PROG: NTP: shmat(18,0,0) succeeded, segment 4
gpsd:PROG: NTP: shmat(19,0,0) succeeded, segment 5
gpsd:PROG: NTP: shmat(20,0,0) succeeded, segment 6
gpsd:PROG: NTP: shmat(21,0,0) succeeded, segment 7
gpsd:PROG: successfully connected to the DBUS system bus
gpsd:PROG: shmget(0x47505344, 24024, 0666) for SHM export succeeded
gpsd:PROG: shmat() for SHM export succeeded, segment 22
gpsd:INFO: stashing device /dev/ttyXTRX0 at slot 0
gpsd:PROG: no /etc/gpsd/device-hook present, skipped running ACTIVATE hook
gpsd:INFO: SER: opening GPS data source type 2 at '/dev/ttyXTRX0'
gpsd:ERROR: SER: device open of /dev/ttyXTRX0 failed: Permission denied - retrying read-only
gpsd:ERROR: SER: read-only device open of /dev/ttyXTRX0 failed: Permission denied
gpsd:ERROR: initial GPS device /dev/ttyXTRX0 open failed
gpsd:ERROR: can't run with neither control socket nor devices open```

I've gone through the setup at [https://github.com/xtrx-sdr/images#non-root-access-for-devxtrx0](url) in order to have non-root access, as well as all other setup procedures in the images repo.  

I am able to see GPS data however by connection directly through serial via `minicom -D /dev/ttyXTRX0` 

Any idea what's going on?

mind22, 2021-08-13:
I had the same problem and solved it by adding "XTRX" into the apparmor config file (Ubuntu 20.04).

Steps to do:
```
sudo nano /etc/apparmor.d/usr.sbin.gpsd

# add 'XTRX' after 'AMA' into following lines:

# ------------------------------------------------------------------------------
# common serial paths to GPS devices
/dev/tty{,S,USB,AMA,ACM,XTRX}[0-9]*    rw,

# default paths feeding GPS data into chrony
/{,var/}run/chrony.tty{,S,USB,AMA,XTRX}[0-9]*.sock rw,
/tmp/chrony.tty{,S,USB,AMA,XTRX}[0-9]*.sock rw,
# ------------------------------------------------------------------------------

sudo systemctl restart apparmor

# test gpsd if service disabled, else restart service gpsd
sudo gpsd -N -n -D 5 /dev/ttyXTRX0 /dev/pps0
```

br

calebschmurr, 2024-09-02:
Thanks for the help - I had a similar issue when using gpsfake.

I was receiving:
``gpsfake:ERROR: SER: device open of /dev/pts/4 failed: Permission denied - retrying read-only``

I had to add
''/dev/pts/[0-9]*    rw,``
to the apparmor file.  I don't think it matters where you stick that line, I placed it under the 
``#pps related devices``
section.

Restart apparmor service as mind22 outlined.