xtrx-sdr/images #96: Python binding for XTRX [QST|

RFNOD, 2020-12-10:
Has anyone been successful in building a python3 binding for the XTRX?

curtcorum, 2021-01-17:
also related to #93 #89 #82 #83

Part of the problem seems to be that the instruction in README.md :

`git submodule update`

Leaves all the individual submodule repositories in detatched head stale states.

It took a long time but figured out that my xtrx_api.h header file was old after fresh build:

```
curt@nye:/usr/local/include$ ls -alh xtrx_api.h 
-rw-r--r-- 1 root root 15K Jan 10 16:36 xtrx_api.h
```

and had the old xtrx_open_list in it:

```
/** Open XTRX device form semicolon separated device list
 * @param devices   Path to XTRX devices, semicolon separated (returned from xtrx_discovery)
 * @param flags     Semicolon separated flags. Can be NULL.
 * @param[out] dev  XTRX device handle
 * @return number of devices on success, errno on error
 *
 * When @ref devices is NULL only first enumerated device is created.
 * Only 'loglevel' flag is parsed.
 */
XTRX_API int xtrx_open_list(const char* devices, const char* flags, struct xtrx_dev** dev);
```
Here is the git submodule state after the 'git submodule update' command:

```
curt@nye:~/src/images$ git submodule foreach git branch -v
Entering 'sources/liblms7002m'
* (HEAD detached at 38bfb48) 38bfb48 Migration to Python 3
  master                     b07761b Migration to Python 3
Entering 'sources/libusb3380'
* (HEAD detached at da900c7) da900c7 proper initialization to work with short packets
  master                     c83d1e9 fix Windows build
Entering 'sources/libxtrx'
* (HEAD detached at 2443a10) 2443a10 gpio & sync API
  master                     acb0b1c xtrx_open_string: properly terminate 4096-byte paramstrings (#19)
Entering 'sources/libxtrxdsp'
* (HEAD detached at 84c6f4c) 84c6f4c make ARCH check valid for all x86 family
  master                     eec2864 Add FFT accelleration functions
Entering 'sources/libxtrxll'
* (HEAD detached at 9106b98) 9106b98 add ability to work with any packet size using short packets
  master                     1b6eddf Fixing typos (#5)
Entering 'sources/xtrx_linux_pcie_drv'
* (HEAD detached at ab29a2b) ab29a2b update package to version 0.0.1-2
  master                     5ae3a3e aarch64: fixup DMA for RK3399
```

I am now trying a build with:

`git submodule foreach git checkout master`

which gives the submodule status:

```
curt@nye:~/src/images$ git submodule foreach git branch -v
Entering 'sources/liblms7002m'
* master b07761b Migration to Python 3
Entering 'sources/libusb3380'
* master c83d1e9 fix Windows build
Entering 'sources/libxtrx'
* master acb0b1c xtrx_open_string: properly terminate 4096-byte paramstrings (#19)
Entering 'sources/libxtrxdsp'
* master eec2864 Add FFT accelleration functions
Entering 'sources/libxtrxll'
* master 1b6eddf Fixing typos (#5)
Entering 'sources/xtrx_linux_pcie_drv'
* master 5ae3a3e aarch64: fixup DMA for RK3399
```

anyway really hopes this explains things!

RFNOD, 2021-03-04:
Was this fixed with [PR#99](https://github.com/xtrx-sdr/images/pull/99)? If so, let's close this issue. Thanks!!