xtrx-sdr/images #97: Dry solder for J2 resistor

trabucayre, 2020-12-15:
This is not an software issue but hardware. I post this story for people facing to the same issue.

I've bought an XTRX pro (xc7a50t) and the PCIe adapter board through mouser website.

I've received both the last week, plugged in many different computers but, each time,  unable to see the board (and no blinking led).

I have updated the bitstream (XTRX powered by USB and with a **digilent hs2** cable). I've used [openFPGALoader](https://github.com/trabucayre/openFPGALoader), since unable to build [xc3sprog](https://github.com/xtrx-sdr/xc3sprog). When powered by USB, the led blink.

Same issue with all computer.

To be more comfortable to continue debugging I've used a ARM based board with PCIe connector. -> Not recognised.

I have built a new firmware from source, tried to flash this new bitstream, with my cable, but, this time, left connected to PCIe instead of using USB cable. Unable to detect the FPGA.

With a multimeter I have seen 3.3V on J2 connector but not on X8.1, second thing, when multimeter probe is applied on J2 resistor led start to blink -> dry solder.

So, if your XTRX is not detected by your computer, check power supply and if J2 resistor is correctly solder.

chemeris, 2020-12-15:
Thank you for sharing this. I'm a bit surprised we missed this during the QC testing. We'll update our QC procedures.

trabucayre, 2020-12-15:
Great!
It's a problem easy to fix but hard to detect and it's a bit frustrating to spend several days to wonder if the board is broken or not.
Thanks