xtrx-sdr/images #98: Ubuntu 20.04 - manual compilation of host libraries does not work

76nic, 2020-12-20:
Hi there!
The manual compilation of host libraries does not work for Ubuntu 20.04. Was someone able to compile them on Ubuntu 20.04? Not sure what the problem is - looks like problems with the qt version.
Can someone please post instructions?
Many thanks!

b3sholt, 2020-12-21:
I just did this.  You can follow the instructions on the images home page. If you post exactly where you get stuck, I may be able to help.

76nic, 2020-12-21:
Many thanks for looking into this! I can not install the libqcustomplot1.3 as required Ubuntu dependency. I was able to install it under Ubuntu 18.04 but not under Ubuntu 19.10 and 20.04. What is the "libqcustomplot1.3 equivalent" under 20.04? Again many thanks!

b3sholt, 2020-12-21:
For 20.04 I just simply left that out and everything appeared to build without problems.

76nic, 2020-12-21:
I did the same but "make" results in the following error:

/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libqcustomplot.so: undefined reference to `QString::arg(QLatin1String, int, QChar) const@Qt_5'
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libqcustomplot.so: undefined reference to `qt_version_tag@Qt_5.12'
collect2: error: ld returned 1 exit status
make[2]: *** [libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/build.make:157: libxtrx/examples/xtrx_fft/mainwindow] Error 1
make[1]: *** [CMakeFiles/Makefile2:687: libxtrx/examples/xtrx_fft/CMakeFiles/mainwindow.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Is 20.04 running QT5? Not sure what to do here.
Many thanks!

b3sholt, 2020-12-21:
I just reinstalled Ubuntu to verify and here is what I did:
Normal installation
-Download updates while installing Ubuntu
-Install third party software

Once installed:
sudo apt update
sudo apt upgrade
sudo apt install gnuradio
reboot

Then, follow the instructions on the images page, removing libqcustomplot1.3 from the dependency install.  xtrx is passing the "xtrx_test".

Are you planning on using GnuRadio?

76nic, 2020-12-21:
Many thanks! Yes, I can confirm that it works on a fresh installed Ubuntu 20.04. There must be something wrong with my "older" installation. Thank your for your help!

However, I was not able to build gr-osmosdr.

make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-osmosdr.so.0.2.0.0'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:399: lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Are you using gr-osmosdr to have a "XTRX source" in GnuRadio?

b3sholt, 2020-12-22:
I too have been having issues with gr-osmosdr.  I have tried a couple different flavors and none have worked thus far.  I think something may be broken with newer versions of GnuRadio and the xtrx, but they are working on it.  See issue #89.  Go ahead and close this issue as we have now resolved it, but keep watching #89 for gr-osmosdr fix.

76nic, 2020-12-22:
Many thanks!!! You have helped me a lot moving things forward a bit!

closed 2020-12-22 by 76nic