xtrx-sdr/images #100: DKMS build error using manually built kernel (ARM64 SBC)

tecandrew, 2021-03-02:
Issue initially came up when attempting to build and DKMS complained about not being able to find it's kernel headers. 

```
andrest@firefly:~/workspace/xtrx/images/sources/build$ sudo /usr/sbin/dkms build -m xtrx -v "0.0.1-2"
Error! Your kernel headers for kernel 4.4.194-59119-g968ba5005831 cannot be found.
Please install the linux-headers-4.4.194-59119-g968ba5005831 package,
or use the --kernelsourcedir option to tell DKMS where it's located
```

I found and installed the headers, and though the build step proceeds it encounters an odd error.

```
andrest@firefly:~/workspace/xtrx/images/sources/build$ sudo /usr/sbin/dkms build -m xtrx -v "0.0.1-2" --kernelsourcedir /usr/src/linux-headers-4.4.194-g968ba5005831

Preparing kernel 4.4.194-59119-g968ba5005831 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make mrproper......(bad exit status: 2)
Warning: using /usr/src/linux-headers-4.4.194-g968ba5005831/.config
(I hope this is the correct config for this kernel)
make oldconfig...
make prepare-all......(bad exit status: 2)

Building module:
cleaning build area....
make -j6 KERNELRELEASE=4.4.194-59119-g968ba5005831 -C /usr/src/linux-headers-4.4.194-g968ba5005831 M=/var/lib/dkms/xtrx/0.0.1-2/build....(bad exit status: 2)
ERROR (dkms apport): binary package for xtrx: 0.0.1-2 not found
Error! Bad return status for module build on kernel: 4.4.194-59119-g968ba5005831 (aarch64)
Consult /var/lib/dkms/xtrx/0.0.1-2/build/make.log for more information.
```

Contents from the `make.log`

```
DKMS make.log for xtrx-0.0.1-2 for kernel 4.4.194-59119-g968ba5005831 (aarch64)
Tue Mar  2 23:47:05 UTC 2021
make: Entering directory '/usr/src/linux-headers-4.4.194-g968ba5005831'
  LD      /var/lib/dkms/xtrx/0.0.1-2/build/built-in.o
  CC [M]  /var/lib/dkms/xtrx/0.0.1-2/build/xtrx.o
/bin/sh: 1: ./scripts/recordmcount: Exec format error
scripts/Makefile.build:283: recipe for target '/var/lib/dkms/xtrx/0.0.1-2/build/xtrx.o' failed
make[1]: *** [/var/lib/dkms/xtrx/0.0.1-2/build/xtrx.o] Error 2
make[1]: *** Deleting file '/var/lib/dkms/xtrx/0.0.1-2/build/xtrx.o'
Makefile:1479: recipe for target '_module_/var/lib/dkms/xtrx/0.0.1-2/build' failed
make: *** [_module_/var/lib/dkms/xtrx/0.0.1-2/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.4.194-g968ba5005831'
```

I'm using an ARM64 SBC (Firefly-RK3399) so it seems I have to rebuild the necessary files from source, including this.

RFNOD, 2021-03-04:
Hi, interesting. Will see if I can replicate this with a Jetson AGX as it is an Arm device. Give me a couple of days.

Interceptor777, 2021-06-21:
Compile the drivers using your kernel sources instead

make -C "path to your kernel sources"

and then you can install the xtrx.ko which results from that.