xtrx-sdr/images #102: xtrx: Unable to initialize UART driver: -16

maxspb89, 2021-03-27:
I've tried to install XTRX with PCI-e adapter on Ubuntu18.04 and Ubuntu20.04. but I've got errors during driver installation.
`sudo modprobe xtrx`

 fails with `"modprobe: ERROR: could not insert 'xtrx': Device or resource busy"`.
results of dmesg:
`xtrx: Unable to initialize UART driver: -16`
Host system is Ubuntu 18.04
uname -a : Linux test 5.4.0-70-generic #78~18.04.1-Ubuntu SMP Sat Mar 20 14:10:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux.

Tried to build driver manually, and recieving the same error after sudo insmod xtrx.ko.
Then I've tried to remove UART checking in static int __init xtrx_init(void) and rebuild again but no PCI device is shown after
`sudo lspci -v -d 10ee:` command so I suppose, that driver hasn't been installed.

cpjenkins, 2021-05-16:
So I just ran headlong into this issue as well (Ubuntu 20.04 5.8.0-53). It looks like the uart driver major version is hardcoded as **234** in [xtrx.c](https://github.com/xtrx-sdr/xtrx_linux_pcie_drv/blob/5ae3a3e877e5b04da6e1cfdcca2d2f124857ddb0/xtrx.c#L529.) as 

I am far from an expert on kernel programming... but I believe 234 is the first major device available in the dynamic allocation block ([kernel docs](https://github.com/torvalds/linux/blob/63d1cb53e26a9a4168b84a8981b225c0a9cfa235/Documentation/admin-guide/devices.txt#L3023))... i.e. some other device on your system is getting a dynamic major number before the xtrx uart driver. 

As a workaround, take a look at your registered devices  (`cat /proc/devices`) and pick a major version that isn't listed. This worked for me.

Interceptor777, 2021-06-22:
I can confirm the solution from @cpjenkins solved this issue for me. However, it's worrying that this is still hard coded.