xtrx-sdr/images #107: xtrx test fails: stuck at writing PCIe registers

samlei-research, 2021-09-10:
I am executing the `./test_xtrx -t l9` and get the following output:

7:44:19.719508 DEBUG:  xtrxllpciev0_discovery:263 [PCIE] pcie: Found "pcie:///dev/xtrx0"

17:44:19.719709 REGS:   internal_xtrxll_reg_in:155 [PCIE] PCI:/dev/xtrx0: Read  [001a] = 04000113

17:44:19.719729 INFO:   _xtrxr3_board_combctrl:513 [CTRL] PCI:/dev/xtrx0: RFIC_GPIO 0x000300

17:44:19.719748 REGS:   internal_xtrxll_reg_out:146 [PCIE] PCI:/dev/xtrx0: Write [0007] = 00000300

17:44:19.719765 REGS:   internal_xtrxll_reg_out:146 [PCIE] PCI:/dev/xtrx0: Write [0003] = 0b008804

17:44:19.719780 REGS:   internal_xtrxll_reg_out:146 [PCIE] PCI:/dev/xtrx0: Write [0003] = 87000000

What could be the reason for that? The driver is loaded correctly and has created `/dev/xtrx0` & `/dev/xtrx1` . Do i need to flash a firmware image before start?

mdiepart, 2022-04-27:
Hi,
I am currently facing the same problem.
I got the device to work on a desktop computer with Rocky Linux but when I switched to a DELL server with Debian Bullseye, I have the same message.

By any chance, did you find a fix for this problem ?

samlei-research, 2022-04-27:
@mdiepart I switched to the USB 3.0 interface, as the PCIe drivers might not work on every system and there is no maintenance at the moment as far as I know.

mdiepart, 2022-05-03:
Just so you know (because the PCIe bus offers higher speed than the USB interface, and because somebody else might encounter this problem).
It seems that there is an incompatibility between the libraries/drivers and the linux kernel version 5 and above.
By installing Debian 10 instead of Debian 11, it is now working as expected.

If somebody wants to try having a look, the device is hanging at the time of enabling the dc/dc converter (line 1200, libxrtxll/xtrxll_base.c). The device tries to pread /dev/xtrx0 but it fails because the file descriptor reports as busy. I do not know if the problem comes from the kernel driver or the libraries.